steff393/wbec #1: Ladestrom regulieren

mli987, 2021-04-16:
Hallo,
das klingt sehr interessant. Danke vorab. 
Ich will die Box ansprechen, um die Ladeströme bedarfsorientiert zu regulieren/steuern. 
Können Sie mir helfen, wie ich die Box am einfachsten ansteuern kann, wenn der Adapter installiert ist?

steff393, 2021-04-16:
Hallo,
danke für das Interesse. 
Ja, das sollte machbar sein. Im Prinzip hab ich es mir so vorgestellt, dass man den Ladestrom über den Aufruf 
http://192.168.xx.yy/json?currLim=130
einstellt (hier z.B. auf 13A).

mli987, 2021-04-16:
Und der Adapter wird wie angesprochen? Einbindung ins Netzwerk? Permanentbetrieb? Wie wurde das gelöst? Schon getestet?

steff393, 2021-04-16:
Bei mir sieht das aktuell so aus:
[image]
Die NodeMCU verbindet sich ins WLAN und ist dann permanent erreichbar. Per http://192.168.xx.yy/json kann man die Werte abrufen (s. Readme). Registern verändern klappt, die Kommunikation läuft auch über mehrere Stunden, aber ich hab's noch nicht mit einem Auto getestet.

derwolff, 2021-04-25:
Hallo, ich klinke mich mal hier ein.
Klasse dass ich jetzt nicht bei null starten muss. Ich setze bei mir Home Assistant ein und aufgrund der guten Integration möchte ich die Steuerung über esphome umsetzen. Falls es zu aufwändig wird würde ich allerdings auf MQTT ausweichen. Die Wallbox ist bestellt, sobald sie da ist werde ich mal damit rumspielen.
Passt der ESP eigentlich direkt in die Box und kann man irgendwo 5V anzapfen?

mli987, 2021-04-25:
Bei mir läuft das jetzt, danke Stefan. 
Was meint ihr zu solaranzeige und evcc? Wird das mit der Heidelberg und uU sogar mit dem ESP laufen? In die Box passt der meiner Meinung nach nicht.

steff393, 2021-04-25:
Bei mir kommt der ESP in ein kleines Gehäuse und das dann in den Zwischenraum zwischen dem inneren Kunststoffgehäuse und der Abdeckung (magnetisch oder mit Klebeband), wo eh die Öffnung ist. Ich stell mal ein Foto ein, wenn ich's hab.
Wg. 5V abgreifen hab ich schon mal geschaut, aber nichts Offensichtliches gefunden. Will auch nicht löten (Garantie), daher nehm ich ein altes USB-Ladegerät.
EVCC gefällt mir sehr gut; hab es gestern probeweise installiert und werd in den nächsten Tagen mal schauen, wie sich die Anbindung am besten bewerkstelligen lässt. Ich glaub das wäre eine sinnvolle Aufteilung: wbec übernimmt die Modbus-Kommunikation und stellt es dann per HTTP/JSON an EVCC kompakt zur Verfügung.

steff393, 2021-04-27:
So sieht es jetzt bei mir aus:
Das graue Kabel geht an Modbus-IN und das schwarze zum USB-Ladegerät.
[image]

s0170071, 2021-04-27:
> 
> 
> So sieht es jetzt bei mir aus:
> Das graue Kabel geht an Modbus-IN und das schwarze zum USB-Ladegerät.
> [image]

Prima! Hab mich gerade vorhin über das RS485 interface geärgert. 
Ich hatte angedacht den ESP am Sicherungskasten zu lassen. Da sitzt er trocken und nahe beim Router. Und von dort mit nem 20m Kabel die RS485 bedienen. Ist angedacht die Modbus ID per Parameter mitzugeben? Ich plane mit zwei Wallboxen und würde die beide an das selbe Interface anschließen.

steff393, 2021-04-27:
Ja, das ist vorgesehen, z.B.:
`http://192.168.xx.yy/json?currLim=60&id=2  --> set current limit to 6A on the box with id=2 (i.e. ModBus Bus-ID=3)`

Über den Parameter id kann man einstellen an welcher Box der Wert gesetzt werden soll. Ist aber 0-basiert (im Gegensatz zur Bus-ID, die mit 1 beginnt).

derwolff, 2021-04-27:
Meine Wallbox ist tatsächlich heute angekommen. Im Gehäuse selber wäre schon genug Platz für den ESP, macht aber nur Sinn wenn man die Versorgungsspannung irgendwo abgreifen kann. Ich schau mir das mal an.
Übrigens ist bei mir der 120 Ohm Terminatorwiderstand bei S6/2, ist das eventuell noch ein Fehler in der Beschreibung der Konfiguration?

steff393, 2021-04-27:
Stimmt, laut Dokumentation müsste es der S6/2 sein. Hab bei mir den S6/1 auf On. Dann hatte ich vermutlich einfach nur Glück, dass mein Kabel recht kurz ist. Ich korrigier es, danke!
Rechts auf der unteren Platine sind 4 Pins, aber ich weiß nicht, welche Funktion die haben.

derwolff, 2021-04-28:
Ich konnte auch nicht rausfinden wofür die 4 Pins sind. Da die Platine der Box wirklich sehr sauber aufgebaut ist und ich daran nicht rumlöten möchte verwerfe ich die Idee mit dem internen ESP. Ich habe in der Garage eh schon ein Gehäuse mit Buskomponenten und Spannungsversorgung.

profex1337, 2021-04-28:
Moin zusammen, 

ich habe 2 Energy Control Wallboxen (über RS485 verbunden (und mir letztens noch einen mobilen Go Echarger gekauft, damit ich anständig denn Überschuss per openWB laden kann.

Diese Lösung hier klingt ja wunderbar und würde das Dasein meiner 2 Heidelberg WB rechtfertigen.

Ich bin zwar durchaus technisch bewandert, aber ohne Anleitung bekomme ich das hier nicht hin.

Bzw. Vorab die Frage: Kann ich die Wallboxen damit über openWB steuern?
Kann jemand eine Anleitung erstellen.
Oder baut mir das Teil jemand für 100€ ? (wäre es mir Wert, wenn es denn funktioniert mit openWB).

Danke!

mli987, 2021-04-28:
Kann man den currentmin-Wert auch runtersetzen?

kerstinbaskakow, 2021-04-28:
Hi mli, 
du kannst den minimalen Strom nur über den Drehschalter S3 einstellen. 0 = 6A. Mittel Reg (04)101 kannst du dann den Wert auslesen. Einstellen per Kommando geht nicht. Macht aber imho nicht viel Sinn, da du ja über Reg (06)261 sowieso dann steuerst und einen max Wert vorgibst. Meine HW-Config ist min. 6A und max. 16A und über den 261 wird dann der max. Ladestrom entsprechend der vorhanden PV-Leistung gesteuert.
1phasig dann 1,4kW bis 3,6kW
2phasig dann 2,8kW bis 7,2kW
3phasig dann 4,2kW bis 11kW

mli987, 2021-04-28:
Ja, so ist es bei mir ja auch. Also 3-phasig. Werte unter 4,2 sind dennoch, meiner Meinung nach, manchmal sinnvoll - wenn die PV zB nur 2,5 bringt.

kerstinbaskakow, 2021-04-28:
Yes, dass hatte ich bei Heidelberg angefragt, weil ich wissen wollte, ob für die auto. Phasenumschaltung nicht mit den vorhandenen Panasonic-Relais funktionieren würde. Aber Antwort war, "nein, das geht nur mit einer externen Schaltung". 
Daher werde ich mir jetzt vor die Wallbox nach dem 3pLS noch ein 3pPhasenschütz klemmen und dann L2 und L3 per remote schaltbar machen. 
Dann musst du entsprechend der vorhandenen PV-Leistung dann die Phasen entsprechend schalten. ABER Vorsicht, nicht alle Fahrzeuge können das.

steff393, 2021-04-28:
@profex1337 Meld dich bitte unter wbec393@gmail.com, ich denke das ist machbar. 
Für EVCC hab ich gestern mal einen Test gemacht, bei dem ich die relevanten Teile der go-e API nachbilde. Das könnte funktionieren... Mit openWB tue ich mich noch schwer, weil ich die shell-Skripte noch nicht ganz verstanden hab. Prinzipiell muss es aber auch gehen.

steff393, 2021-04-28:
@kerstinbaskakow Welches Schütz würdest du da nehmen?

kerstinbaskakow, 2021-04-28:
> @kerstinbaskakow Welches Schütz würdest du da nehmen?
z.B. ABB ESB25-40N 25A 240V für 16€ beim Hornbach. Ansteuerung über Hilfskontakt 13,14 mittels eines Tasmota Switch oder Shelly. 
L1 und N liegen dann an der WB immer an, L2 und L3 zuschaltbar. 

Ablauf:
L1, L2, L3 geschalten, 
Phasenerkennung nach Betriebszustand >=4 aktiv, 
PV-Überschuss < 4kW mit Reg 261 = 0A
L2 und L3 abschalten. ca. 1min Warten (wg. Abbau und Aufbau HV-Zwischenkreis im Fahrzeug)
Ladestart mit Vorgabe Reg 261 mit L1 und einphasigem Laden. 

Wenn dann die PV-Überschussleistung > 4kW wird, kommandiere ich wieder auf 0A (261) und Schalte Phase L2, L3 dazu, nach einer Minute dann wieder Stromvorgabe mit >=6A (aber diesmal 3 phasig)

Wichtig... ich weiß nicht ob das für einen ZOE oder KIA funktioniert.

profex1337, 2021-04-30:
@kerstinbaskakow Schafft das der Shelly ? Ich hab hier noch 2Stk Shelly 2.5 rumliegen (wollte ich eigentlich für Raffstore Steuerung). Die können nur 10A max (pro channel)

kerstinbaskakow, 2021-04-30:
Mit dem Shelly schalte ich nur das Phasenschütz. Die tatsächlichen Phasen werden dann über das Schütz selbst geschalten.

derwolff, 2021-05-02:
Falls jemand die Heidelberg Wallbox über ESPHome und Home Assistant einbinden möchte habe ich zufällig dieses universelle ModBus Projekt gefunden.
[https://github.com/martgras/esphome/tree/modbus_component/esphome/components/modbus_controller](url)

Eine Diskussion und meine Beispielkonfigurationsdatei ist hier zu finden:
[https://github.com/martgras/esphome/discussions/11](url)

steff393, 2021-05-02:
@kerstinbaskakow Hast du einen guten Kontakt zu Heidelberg oder gehst du über den normalen Support? Mich würde noch interessieren, wie man eine neue Firmware über Modbus auf die Wallbox flashen kann. Mit der Combox geht das ja. Aber das Protokoll habe ich bisher nirgends gefunden.

@mli987 Bzgl. EVCC hab ich einen eigenen Issue #2 gestartet

kerstinbaskakow, 2021-05-03:
@steff393 leider nicht. Hatte mal versucht per normalen Support Kontakt zur Technik aufzunehmen, leider ohne Erfolg.

Am Update wäre ich auch interessiert. Sollte aber imho kein großes Problem sein. Firmware + Registerbereiche. Ich denke, dass Fa. Heidelberg hier schon bald mal was zur Verfügung stellen wird. Die Combox ist mir zu teuer ;-)

derwolff, 2021-05-05:
@steff393 Du schreibst ja zurzeit alle 10 Sek. in das Register 258 den Wert 4 um den Modbus Watchdog zurückzusetzen. Also "disable Standby Funktion" .
 Das ist gar nicht nötig wenn du eh lesend auf einige Register zugreifst. Die ESPHome Erweiterung polled z.B. alle 60 Sek. die Register. Es reicht dann die Watchdog Time (Reg. 257) größer 60000 ms zu setzen.
Ich werde aber auch die Standbyfunktion der Wallbox aktiveren (Reg 258 auf 0).  Die Box und damit auch der Modbus wird erst wieder mit Verbindung zum Auto aktiviert. Warum sollte man auch die Box steuern wenn kein Auto verbunden ist?
Ich habe den Stromverbrauch der Box im Leerlauf noch nicht nachgemessen aber die PCB Temperatur liegt fast bei 40 Grad und in einem Video wurde ein Verbrauch von über 6 Watt gemessen.
Geht die Box nach ein paar Minuten in den Standby fühlt sich alles kälter an und das Schaltnetzteil fiept auch nicht mehr so unerträglich laut ;-) .

profex1337, 2021-05-05:
@derwolff Sehr guter Punkt mit dem Standby!

steff393, 2021-05-06:
Das ist ein guter Punkt. Bei meinen Tests (ohne Auto) konnte ich den Standby immer nur durch Power On Reset beenden. 
Aber wenn die Verbindung mit dem Auto das auch bewirkt, dann werd ich das demnächst mal ändern.

steff393, 2021-05-07:
Ich hab das jetzt mal geändert:
- Modbus Timeout und Standby sind jetzt konfigurierbar
- Beide Register werden nur noch beim Startup geschrieben (an alle zu diesem Zeitpunkt aktiven Slaves)
- Bei Timeout einer Box (der ja normal durch Standby verursacht werden kann) werden nicht gleich alle Werte verworfen

steff393, 2021-05-08:
Hab's heute mal getestet (Wallbox nur einphasig angeschlossen, Annahme 30ct/kWh):
```
1) Standby     1,1W =>  2,89€ pro Jahr
2) An, LED aus 3,1W =>  8,15€ pro Jahr
3) An, LED an  4,8W => 12,61€ pro Jahr
wbec alleine   1,4W =>  3,68€ pro Jahr (für bis zu 16 Boxen)
```
Allerdings erreicht man 1) ja nur wenn man nie ein Auto ansteckt. Der Zustand 3) ist normalerweise auch nicht dauerhaft. Unter der Annahme, dass während 50% der Zeit ein Fahrzeug angeschlossen ist, dürfte der Standby ca. 2,50€ pro Jahr und Box sparen. Mit PV-Strom vermutlich nochmal deutlich weniger (da die meisten Autos unterm Tag, wenn die Sonne scheint, unterwegs sein dürften).

Anderseits sehe ich aber auch nichts, was gegen den Standby spricht.

kerstinbaskakow, 2021-05-25:
Hey, wir haben heute bei dem wechselhaften Wetter versucht mit Überschuss zu laden, allerdings hatten wir zwei mal einen Abbruch am Fahrzeug. Nach genauer Analyse haben wir jetzt die WB in Verdacht. Bei MaxCurrCommand 0A wird die Spannung und CP abgeschalten, bei einer Stromvorgabe wieder eingeschalten. Beim 10. Zyklus innerhalb kurzer Zeit wurde allerdings nur noch CP geschalten und die Spannung nicht mehr.

Hattet ihr auch schon mal ein ähnliches Verhalten? Bei uns heute reproduzierbar bei kurzen Ladepulsen. (z.B. 10x in 9min). 

Am Bild mal ein Export aus dem Log. 

[image]

derwolff, 2021-05-27:
Meiner Meinung nach wirst du mit dieser Schaltfrequenz die Relais der Wallbox und wahrscheinlich auch die Ladeelektronik des Autos beschädigen. Ich würde die Ladung nur freigeben wenn z.B. 10 Min. ein stabiler Ertrag anliegt und dann auch für mindestens mehrere Minuten die Ladung aktiv lassen. Gegebenenfalls den Ladestrom auf das Minimum von 6A herunterfahren. Ich gehe davon aus dass bei einem Ladestrom von 0A die Wallbox die Verbindung zum Netz trennt.

kerstinbaskakow, 2021-05-28:
@derwolff danke für deine Rückmeldung. Jap, da hast du recht. Ich hatte noch einen Fehler im Code.... die min. Einschaltdauer stand auf 10s statt 10min... Bei 0A wird fahrzeugseitig die Phasen und CP getrennt. Somit sieht es für das Fahrzeug wie einer neuer Ladevorgang aus.

rantanplan1980, 2022-01-05:
Hallo zusammen, entschuldigt die unqualifizierte Frage. Aber wo bekomme ich die Hardware hierzu her. (Ich meine nie Plantine (oberer Teil). Wie ich einen ESP8266 kaufe weiß ich. :) Vielen Dank!

steff393, 2022-01-05:
Hi @rantanplan1980, schick mir bitte kurz eine Mail

DrGalvany, 2022-01-21:
Hallo, dieses Projekt scheint genial zu sein, über einen NodeMCU das zu steuern.
Ich weiß nicht, ob ich es irgendwo übersehen habe, aber wie, oder welche Pins der NodeMCU werden angeschlossen.

Recht herzlichen dank für deine Erfindung.
Viele Grüße Micha

zivillian, 2022-01-21:
Die Pins sind hier konfigurierbar: https://github.com/steff393/wbec/blob/cb618d805cca1a2de52d85db5a22a3b56b09ac1e/src/globalConfig.h#L17-L21

DrGalvany, 2022-01-21:
> Die Pins sind hier konfigurierbar:
> 
> https://github.com/steff393/wbec/blob/cb618d805cca1a2de52d85db5a22a3b56b09ac1e/src/globalConfig.h#L17-L21

Hallo, das ging ja fix. 
Danke für deine Antwort.
Ich sehe, da muss, ich mich doch noch etwas tiefer mit beschäftigen, (recht frisch unterwegs mit sowas).
Bin auf der Suche mit meiner Heidelberg auf diese geniale Erfindung gestoßen.
Diese Möglichkeit ist so toll und (neu?) dass es noch nicht mal einen Beitrag auf YouTube gibt. 

Nochmals Danke

profex1337, 2022-01-21:
Es ist genial. Funzt an meinen 2 heidelberg seit fast nem Jahr wunderbar

steff393, 2022-01-21:
Hallo @DrGalvany, vielen Dank für das Lob! Wenn du noch weitere Fragen zur Hardware hast, schick mir gern einfach kurz eine Mail (wbec393@gmail.com). 
Meine Video-Skills sind nicht so toll ;-)

botox-100, 2022-02-24:
Hallo, 
ich bin gerade dabei das wbec mit meiner Heidelberg energy control einzurichten. Dabei möchte ich gerne einen normalen  Schalter an der Heidelberg bzw. dem wbec anbringen mit dem man "hart" zwischen PV Überschussladen und normalem Laden umschalten kann. 
Mein Frage ist: Gibt es eine Funktion im wbec mit der man den Zustand eines der nicht verwendeten gpio pins reporten kann? Sozusagen möchte ich die Schalterstellung mit in der json Antwort auslesen. Geht das? Ich habe mal ein wenig in den Code geschaut, aber meine Fähigkeiten reichen nicht aus, dass mal eben hinzuzufügen. Evt. gibt es ja einen Parameter/eine Funktion, der/die für diesen Zweck nutzbar ist.
Abhängig von der Schalterstellung würde ein Script auf einem raspi  dann die zugehörige Konfiguration  vornehmen.  Ich möchte das zunächst mit einer Phase machen, dann aber, in einem zweiten Schritt, die weiteren Phasen (wie hier schon mal beschrieben) zuschalten.
Gruß Thomas

DrGalvany, 2022-02-24:
Hallo,
PV Überschlussladen muss doch durch eine Logik erfolgen oder übersehe ich da was?
Mir ist gar nicht bekannt, dass das "wbec" kann.
Meine "wbec" hat auch nicht wie auf dem Bild hier die Option.
"wbec" müsste doch sonst den Smart Meter auslesen oder bin ich da falsch?
Mein Überschussladen werde ich demnächst mit IOBrocker versuchen umzusetzen, das auslesen und steuern geht schon.

Michael
[image]

profex1337, 2022-02-24:
Hallo zusammen,

Ich kann openWB als Steuerung empfehlen. Funktioniert klasse!

Grüße

Stefan

steff393, 2022-02-24:
@botox-100 : Schau dir am besten mal mein Projekt hgdo an und dort die Datei src/button.cpp (speziell btn_setup() und btn_loop()). Du könnstest da die relevanten Codeteile rausziehen und bei wbec in der Datei webServer.cpp den Status btn.Pressed() abfragen und per JSON ausgeben.
Ich würde den Pin aus dieser Funktion hier nehmen (das ist D3) bzw. diese Funktion umschreiben:
https://github.com/steff393/wbec/blob/1d7db976688a5f9d187777a3d62c13221ff012a1/src/webServer.cpp#L148

steff393, 2022-02-24:
@DrGalvany : Es gibt seit kurzem ein Modul pvAlgo, das eine **sehr einfache** PV-Überschussregelung steuern kann. 
Input ist die aktuell verfügbare Leistung für das Laden, aktuell aus 3 möglichen Quellen:
- powerfox poweropti (liest den Hausstromzähler aus), s. Wiki
- per HTTP-Aufruf von http://x.x.x.x/pfox?pvWatt=1234
- SolarEdge Wechselrichter (von AMiketta)
Ich möchte aber betonen: Das ist stark vereinfacht und kommt nicht an die Genauigkeit/Komfortfunktion von openWB o.ä. ran.

Die Buttons auf deinem Screenshot sind normalerweise versteckt und werden nur dann sichtbar, wenn man diesen pvAlgo konfiguriert (cfgPvActive).

botox-100, 2022-02-25:
@steff393 Danke für den Hinweis. Ich schaue es mir an und versuche die Funktion einzubauen.

andy5macht, 2022-03-13:
> 

Genau das würde ich auch gern realisieren. Ich schau mir den Hinweis von Steffen mal an.
Falls das offiziell umgesetzt werden würde wäre natürlich der Bringer. Aber ich glaube ich schaff das schon den Code entsprechend abzuändern

botox-100, 2022-03-13:
Hallo andy5macht,

ich habe das letztendlich doch nicht mit einem Hardware - Schalter sondern mit einer Webseite gemacht, auf der man dann den Schalter "umlegen" kann. Funktioniert bis jetzt prima. Falls du dir überlegst, dass auch in SW  umzusetzen, kann ich dir gerne meinen Code als Anregung geben.
[image][image]

Die gesamte "Ladelogik" und die Anbindung an meinen Wechselrichter (Huawei, der auch Modbus spricht) habe ich per Python Script auf einem Raspi realisiert. Falls jemand eine ähnliche Anbindung machen will, kann ich meine Erfahrungen gerne weitergeben. 
Gruß Thomas

andy5macht, 2022-03-14:
Hallo @botox-100
als Hilfe, welche Variable ich überschreiben muss, damit er den Ladestrom maximal einstellt, obwohl nicht genug überschuss da ist, wäre dein Code sicherlich hilfreich.

@steff393:
ich bekomme demnächst einen Fronius Wechselrichter samt Smartmeter.
Ich werde mich daran versuchen, diesen mittels wbec direkt auszulesen und eine Überschussregelung dann ohne zusätzlichem Raspberry zu realisieren.
Hilfestellung gibt mir diese Seite:
https://www.onderka.com/hausautomation-technik-und-elektronik/nodemcu-json-daten-von-fronius-wechselrichter-einlesen

evtl. werde ich mich nochmals bei dir melden, falls ich nicht weiter komme.

botox-100, 2022-03-14:
Hi andy5macht,

du musst die Variable currLim = 160 setzen, dann lädt die WB auf allen drei Phasen mit 16 Ampere, also 11Kw, Der zugehörige Befehl lautet in etwa: http://IP-Adresse-WBEC/json?currLim=160
Mit currLim=0 schaltest du die Ladung ab
mit currLim zwischen 60 und 160 kannst du den Ladestrom variieren.
Mehr ist gar nicht nötig.
Ich versuche heute mal meine Script in Github zu laden, wenn ich etwas Zeit habe.
Viel Erfolg, Thomas

andy5macht, 2022-03-14:
Generell klar, nur wo ist der richtige Punkt den Wert zu überschreiben?
Wichtig ist hier ja, dass der Wert gespeichert bleibt und nicht über WLAN ein anderer Wert wieder geholt und gesendet wird.

botox-100, 2022-03-14:
Bin mir nicht ganz sicher ob ich die Frage verstanden habe, ich versuche mal eine Antwort.
Wenn du den Wert einmal setzt bleibt er in der WB gespeichert (vermutlich bis zum reboot der Box). Das WBEC setzt von selbst keine neuen Werte, es ist quasi das Interface für ein System wie openBW oder (wie in meinem Fall)  ein Script, das die Werte dann aktiv setzt. 
Steff393 kann dir da sicher mehr zu den Anbindungsmöglichkeiten sagen. Ich benutze nur den Weg über die einfachen HTTP-GET Requests.

steff393, 2022-03-14:
Ja, das stimmt soweit.
Es gibt mehrere Quellen, die den Stromwert setzen können. Alle nutzen (im Source-Code) die Funktion lm_storeRequest(). Aktuell sind das:
- Webserver (d.h. die HTTP-GET-Aufrufe wie oben, auch die go-e-Emulation und die Wallbox-Steuerung-App)
- WebSocket (z.B. das neue Web Interface)
- MQTT
- die RFID-Funktion
- der pvAlgo (z.B. powerFox, SolarEdge)

Wichtig ist aber nicht mehrere Wege gleichzeitig aktiv zu nutzen, sonst toggelt der Wert fröhlich hin und her :-)

@andy5macht: Wenn bzgl. der Fronius-Steuerung noch Fragen sind, sollten wir das vielleicht in einem eigenen Issue bearbeiten.

chris-wein, 2022-03-21:
Hat jemand eine Anleitung, wie man das mit ESPHome und HomeAssistant zum Laufen bekommt? 

Wäre unglaublich dankbar für eine Hilfestellung! Wir haben zwei Heidelberger Wallboxen im Hof, die bisher leider nur dumm mit Netz als auch Sonnenstrom laufen. Unser Haus ist dagegen mit HomeAssistant schon relativ smart unterwegs :-)

steff393, 2022-03-21:
Hallo Chris,
HomeAssistant sollte eigentlich auch die HTTP-Aufrufe wie oben beschrieben ausführen und den JSON-String, der als Antwort kommt parsen können.

steff393, 2022-03-21:
-------
Bitte für neue Themen eigene Issues anlegen
-------
Ich habe diesen Issue offen gelassen, damit er schnell gefunden wird, weil er doch einige interessante Infos gibt.
Mittlerweile wird der Thread aber doch recht lang (50 Beiträge aktuell) und unübersichtlich.
Bitte erstellt für neue Themen/Anfragen lieber neue, eigene Issues.
Natürlich könnt ihr auch gern eine Mail an wbec393@gmail.com senden.

raddis666, 2023-11-21:
Moin,

gibt es vielleicht eine Anleitung wie ihr das umgesetzt habt. Ich hab alles durchgelesen aber komme nicht ganz mit wie ich das integrieren soll. Hab eine Heidelberg und ein Mod Bus Gateway auf Gutscheine mit wlan. Womit besorgt ihr euch die Daten aus dem Gateway? Und wie sieht eure config dann aus?

steff393, 2023-11-21:
Hallo @raddis666 
es gibt fertige Module mit Anleitung auf https://steff393.github.io/wbec-site/

raddis666, 2023-11-21:
Hallo steff393, 

ich interessiere mich für wbecLan und wbecPremium. Bitte sende mir ein Angebot zu. 
ggf. weitere Infos... 

Zwei drei Fragen hätte ich noch, du schreibst er kommuniziert mit den SMA-Wechselrichtern und nicht mit dem SHM, ist es ein Problem das ich davon 2 habe und zusätzlich noch einen Laderichter?

Und am liebsten würde ich ja meine eigene Steuerung einbringen über HA, sollte aber über MQTT möglich sein. Weißt du auch über wieviel Meter ich das Bus Kabel verlegen kann ohne Signalverlust?

 

Mit vielen Grüßen

 

Florian Prüß

closed 2024-03-24 by steff393

frankomio, 2024-10-14:
Hallo (ich hoffe ich darf hier mal einhaken),
bei mir lässt sich weder CurrLim noch CurMax von 12a auf 16A umstellen.

Ich nutze Wbec 0.5.2 an der EnergyControl, dahinter zur Regelung EVCC  (wo auch die 16A eingestellt sind).
Aber immer wenn ich in der GUI den Regler auf 16A stelle springt er auf 12 zurück (und über die JSON habe ich quasi den selben Effekt)

Der Drehschalter für dcen Ladestrom hatte ich bei WBEC installation gar nicht verändert müsste also bei 16A (Default) stehen,,,

habe ich einen Denkfehler ?