steff393/wbec #2: Support für openWB und EVCC - Tester gesucht

steff393, 2021-05-01:
Hallo zusammen,

ein vielfacher Wunsch war, die Heidelberg Wallbox Energy Control mit openWB oder EVCC steuern zu können.

Ich habe die letzten Tage mal probeweise den Support für openWB und EVCC eingebaut.
wbec bildet jetzt den Teil der API vom go-eCharger nach, der von openWB und EVCC genutzt wird. Es ist somit keine Änderung am Code von openWB/EVCC notwendig. Stattdessen konfiguriert man einfach einen go-eCharger mit der IP-Adresse von wbec.

Die Firmware steht als Release v0.0.2 zur Verfügung: https://github.com/steff393/wbec/releases/tag/v0.0.2

Ich würde mich über Freiwillige freuen, die das Ganze mal testen und mir Rückmeldung geben. Selbst kann ich leider nur mit EVCC recht limitiert testen (kein Auto). Kann daher sein, dass gerade die Zustandswechsel/Timings noch nicht sauber funktionieren.

Hier z.B. die Konfiguration für die evcc.yaml:
```
chargers:
- name: wbec
  type: go-e
  uri: http://192.168.xx.yy
```
Einschränkung: Die go-eCharger API bietet nur eine Wallbox pro IP-Adresse an. Es funktioniert daher aktuell nur mit der Wallbox mit Modbus-Adresse 1.

profex1337, 2021-05-02:
Danke, ich freu mich auf dein Paket und werde das nächste Woche dann testen und wir bleiben in Kontakt :). Nächsten Freitag kommt mein Benz 300 T de PHEV noch hinzu, meinst Du es wird noch möglich sein die 2. Wallbox mit anzusteuern? Verbunden wären sie entsprechend über Modbus, die eine als Master, die andere als Slave...
Bzw. ich nehme an, WBEC übernimmt dann die Master Rolle.

steff393, 2021-05-03:
Das wird noch schwierig... Die go-e API gibt's nicht her, dass man unter der gleichen IP 2 Ladepunkte haben kann. Da müsste man dann openWB ändern.
Ja, WBEC übernimmt die Master-Rolle, d.h. die Wallboxen müssen alle als Slave eingestellt sein. Lass uns mal schauen wie der erste Test läuft und dann sehen wir weiter.

profex1337, 2021-05-16:
Die Tests jetzt am Wochenende liefen super! 
Ich hab nur ein Problem mit meinem Tesla.. Ich glaube mein interner Charger ist defekt, da er nur noch mit 2 Phasen läd.

Ansonsten hier mal ein paar "Kurven" von heute. (Blau umrandet = PV Überschuss Laden)
[image]

therobbot, 2021-05-17:
Hi, ich bin auf dieses Projekt gestoßen. Ich habe mich leider erst richtig eingelesen, als bei uns schon die Heidelberg Wallbox bestellt wurde. Ich stehe jetzt vor der Entscheidung, die Heidelberg Energy Control zu nehmen oder doch lieber eine OpenWB zu bestellen und nochmal 3-4 Monate zu warten. Ist denn der Status von dem Projekt so, dass ich mit Programmier- und IT-Kenntnissen es hinbekommen würde, mit der Heidelberg Wallbox und einem Kostal Smartmeter Überschussladen zu konfigurieren? Geht es nur mit OpenWB oder auch mit EVCC?

Clanchef, 2021-05-18:
Ich habe seit gestern die Heidelberg in der Garage hängen, Box angeschlossen und in OpenWB integriert, es funktioniert. Mein Elektriker war von der Heidelberg begeistert, "sehr solider Aufbau". Die OpenWB Wallbox kenne ich nicht so gut, hat scheinbar viele Funktionen, ist jedoch auch teurer als die Heidelberg. Mit der KfW Förderung ist diese Lösung fast "geschenkt" und scheint flexibler. Mit EVCC habe ich es noch nicht getestet. Da sich wbec als Go-e charger verhält ist eine Integration denkbar einfach.

profex1337, 2021-05-18:
@Clanchef 
Ich teste es heute mal mit MQTT statt go E Charger.

Ich hab 2x Heidelberg. Hab es bis jetzt aber noch nicht hinbekommen beide mit einer WBEC zu steuern.
Steff und ich sind aber dabei eine Lösung zu finden.

profex1337, 2021-05-18:
Also, auch mit MQTT und einer Box läuft es ganz gut:
[image]

profex1337, 2021-05-19:
Und dank WBEC wird ein Tag wie gestern wunderbar genutzt (der Benz läd maximal 2 Phasig):
[image]

Lader2021, 2021-06-17:
Hi zusammen ich habe die Heidelberg Wallbox Energy Control und bekomme dieses Jahr noch eine Photovoltaikanlage. Geht das was du mit openWB gemacht hast mit Solaredge Wechselrichter?
Da ich noch die Wahl habe wäre cool hier etwas Hilfestellung zu bekommen. Danke.

steff393, 2021-06-17:
Hi,
wbec hat selbst bisher keine direkte Anbindung an Wechselrichter, weil es da zu viele verschiedene gibt. Man kann aber openWB oder EVCC auf einem Raspberry installiereren und dort dann wbec als Ladepunkt, Solaredge als Wechselrichter und ggf. weitere einrichten.
Oder die Wallbox Steuerung App sollte auch funktionieren, siehe auf der [Seite ](https://android.chk.digital/ecar-charger-control/)von Christian:
- SolarEdge (das läuft nicht ideal, da man nur alle 5 Minuten an die aktuellen Daten darf, aber besser als nichts. Suche noch weitere Tester, idealerweise mit Speicher)
- SolarEdge lokal (ohne Heimakku)
...

Clanchef, 2021-06-18:
Hallo, ja das geht. Lass Dir einen vernünftigen Leistungsmesser in die PV mit einbauen. Normalerweise machen die Solarteure das über einfache pulsgesteuerte Leistungsmesser die der Solaredge zwar auswertet, nicht aber über das Heimnetzwerk rausgibt und nur übers Portal. Ich habe inzwischen einen Powerfox am easymeter Stromzähler dran, das funktioniert sehr gut.
Die Box von Stefan ist wirklich TOP!!! Danke Stefan für diese tolle Arbeit hier!

Teimue, 2022-06-27:
Als ich das gelesen hatte, dass ein Teil der go-eCharger API emuliert wird,  hatte ich gehofft, dass dann auch das iobroker Modul für den go-eCharger funktioniert. Das scheint aber nicht der Fall zu sein.
Das Modul connectet scheinbar richtig. Aber der Response Json String scheint nicht korrekt zu sein.
Es wäre gut, wenn das klappen würde. Kann man aus der Fehlermeldung herauslesen, was nicht stimmt?
[image]

steff393, 2022-06-27:
Hallo, 
wbec emuliert nur die Teile, die für die Heidelberg wirklich relevant sind. Das ioBroker-Modul prüft aber, ob wirklich alle Werte gesendet werden, auch diejenigen die eigentlich nicht benötigt werden, z.B. cch und cid (Farbwerte der LEDs am go-e-Charger).

Teimue, 2022-06-28:
OK, also müsste der go-e Adapter toleranter bei fehlenden Werten sein. Ist auch nicht mehr so wichtig.
Ich habe mir einen eigenen Import gemacht, der die Werte aus dem Json String in eigene Datenpunkte schreibt. Das ist ausreichend.

premultiply, 2022-07-11:
evcc spricht direkt nativ mit der Heidelberg EC, dort wird daher keine go-e-Charger-Emulation benötigt.
Hier wäre es daher geschickter wenn wbec nur eine einfache Modbus-RTU-IP-Umsetzung machen würde.

mmoeller53, 2023-02-23:
Hallo,
bin auch freudiger Besitzer der fertigen Version von wbec,
Nun steht die Einbindung in evcc an, ist das noch der aktuelle Stand mit der Go e Emulation? Weil auch zwischendurch was mit mqtt Einbindung gestanden hat.

steff393, 2023-02-23:
Meine Empfehlung wäre MQTT: https://github.com/steff393/wbec/wiki/Configuration-Example:-EVCC-via-MQTT

closed 2023-02-23 by steff393