steff393/wbec #7: Automatische Umschaltung zwischen 1 und 3 Phasen für die Heidelberg

steff393, 2021-05-29:
Hallo zusammen, 

ich hab mir jetzt nochmal Gedanken zu der automatischen 1-/3-Phasen-Umschaltung gemacht. Bin mir nicht sicher, wie viel Sinn es in der Praxis macht, aber in den Foren ist es ja ein vieldiskutiertes Thema/Kaufkriterium.

Auf Basis von des Ansatzes von @kerstinbaskakow (https://github.com/steff393/wbec/issues/1#issuecomment-828647021) müsste die Schaltung ja so ausschauen, oder?
[image]

Ich hab wbec jetzt mal ein Modul (src/phaseCtrl.cpp) spendiert, was den Schaltvorgang des Shelly durchführen würde. Der Ablauf wäre wie folgt:
[image]

Wenn sich die Anforderung ändert wird der verfügbare Strom (Reg. 261) auf 0A gesetzt. 
Wenn der tatsächliche Strom für 60s lang an allen 3 Phasen 0A ist, wird per http-Aufruf der Shelly An oder Aus geschalten.
Danach wird im Zustand INIT darauf gewartet, dass die Spannung an allen 3 Phasen stabil ist (entweder L1>200V und L2,L3 < 15V oder alle 3 Phasen > 200V für jeweils 60s).
In den Zuständen INIT und WAIT_0AMP wird jede Veränderung von Reg. 261 unterdrückt (aber gespeichert und bei Eintritt in NORMAL_xP wieder eingestellt).

Allerdings ist das Thema ja sehr heikel (s. Probleme bei Zoe, etc.). 

**Die Nutzung erfolgt explizit auf eigenes RISIKO! Beschädigungen an Fahrzeug oder Ladestationen sind nicht ausgeschlossen!**

Daher ist der Funktionsaufruf in main.cpp noch auskommentiert. Die Funktion ist auch nur für eine angeschlossene Wallbox (cfgCntWb=1) vorgesehen, um Timing-Probleme am Modbus (veraltete Werte) möglichst auszuschließen.

Folgende Punkte sind mir noch unklar:
- [ ] Stimmt die Schaltung so? Oder widerspricht sie gewissen Vorgaben?
- [ ] Seht ihr noch Risiken, wie ungewollt eine Änderung der Phasen während aktivem Laden stattfinden könnte? Verbindungsprobleme, Stromausfall (kurzzeitig), Relais klebt, ...
- [ ] Weiß jemand, wie man von openWB & Co. die für's Laden zur Verfügung stehende Leistung bekommt? Letztlich müsste man anhand davon die Zustände wechseln (z.B.: wenn > 4,1kW dann 3-phasig; wenn < 3,7kW dann 1-phasig)
- [ ] Sollte man die Anzahl der Umschaltungen pro Zeit noch begrenzen?
- [ ] Kann die "Pause" während des Umschaltens dazu führen, dass das Stromkontingent vom Lastmanagement an eine andere Wallbox vergeben wird?

Man könnte natürlich statt dem Shelly auch direkt ein Relais von wbec schalten lassen (oder wbec auf den Shelly packen, o.ä.). Aber preislich ist der Unterschied minimal und eine Kapselung der Funktionen macht ja auch Sinn.

kerstinbaskakow, 2021-06-10:
Hi @steff393, yes genau so habe ich es vor zu installieren. 
Ich bastle momentan aber noch an einer "Fahrzeugerkennung" um zu vermeiden, dass Nachbars Kia oder ZOE zerstört wird. So eine richtige Idee habe ich ab noch nicht. Evtl. mit einem PLC Sniffer.

iot4all, 2021-08-05:
Ihr schaltet VOR der WB mit einem extra Schütz L2/L3 on/off?
Manche WB prüfen L2/L3 ab. Da könnte es Probleme geben. Wenn es die Heidelberg nicht prüft, sollte es funktionieren.

Kann de Heidelberg CP-Unterbrechung? => Wenn nicht, kann es gefährlich werden, weil manche EV's dann keine Ladeunterbrechung detektieren

Was macht Ihr, wenn der Shelly einen Aussetzer hat oder einen reboot hinlegt? Er hat eigenen uC und ist per WLAN angebunden. Das ist keinesfalls sicher. Das können auch die 60s-Schleifen nicht zu 100% abfangen. Hinzu kommt, dass der wbec-ESP auch mal hängen kann. 

Sind das 2x 60s für eine volle Umschaltung oder nur 1x?
Bei PV-Laden ist alles >30s unangenehm. An wolkigen Tagen wird man da nicht glücklich.
Hysterese bräuchte es auch noch.

steff393, 2021-08-15:
Danke für die Anregungen. Momentan ist das Feature ja noch deaktiviert. Vor Einsatz sollte man sich natürlich damit auseinandersetzen zu was das eigene EV in der Lage ist.

Reboot vom Shelly halte ich für unwahrscheinlich, da er in diesem Szenario nichts anderes tun muss, als auf Kommando "Ein" und "Aus" zu schalten. Wenn diese Minimalfunktion nicht stabil funktionieren würde oder er Verbraucher ab und an ungewollt einschalten würde, dann würde sich das Ding nicht verkaufen :-)

Ein Hängen vom wbec-ESP dagegen wäre absolut unkritisch, da dann alles einfach so bleibt wie es ist. Das war eben auch der Sinn hinter der Aufgabenteilung wbec/Shelly.
Die Zeiten können gern parametrierbar gemacht werden und Hysterese ist ja oben bereits erwähnt.

closed 2021-08-15 by steff393

profex1337, 2021-09-23:
Hi Stefan,
der Winter rückt näher und dieses Thema wird tatsächlich für meinen Tesla interessant...

Momentan nutze ich bei bewölktem Wetter den GoE Charger an der Haushaltssteckdose.
Optimal wäre natürlich eine automatische Umschaltung der Phasen.

Einen der o.g. Shelly hab ich daheim. An die Umsetzung trau ich mich aber ehrlich gesagt nicht ran...
Bzw. wie könnte ich es dann am einfachsten, ohne viel Bastelei einbauen?

Grüße

Stefan

steff393, 2021-09-23:
Hi Stefan,
es scheitert u.a. noch an dem Trigger von z.B. openWB (das "On Request for x Phases"). Ich wüsste aktuell nicht, wie man das rauskriegt.

profex1337, 2021-09-23:
Ok schade!

Die neue Generation der GoE Charger kann die Phasenumschaltung... Hab meinen da wohl ein paar Monate zu früh gekauft.. Mist.

emmrichd, 2023-01-03:
Hallo,
ich die Umsetzung, nachdem nun einige Zeit verstrichen ist, zu empfehlen oder nicht?
Die Heidelberg WB verträgt das? Auch dauerhaft?
Woran würde ich denn bitte sehen können, ob mein Auto das verträgt?
Wäre nicht die "kleine Lösung", einfach noch eine Einzelsicherung für eine Phase zu setzen, und die im Winter raus zu nehmen?
Viele Grüße, Dieter

miguelito62, 2024-01-11:
Zuerst mal danke an Steff für seine Arbeit und an alle für die wertvollen Infos hier.

Ich habe zuerst gar nicht gerafft, dass die HB Energy Control die Phasenum(ab)schaltung gar nicht kann! Nach Inbetriebnahme und herumspielen mit wbec / Wallbox drauf gekommen und dann durch Infos hier bestätigt - Irgendjemand hatte bei HB direkt angefragt.

Egal, ich habe nun die externe Umschaltung aufgebaut. Einen Shelly zur Ansteuerung des Schütz aufzubauen ist mE etwas overengineered: Steff hat ja breits D3 als custom DOUT unterstützt. D.h. ich musste nicht mal etwas ergänzen an wbec.

An D3 habe ich für das Koppelrelais einen Open Collector Ausgang aufgebaut. Aber wer sich nicht zutraut Transistor + Basiswiderstand ran zulöten, für den gibts auf Amazon Chinesen Relaisplatinen, wo das Relais direkt an den uC Ausgang angeschlossen werden kann.

Die Umschaltung übernimmt wbec.local/gpio?on / off

Wollte zuerst auch die Umschaltlogik in wbec integrieren aber das zahlt sich nicht aus. wbec sollte schön simpel bleiben - ein WLAN interface für die HB. Die Umschaltung soll das HEMS machen. Ist ja nicht zeitkritisch. currLim rücksetzen, strom prüfen, gpio setzen, Spannung der phasen prüfen, currLim wieder hochfahren. Btw. Ich fahre mit FHEM.

Zum Aufbau: Abschaltung Ph2, Ph3 mit 2 Phasen 20A Installationschütz - siehe Bilder.
Koppelrelais hatte ich zuerst Triac basiertes aus China von Amazon - hat aber nicht mehr abgeschaltet das Teil. Also zurück und bei Reichelt relaisbasiertes Weidmüllerteil. Das funzt zuverlässig.

Btw. ich habe die Initialisierung und das Low Setzen von D3 an den Anfang von setup() gesetzt. Sonst ist das Relais beim Kaltstart eine relativ lange Zeit an. Wenn RFID verwendet wird überschreibt der I2C (oder war es SPI?) code die D3 Einstellung ohnehin wieder.

Bilder des Aufbaus:
[image]
[image]
[image]

hawa-lc4, 2024-01-11:
Hallo Michael,
zuerst einmal vielen Dank für deinen ausführlichen und gut dokumentierten Beitrag zu diesem Thema.
Und natürlich auch ein großer Dank an die engagierte Arbeit von Steff an dieser Stelle!

Ich habe an dieser Sache auch schon gebaut und bin bei der getrennten Lösung geblieben; leider ein WiFi Gerät mehr (gefällt mir auch nicht aber der Schutz meines Autos ist mir wichtiger).
Ich benutze die HB energy control, dann wbec als Steuerung der WB und als EMS nutze ich openWB.
Die Phasenumschaltung macht bei mir ein Tasmota basiertes Element mit einem Hilfsschütz (du hast eines von ABB, ich hab eines von Kopp), wichtig ist hier vor allem der Dauer-Bemessungsstrom.

Um kurz auf die Frage von Dieter im Januar 2023 einzugehen: der Wallbox ist die Phasen-Umschaltung ziemlich egal, im Grunde ist das nur eine intelligente Drohstromsteckdose, das ist jetzt sehr einfach ausgedrückt aber im Grunde ist es so.
(Solange wir hier immer von der Schaltung von L2 und L3 reden und nicht von L1 denn an der hängt die Elektronik der WB)
Der entscheidende Teil ist das EV, und ob das eine Phasenumschaltung verträgt kann nur der Hersteller sagen.

Zu der direkt schalten Lösung ist mein Rat aber ganz genau aufzupassen das dieser GPIO Ausgang keinen "Mist" macht wenn die MCU von wbec mal nicht so laufen sollte wie geplant. Wenn das Schütz im Ladevorgang ohne Vorwarnung umschaltet wird es der WB egal sein, was aber das EV daraus macht ist Glückssache. Eine kaputte Ladeelektronik ist aber kein Spaß.
Wenn wbec hier einen reboot macht paßt der Zustand des GPIO mit großer Wahrscheinlichkeit (50%) nicht zu dem Zustand den er haben sollte und diesen erst erreicht wenn wbec vollständig läuft.
Welcher GPIO ist da eigentlich egal, du hast D3 (GPIO0) genommen der beim "normalen" boot high-Potential hat, ich habe mit D8 (GPIO15 = low-Potential beim boot) experimentiert und bin auch da nicht glücklich geworden.
Anmerkung: ich habe den reboot im Web-Server Modul (webServer.cpp)  mit einem Codewort/Kommando gesichert damit der nicht versehentlich ausgelöst wird.

Meine Überlegungen sind an dieser Stelle das nur ein voll funktionstüchtiger wbec in der Lage sein sollte korrekte http-Kommandos an die WiFi-Schaltdose zu senden und damit keine Beeinflussung dieses Zustandes durch einen reboot oder Ausfall des wbec auftreten sollte. Da die Schaltdose eine separate Einheit und meine Erfahrung ist mit Tasmota das die Teile wirklich sehr stabil laufen.
Achtung!! Um nicht mißverstanden zu werden: auch wbec läuft sehr stabil!!!
Aber bei massiven modbus Problemen kommt es auch schon mal zu Fehlverhalten. Klar kann wbec nichts für diese modbus Probleme aber es ist auf alle Fälle hier der leidtragende Teil.

Gruß Günter

miguelito62, 2024-01-15:
Günter, danke für Dein Feedback. Und ja, der Punkt geht an Dich.
Das Problem, dass die Phasen während des Ladevorganges abfallen wg Fehlverhalten wbec Knoten habe ich nicht bedacht / als kritisch eingestuft.
Aber ganz klar: sobald Elektronik im Spiel ist muss man div. Fehlermodelle annehmen. Un dass so ein Teil mal abschmiert unabhängig von der Qualität der wbec Soft ist nicht unwahrscheinlich.
Also mal schauen was ich daraus mache. Wollte ursprünglich statt des Installationsschütz einen Stromstoßschütz einbauen. Bzw stattdessen könnte auch das Koppelrelais durch Stromstosschalter ersetzt werden. Damit muss wbec / oder das EMS vor dem Ladestart ermitteln in welcher Stellung der Schalter steht - wäre ja kein Problem.

Btw. durch Einsatz shelly kannst das Problem auch nicht ausschließen, denn für den gilt das gleich wie für die wbec Platine / ESP..  (Ausfallraten komplexe Elektronik vs mechanischer Schalter).

Andere Möglichkeit wäre noch die Ansteuerung des Koppelrelais über diskrete Teile wie Flip flop / buffer. Sodass der ESP eine definierte Schaltsequenz senden muss, damit das Koppelrelais schaltet. Also Absturz ESP / wbec bewirkt nicht Abfallen des Schütz.
Ich denke allerdings der Wegfall der Phase(n) während des Ladevorganges ist nicht kritisch für die Ladeelektronik in der Karre bzw weniger kritisch als das Aufschalten der Phase. Mal sehen ob ich noch Maßnahmen ergreife oder das Zeug so lasse.
Grüße
Michael

abuesen, 2024-01-15:
Hi Leuts.
Reine Neugier: Wozu ist die Phasenumschaltung gut? 
Welche Autos brauchen wirklich 6A pro Phase?

Die Heidelberg-Box macht alles unter 6A einwandfrei mit. Seit dem Herbst lade ich Steff sei Dank auch geregelten PV-Überschuss. Erst nur mit wbec jetzt mit wbec+evcc. Bei beiden Aufbauten beginne ich mit 3P-3A, rund 2 kW, mit der Ladung und schalte bei Unterschreitung von 1,3kW wieder ab. Hab testweise auch schon mit 500W geladen. Geht, aber am Ende sind die Verluste im OBC viel zu groß.

LG, Andre.

miguelito62, 2024-01-15:
Hi interessanter Kommentar! - bei meiner HB energy control ist ein Drehschalter für min. Ladestrom. Die kleinste Einstellung ist 6A. Darunter kann ich über wbec auch nix einstellen. D.h. Anforderungen unter 6A führen zu currLim = 0. Käme ich bis auf 3A runter bräuchte ich auch keine Phasenabschaltung!. Mein ID3 lädt nur auf 2 Phasen ;)

miguelito62, 2024-01-15:
Wenn das ginge wäre es fast schade: Jetzt habe ich extra die Zuleitung zur WB unterbrochen um die Phasenabschaltung einzubauen.. ;)

abuesen, 2024-01-15:
Vor allem hat man ohne Umschaltung einen bequemen und geschmeidigen Übergang von 2kW-11kW :-)

hawa-lc4, 2024-01-15:
@abuesen : der Punkt geht an dich ;)
Ich werde das umgehend mal testen, auch mein Skoda citigo-e lädt nur mit zwei Phasen, mal sehen was der zu 3 Ampere meint.
Damit wird die ganze Sache noch unkomplizierter und entspricht meinem Motto: keep it simple.

Zitat Michael: "Btw. durch Einsatz shelly kannst das Problem auch nicht ausschließen, denn für den gilt das gleich wie für die wbec Platine / ESP.. (Ausfallraten komplexe Elektronik vs mechanischer Schalter)."
Damit hast du völlig Recht, auch der Shelly (oder Tasmota) ist eine MCU und die kann Mist machen.
Allerdings habe ich mit meinen einfachen Tasmotas (nur als reiner Schalter, ohne irgendwelche Sensoren dran) noch nie Probleme gehabt (ok, sag niemals nie) und damit habe ich den komplexen Teil wie du ihn weiter beschrieben hast realisiert.
Hier ist es eben eine korrekte http-Nachricht vom wbec an die Schaltbox, bei dir wäre es ein "Bitmuster" auf Hardwareebene.

Aber wenn's ohne Umschalter geht: noch viiel besser.
Wenn's wbec+evcc geht sollte ich das auch mit wbec+openWB hinbekommen, wäre doch gelacht. :)

miguelito62, 2024-01-15:
Btw gleiches auch mit der wbec webserver oberfläche: Stromvorgaben unter 6A gehen bei mir nicht einzustellen -  siehe Betreibsanleitung 1.6.3 Drehschalter S3 min. Ladestrom. Deswegen meine Überzeugung weniger geht nicht. Habt ihr ne andere HB Energy control Release? ;)

miguelito62, 2024-01-15:
Oder ich habe etwas verkonfiguriert und S3 zieht nicht in jeder Konfiguration...

miguelito62, 2024-01-15:
Ok ich habe 00.779.2885/04 von 21/2022. Punkt 1.5.3 Bedienungsanleitung für externes Lastmanagement sagt bzgl. des S3 das gleiche aus wie das Kapitel für lokales Lastmanagement - unter 6A geht nix.

abuesen, 2024-01-15:
Danke für die kritischen Nachfragen! 
Die WB geht nicht unter 6A, macht dann aber nur 3.2kW, gemessen mit einer geeichten Messung. Die WB "sagt", das seien 3,8kW.  Alles bei einer Spannung von 230V +-.

Bei den Tests habe ich immer nur geschaut ob es lädt oder nicht und alle Tests waren bei ausreichend Sonne, so dass die 3,2kW absolut plausibel und unter 6A waren! Später hab ich nur noch gemerkt, ob es nun geladen hat oder nicht. Ich fand die Welt in Ordnung.

Hier wird es dann spannend. Geht man mit PV-Überschuss "Start" (und "Stopp") unter die 6A, wird der Überschuss genutzt und der Rest aus dem Netz gezogen. Das ist mir somit bis heute nicht aufgefallen danke @miguelito62 .

Somit startet das Laden durchaus mit 1A auf drei Phasen, die restlichen 5A werden aber aus dem Netz dazu geholt. Fällt der PV-Überschuss unter "Stopp", so stoppt das Laden auch ordentlich. Eigentlich ein gutes Verhalten. Damit kann man mit wenig Aufwand und etwas "Zukauf" trotzdem die eigene PV-Energie nutzen.

Ich nehme das also zurück, Die Phasenumschaltung macht Sinn, wenn man mit der EC nur PV-Überschuss laden will. Un meine (absolut ernst gemeinte) Frage ist damit beantwortet.

Sorry. dass ich euch wuschig gemacht habe 😃 

[image]

[image]

miguelito62, 2024-01-15:
abuesen,

Im Gegenteil! Super, danke für's Nachsehen! Gemeinsam kommen wir der Sache schon auf den Grund. Wäre ja super wenn es doch klappen würde...

Ich habe auch gerade in wbec nachgesehen wg. Ladestromvorgabe:

#define CURR_ABS_MIN       60   // absolute possible lower limit for current

wbec prüft als untere Grenze für die Stromvorgabe auf 6A. Habe jetzt wg MQTT nicht explizit nachgesehen, aber das geht ganz sicher auch über die Funktion lm_storeRequest(), die dann das entspr. modbus reg. der WB schreibt.

Ich werde die Begrenzung für die Ladestromvorgabe mal testweise runtersetzen in wbec, wenn mal Zeit zum Spielen ist ;) Bin aber sicher, die WB selbst wird Werte unter 6A eben nicht annehmen bzw. es wird dann nicht geladen. Aber wie es so schön heißt: Versuch macht kluch.

Michael

steff393, 2024-01-15:
Hallo zusammen,
sorry für die späte Rückmeldung. 

Die Wallbox begrenzt ja immer den Strom nach oben hin. Das Auto entscheidet dann und kann natürlich auch weniger nehmen.
Man kann der Heidelberg (wie auch den meisten anderen Wallboxen) keinen kleineren Wert als 6A als "Ladestrombegrenzung" einstellen. Nichtsdestotrotz kann das Auto dann z.B. mit 4A laden, man hat es jedoch nicht unter Kontrolle.
Wenn man z.B. 3A per Modbus schickt, ignoriert die WB den Wert, oder geht auf 0A.

cfgPvLimStart/Stop:
Hier kann man niedrigere Werte einstellen. Dies führt dazu, dass die Überschussregelung noch nicht gleich komplett abschaltet, sondern auch noch weiterlädt, wenn z.B. nur noch Leistung für 3A zur Verfügung steht. Somit kann man quasi bei schlechterem Wetter weiterladen, aber halt mit entsprechend Netzbezug, der die Lücke ausgleicht.
Ob es viel Sinn macht ist fraglich, aber die Möglichkeit gäbe es.

Phasenumschaltung:
Ja, die Schaltung ist gut, hat aber eben die beschriebenen Risiken. Daher empfehle ich sie nicht, wenn man das Risiko nicht bewusst eingehen möchte. 
Zudem bin ich mittlerweile von der autom. Phasenumschaltung auch nicht mehr überzeugt. Es kann in manchen (ich denke seltenen) Konstellationen Sinn machen, aber man sollte sich mal Aufwand/Kosten zu Nutzen vorher gut anschauen. 
Jeder, der ausreichend Zeit hat, an der das Auto angesteckt bleiben kann, dem reicht auch (meistens) das 1-phasige Laden. 

Anders ausgedrückt: Ob ich jetzt bei einer typischen Gaußkurve der PV-Einstrahlung den oberen (durch 3 Phasen) oder den rechten Teil (durch längere Ladezeit) ins Auto lade ist oft relativ egal. Wenn man sich das selbst bastelt ist es egal und ein schönes Gimmick. Wenn man dafür aber 200...300 Euro mehr ausgibt, muss man schon überlegen, ob sich die Schalterei lohnt. Aber das ist eine Glaubensfrage, die ich eigentlich auch gar nicht mehr diskutieren mag ;-)

miguelito62, 2024-01-16:
Steff, danke für Deine insights! Ich bin ganz bei Dir, dass die ständige Schalterei der Phasen während des Ladevorganges nicht erforderlich ist.

Ich habe meine PV jetzt in der Winterzeit bekommen und meine ersten Schritte gemacht. An Sonnentagen kommen gut 3kW/h raus (habe 5kWp Ost + 5KWp west).  Bei tw Bewölkung dann "nur" 2kW/h oder weniger - da macht es Sinn 1phasig laden zu können.

Ganz ehrlich kann man meinen use case leicht mit 1 Phase abdecken. Habe Doppelgarage wo die Karre gemütlich dauerladen kann. Die restliche Energie kann von andere Verbrauchern HEMS gescheduled konsumiert werden.

Möchte aber trotzdem auch 2 phasig laden können: Wenn die Sonne im Zenith steht könnten es dann doch auch mal >> 5kW/h werden - Die Erfahrung wird es ab dem Frühjahr zeigen.

Ich denke dass bei mir, wenn überhaupt, an schönen Tagen 1x die Phasenumschaltung während des Ladevorgangens aktiv wird: Wenn man Morgens mit 1 Phase losfährt oder wenn man Abends mit 1 Phase den Ladevorgang abschließt. Bei Bewölkung würde ich nicht ständig hin und her schalten -> Die Erfahrungen mit der Energieausbeute im Sommer werde ich nun sammeln und dann auf dem FHEM ein modul umsetzen oder etwas fertigen entsprechend konfigurieren, wo die Hysterese entsprechend meiner örtlichen Gegebenheiten passt.

Denke, wenn man wie ich gerne mal bastelt macht die Phasenumschaltung ggF Sinn. Sonst eher nicht.

Michael

abuesen, 2024-01-16:
> Somit kann man quasi bei schlechterem Wetter weiterladen, aber halt mit entsprechend Netzbezug, der die Lücke ausgleicht.
> Ob es viel Sinn macht ist fraglich, aber die Möglichkeit gäbe es.

Das macht absolut Sinn!
Im Grunde regel ich im Winter genau damit. Hab 10kWp West, 5kWp Ost. Ab Mittag ist bei etwas Sonne der Hausakku voll und  es beginnt die Zeit, das Auto zu laden. (Ich _dachte_ ja nur, ich würde ausschließlich PV-Überschuss laden)

Würde ich gleich 11kW laden wäre häufig mein Bedarf gedeckt, bevor die Sonne weg ist. Folglich würde dann eingespeist. Man 
kann die Leistung zwar von Hand runter drehen, ist aber nur bei Sonne relevant und ich mag nichts manuelles. Also lädt das Auto PV mit Zusatzleistung aus dem Netz, falls verfügbar auch mehr als 80% Auto-SoC. Ohne PV stoppt auch das Laden In der Nacht wird der benötigte Rest mit 11kW nachgeladen.

**Voraussetzung**: Man verbraucht genug Strom mit dem Auto und die Schicht ist Mittags rum (wie bei uns), damit das Auto daheim laden kann :)

Die Vorteile sind klar: 
1. Es würde eh Strom dazu gekauft. Wenn ich tagsüber so die PV nutzen kann ist dies ein Vorteil.
2. Laden des PV-Ertrages, ohne eine Phasenumschaltung implementieren zu müssen. 
3. Vollautomatische Steuerung, Auto anstecken und gut.

Eine Phasenumschaltung finde ich als Nachrüstlösung viel zu aufwendig für den Nutzen, den ich damit erzielen kann. Zudem kann unser Auto das nicht.

PS OT: Ich lade eigentlich erst in der Nacht (ca 4 Uhr) aus dem Netz, weil dadurch meine ungeheizte Garage wärmer gehalten wird :-) Das merkt man stark.

miguelito62, 2024-01-17:
Echt ist die Abwärme des Laders spürbar?? Mir ist nur im Sommer 1x mal aufgefallen, dass der Lüfter beim Laden aktiv wurde - sonst nie. Wahrscheinlich fällt bei 11kW Laden doch einiges an Abwärme an. Bei mir ist die Garage im Haus integriert und wird dadurch mitgeheizt - d.h. im Winter so zwischen 10 und 5 Grad. Das ist natürlich super für den Akku.

Denke wir können hier abschließen:
- **HB Energy Control Ladestrom Minimum 6A**, darunter geht nix.
- Phasenumschaltung mE zwar nicht schwer zu implementieren: Man muss aber bistabiles Koppelrelais oder Schütz aufbauen, damit bei Absturz wbec Knoten während des Ladens sicher nix passieren kann. Ob es sich auszahlt, wenn man nicht gerne selber Bastelt: eher nein. Dann gleich ein Gerät kaufen, das das bereits unterstützt.

Danke an alle für Eure Inputs und Mithilfe

miguelito62, 2024-04-03:
Hallo noch einen weiteren Kommentar zur Phasenumschaltung von mir:

Es ist wirklich so, dass der ESP8266 sehr touchy ist, wenn man an manchen Pins (z.B. D3) beim Hochlauf etwas angeschlossen hat. Meiner bleibt gefühlt jedes 3. hängen. Viele Pins hat mein Derivat nicht, so bleiben nicht viele Optionen. Die Modbus Schnittstelle zu verlegen und dann wieder zu riskieren, dass das Teil wegen des angehängten PHYs hängen bleibt, die Übung spare ich mir.
Deswegen folgende Lösung als Nächstes: Das Umschaltrelais für die Phasenumschaltung einfach auch per Modbus ansteuern. Wollte schon eine Lösung mit Arduino + Modbustreiber machen aber auf Amazon gibt es fertige Modbus Relais um 10.- zahlt sich also nicht aus. (Jaja auf Aliexpress gibts die ab 3EUR - die kommen dann wahrscheinlich zu Weihnachten an). Einzig die Größe könnte herausfordernd werden, will das ja in meine bestehende Dose rein - siehe Bilder oben.
Die Teile sind auch konfigurierbar in Punkto Baudrate, soll ja an den Bus zur Wallbox dazugehängt werden. Ich berichte jedenfalls kurz, wie es geklappt hat. Ist ja vielleicht eine Option für den einen oder anderen.

Michael

abuesen, 2024-04-03:
Da mache ich beim Kommentar doch mit den Winter-Erfahrungen mal mit. Aber ohne Phasenumschaltung (die unser Kona eh nicht beherrscht).

Nach den o.g. Erkenntnissen wollte ich natürlich auch nicht den guten Sonnenstrom verschenken oder immer von Hand das Laden ein- und ausschalten. Zum Regeln nutze ich seit Herbst evcc [1]. Es lässt die Angabe von Thresholds zu, damit das Laden schon vor dem notwendigen Überschuss anfängt und bei zu viel Bezug wieder aufhört. Gelernt über github [2].

Grundüberlegung: Laden muss ich eh, also lade ich, wenn etwas Überschuss da ist und kaufe damit weniger Energie beim Versorger.

Nach etwas probieren lief diesen Winter die Konfiguration sehr gut: 
Ab 2kW Einspeisung (für eine Minute) startet der Ladevorgang. Der fehlende Strom wird aus dem Netz geholt. 
Ab 3kW Bezug für mehr als 4 Minuten wird das Laden wieder eingestellt.

Damit habe ich fast meinen gesamten Ertrag selbst mit Haus(-akku) und Auto verbraucht. Trotzdem musste ich noch oft genug zusätzlich Laden, was aber nachts die Garage etwas temperiert. Mit dem Vorgehen ist für mich eine Phasenumschaltung so gar nicht erforderlich. Das liegt aber sicherlich an meinen Gegebenheiten, also Anlagengröße und Bedarf.

BTW: Trotz der Einstellung '6 A' auf drei Phasen wird nur mit 3,8kW geladen. 

[1] https://evcc.io/
[2] https://github.com/evcc-io/evcc/discussions/11756#discussioncomment-8356523