steff393/wbec #9: Improve docu on MQTT and logging

oldgitdaddy, 2021-08-15:
Hi all,
Would it be possible to spend more words on the mqtt feature? and also send some logging over the mqtt channel, so that we have more options for debugging? Documentation should cover an example of the config and also the particular mqtt topics wbec publishes. I understand that the mqtt feature is preliminarly developed to support openwb, but I also think that it can be of general use for integration into other systems.

Thank you.

steff393, 2021-08-15:
Hi, 
I tried to improve the the wiki in this regard: https://github.com/steff393/wbec/wiki/openWB-configuration-via-MQTT
Yes, main intention is for openWB. But I know some users, which use it also with other brokers.

Regarding logging: Do you have specific infos (e.g. from the /json in mind)?

oldgitdaddy, 2021-08-17:
Regarding MQTT enablement, I see that MQTT only works, if `cfgMqttLp `is set. Correct? see also: https://github.com/steff393/wbec/blob/efd62f7e3569f6cb03daaac0fbf63935d08b1375/src/mqtt.cpp#L105

What would be the best config, for MQTT communication with EVCC?

In case of logging, I would log all debug info via MQTT. This would may require some extra config parameter.

steff393, 2021-08-18:
Yes, that's right. Because the openWB MQTT topic contains the number of the loadpoint. And cfgMqttLp describes the mapping between the loadpoint nr. and the wbec internal box id. In the simplest case it's just a [1].

MQTT communication to EVCC is a good idea (because it would allow to get rid of the go-e API). But how does it work? As far as I understood up to now, there is no MQTT broker contained in EVCC, right? I only found the following [link](https://github.com/andig/evcc-config#generic-mqtt-1).

Logging the debug messages via MQTT is an excellent idea! I've to think how to implement it. 
I just figured out, that the tool which I typically use (MQTT Explorer) also has a "History" section that can be used for this purpose.

oldgitdaddy, 2021-08-19:
Basically, it would be good, that the IP address of the broker is configurable. In some setups, there is a central MQTT broker in the network. For Home Assistant there is an Add-on for mosquitto broker, which is also the preferred choice for evcc.

In general, I would refrain from incompatible changes regarding the API compatibility.

steff393, 2021-08-29:
Logging via MQTT works now. The debug messages are published to "wbec/log":
[image]

oldgitdaddy, 2021-09-30:
Will you create a new release, so that we can consume it over the air? Highly appreciated 🥇

steff393, 2021-10-01:
Just sent the binary to you by mail. I think it's not yet enough function increase for a new release.
In case someone else is interested, then please let me know.

Unfortunately I didn't manage yet to work on the EVCC/MQTT topic...

steff393, 2021-11-02:
Both topics are included in wbec 0.4.2

closed 2021-11-02 by steff393