steff393/wbec #14: Problems using ioBroker/go-e adapter

moddmylife, 2021-12-12:
hello anyone, 
i tried using the Go-E-Charger adapter to include the WBEC into my HA-System. But everytime i got an API-Error
[image]

any ideas to solve this?

steff393, 2021-12-12:
Hi 
sorry, wbec emulates only a part of the go-E API (i.e. the main variables needed by Wallbox Steuerung App, openWB and EVCC). 
The API seems to increase and increase every few weeks, so it's not possible to remain always up to date. And it doesn't make sense to send dozens of values just with constant default values (from my point of view).

So I'd suggest to solve this on the ioBroker side. Maybe the checks of the unneeded variables can be disabled?

moddmylife, 2021-12-12:
> Hi sorry, wbec emulates only a part of the go-E API (i.e. the main variables needed by Wallbox Steuerung App, openWB and EVCC). The API seems to increase and increase every few weeks, so it's not [...]

No problem. i'll do a little bit of json-magic and see what happen. 
but thank you for your quick reply!

closed 2021-12-19 by steff393

maxpd1, 2022-04-22:
Is there an official manual how to do the json magic?