steff393/wbec #16: Port ändern MQTT

anselra, 2022-01-03:
Hallo zusammen,
erstmal ein Gutes Neues und Glückwunsch zu diesem tollen Projekt.
Ich binde WBEC mit Hilfe von MQTT an meinen IOBroker an. Leider habe ich keine Möglichkeit gefunden den Port anzupassen. Dies ist bei mir aber leider notwendig. Kann mir da jemand weiterhelfen, vielen Dank

steff393, 2022-01-04:
Hallo anselra,
dankeschön, wünsch ich dir ebenso. Du meinst hier statt 1883 einen parametrierbaren Wert, oder?
client.setServer(cfgMqttIp, 1883);

Ja, das könnte ich zur nächsten Version umsetzen.

anselra, 2022-01-04:
Ja genau, ein parametrierbarer Wert wäre am besten.
"cfgMqttIp":"192.168.178.1:xxxx"

klasse Vielen dank

steff393, 2022-01-04:
Wird in der nächsten Version drin sein als neuer Parameter, den man z.B. so setzen kann: "cfgMqttPort":42
Default ist 1883.
Den Code habe ich vorher schon hochgeladen.

anselra, 2022-01-04:
super, danke

closed 2022-01-04 by anselra