steff393/wbec #21: Ethernet Anbindung Wbec

taschaue, 2022-02-20:
Guten Tag, 
hat jemand eine Anbindung per Ethernet statt Wifi realisiert? Wird das unterstützt?

Danke
taschaue

steff393, 2022-02-20:
Hallo taschaue,
mit dem verwendeten Mikrocontroller geht es meines Wissens nicht. 
Mögliche Alternative, wenn schon ein Netzwerkkabel bei der Wallbox liegt: Das Netzwerkkabel für den Modbus nutzen und wbec am anderen Ende (ist ja in der Regel neben dem WLAN-Router) platzieren.
Ist dann fast wie mit Ethernet und hat den Vorteil, dass die Router (z.B. Fritzbox) oft auch gleich einen USB-Anschluss zur Stromversorgung haben, d.h. man spart sich das USB-Netzteil.

taschaue, 2022-02-20:
Danke für die Info.
Gab es mal einen Test mit einem ESP32? Weil da gibt es z.B. von Olimex den ESP32-PoE?

steff393, 2022-02-20:
Bisher noch nicht. Soweit ich weiß, müsste man für den Umstieg vom ESP8266 auf den ESP32 einiges an den Libraries ändern/austauschen (z.B. SPIFFS/LitteFS, etc.). Hab ich bisher nicht in Angriff genommmen.

taschaue, 2022-02-20:
Danke für die Info. Ist übrigens ein echt tolles Projekt!

closed 2022-02-20 by taschaue