steff393/wbec #22: Probleme mit EVCC und WBEC

oldgitdaddy, 2022-02-23:
Hallo zusammen,
Mein aktuelles setup besteht aus EVCC, was auf Home Assistant läuft und per Wifi angebundenen WBEC Installation. Im letzten Jahr hat das setup bereits etwas rumgezickt. Dies ist jetzt mein nächster Versuch, das gut ans laufen zubekommen. Generell funktioniert das setup, leider verliert es EVCC zum WBEC immer die Verbindung. Ich habe das Problem sowohl wenn EVCC per REST mit dem WBEC oder per MQTT kommunziert. EVCC stoppt das Laden, bzw. führt es nicht weiter eher zufällig. Sobald ich aber das Auto wieder neu an die Heidelberg Connect anhänge funktioniert es dann wieder für eine Weile. 

Gerne bin ich bereit jeweilige Logs zur Verfügung zu stellen, aber über MQTT sehe ich keine nennenswerten Fehler. Danke für Eure Hilfe!

steff393, 2022-02-23:
Hi,
bei der Variante über MQTT gibt es eigentlich keine direkte Verbindung von EVCC und wbec, es geht ja alles über den Broker.
Und es gibt nur 4 relevante Werte:
topic: wbec/lp/1/status   --> von wbec an EVCC: gesteckt, ladend, usw.
topic: wbec/lp/1/enabled --> von wbec an EVCC: 0, wenn 0A eingestellt, sonst 1
topic: wbec/lp/1/enable  --> Kommando von EVCC, d.h. AUS (0A) oder AN (6-16A, s.u.)
topic: wbec/lp/1/maxcurrent --> Stromvorgabe von EVCC (6-16A)

Die anderen Werte (Meter) sollten nicht so wichtig sein, schau aber vielleicht der Vollständigkeit halber auf welchen Werten die stehen.
Wenn man jetzt z.B. den echten Zustand am Fahrzeug, den Wert im Broker und den Wert in EVCC anschaut, dann würde man doch sehen, was nicht stimmt.
Vielleicht wäre noch hilfreich zu wissen, ob die blaue LED bei wbec noch alle 10s (cfgMbCycleTime) kurz flackert, das zeigt eine aktive Kommunikation am Modbus.

Nutzt du den Standby (cfgStandby)?
Passiert das im Zustand "PV-Laden" oder auch bei "Min+PV"? Ich glaub manche Autos setzen das Laden nicht automatisch fort, wenn zwischendurch mal der Strom auf 0A begrenzt wird.

oldgitdaddy, 2022-02-24:
Das Prinzip von MQTT ist mir bewusst. Und natürlich hast Du recht, es ist ein broker dazwischen. 

Das ist die Konfiguration:
`"cfgStandby":4`

Macht es Sinn, diesen fix mal einzuspielen? Wann gibt es den release mit dem pull request? https://github.com/steff393/wbec/pull/19

steff393, 2022-02-24:
Ich werd demnächst ein Release 0.4.4 erstellen, in dem die Änderungen aus #19 enthalten sind. In den aktuellen master-Branch sind die Änderungen auch schon eingecheckt.

oldgitdaddy, 2022-02-25:
Super danke. Nach dem neuen update werde ich dann nochmal schauen. Akku ist aktuell sowieso voll :-)

oldgitdaddy, 2022-03-01:
Also mit release 0.4.4 läuft wbec deutlich stabiler. Die erste Session hat bereits gut funktioniert. Nutze folgenden Config Parameter
"cfgLoopDelay": 5

Bei der Fülle an Parametern, denke ich wäre es aber gut eine separate Doku und Erklärung noch bereitzustellen, für die jenigen, die die "Historie" jedes einzelnen Parameter nicht kennen. Ich weiss, dass die Parameter im Code bereits etwas beschrieben sind.

closed 2022-03-01 by oldgitdaddy