steff393/wbec #24: Support Fronius Inverter / Smart Meter

andy5macht, 2022-03-15:
Da die Firma Fronius den Modbus auch über TCP verbreitet und auch die Daten dazu veröffentlich hat, möchte ich die Fronius WR samt Smart Meter mit in dieses Projekt einbinden.

**Das ganze hat das Ziel die Wallbox automatisch auf die Ladeleistung zu setzen, die dem PV überschuss entspricht.**

Es gibt bereits hilfreiche Informationen zum Thema Fronius Modbus TCP und dessen Register.
https://github.com/Brovning/fronius/blob/main/README.md

steff393, 2022-04-29:
Hi @andy5macht,
hab gerade deinen Code aus dem Fork rübergezogen. 
Von einem anderen Nutzer hab ich das hier bekommen, bzgl. Kostal KSEM:
https://www.kostal-solar-electric.com/en-gb/products/accessories/smart-energy-meter/-/media/document-library-folder---kse/2020/12/15/14/22/ba_kostal_interface_ksem_de.pdf

Das hab ich jetzt noch angefügt.

Die nötigen Parameter wären jetzt:
- cfgInverterIp: IP-Adresse vom Wechselrichter oder Zähler (ersetzt das alte cfgSolarEdgeIp)
- cfgInverterType: 0=off, 1=SolarEdge, 2=Fronius, 3=Kostal

Aktuell ist aber alles noch ungetestet.

anselra, 2022-05-01:
Hi @steff393 ich hänge mich mal dran, habe ebenfalls einen Fronius WR und den Fronius SmartMeter dazu. 
leider bekomme ich die Werte nicht angezeigt. Verstehe ich es richtig, dass ich nur die cfg.json anpassen muss?

{"cfgApSsid":"wbec","cfgApPass":"wbec1234","cfgCntWb":1,"cfgMqttIp":"192.168.0.xxx","cfgMqttPort":1884,"cfgMqttLp":[3,1],"cfgMqttUser":"xxx","cfgMqttPass":"xxx","cfgInverterIp":"192.168.0.xxx","cfgInverterType":"2"}

würde es gern weiter testen und gerne auch dokumentieren,
Gruß

steff393, 2022-05-01:
Hi Ralf,
danke für die Unterstützung!
"cfgInverterType":2
d.h. die Anführungszeichen müssten noch weg. Danach bitte wbec resetten (/reset).
Wichtig ist auch, noch zu beachten, was per MQTT parallel an Anforderungen kommt. Nicht dass sich da Änderungen überlagern.
Bin schon gespannt ;-)

steff393, 2022-05-01:
Für's Debuggen eignen sich wahrscheinlich die Seiten /inverter (Rohwerte) und /pfox (--> watt = Bezug/Einspeisung)

anselra, 2022-05-02:
Danke für den Hinweis :-)
unter /inverter bekomme ich diese Ausgabe, nach ca 30 sec nach reset allerdings immer die gleichen Werte
{"inverter":{"isConnected":"1"},"power":{"AC_Total":"0","house":"0","inverter":"65","inverter_scale":"18034","meter":"26997","meter_scale":"29440"}}
allerdings verändert sich diese nicht.

unter /pfox
{"box":{"chgStat":0,"power":0,"currLim":0,"resCode":"e4"},"modbus":{"millis":191811},"pv":{"mode":0,"watt":0}}
hier ändern sich die millis

über MQTT kommen bei mir keine zusätzliche Werte an.
Testaufbau ist ein NODEMCU, dieser ist nicht verbunden mit der Wallbox, da ich davon ausgehe, dass dies für PV nicht notwendig ist, falls doch auch kein Problem :-)

andy5macht, 2022-05-02:
Cool, dass es hier weiter geht mit Fronius.
Ihr seid mir zuvor gekommen! Die Adressen für Fronius waren bisher nur als Dummies ausgeführt:
```
#define REG_FR_M_AC_Power       40006 // modbus register for "Total Real Power (sum of active phases)" int16 in Watts
#define REG_FR_M_AC_Power_SF    40007 // modbus register for "AC Real Power Scale Factor" int16 SF
```

diese beiden Zeilen müssen also angepasst werden. Hierfür wäre mein erster Versuch:
```
#define REG_FR_M_AC_Power       40084
#define REG_FR_M_AC_Power_SF    40085
```

Damit das auch funktioniert muss der Fronius Wechselrichter seine Daten auch im Integer-Format ausgeben.
Die ganzen Informationen stehen aber oben in meinem Link zur PDF

anselra, 2022-05-03:
ich benutze noch den IoBroker mit der Fronius App, da ist mir aufgefallen, dass man (zumindest bei der App) nur positive Werte bekommt und nicht einen positiven oder negativen auf einer API wie z.B. bei Kostal wenn ich mich recht entsinne

andy5macht, 2022-08-27:
Hallo,
ich konnte endlich meinen Fronius Wechselrichter in Betrieb nehmen. Die Register dazu sind:
#define REG_FR_I_AC_Power       40083 // modbus register for "AC Power value", int16 in Watts 
#define REG_FR_I_AC_Power_SF    40084 // modbus register for "AC Power scale factor" int16
#define REG_FR_M_AC_Power       40087 // modbus register for "Total Real Power (sum of active phases)" int16 in Watts
#define REG_FR_M_AC_Power_SF    40091 // modbus register for "AC Real Power Scale Factor" int16 SF

Die Einstellungen am Wechselrichter müssen dann folgende sein:
[image]

Ich verstehe allerdings nicht, wieso der Hausverbrauch falsch berechnet wird?
Das ist laut source eine einfache Subtraktion...
[image]

steff393, 2022-09-03:
Hallo Andy,
super, danke für die Anleitung. Ich hab's gerade so angepasst (die Invertierung hab ich ein Stück weiter oben gemacht, damit die anderen Wechselrichter nicht beeinflusst werden). Wäre super, wenn du's damit nochmal testen könntest!

steff393, 2022-09-03:
@andy5macht @anselra @AMiketta
Ich habe jetzt mal versucht, die PV-Funktion übersichtlicher zu beschreiben: https://github.com/steff393/wbec/wiki/PV-Überschussladen

Da ich selber keinen der genannten Wechselrichter habe, wäre interessant, welche Typen ihr habt und welche Einstellungen man ggf. dort vornehmen muss (ähnlich wie Andy es oben für den Fronius gemacht hat). Wäre super, wenn ihr das im Wiki ändern könntet, oder hier im Issue beschreiben könnt (dann würd ich es übertragen).
Ich glaube das wäre für andere Nutzer ganz hilfreich.

anselra, 2022-09-04:
@andy5macht @steff393 
ich habe es gerade getestet allerdings passt es noch nicht ganz.
Hardware, Fronius Symo 10.0-3-M und Fronius Smart Meter, 
Wechselrichter Einstellungen wie andy beschrieben, die API passen auch. Ich bekomme folgende Werte ip/inverter

{"inverter":{"isConnected":"1"},"power":{"AC_Total":"0","house":"11515","inverter":"6023","inverter_scale":"0","meter":"-5492","meter_scale":"0"}}

house wird addiert, da der Wert negativ übermittelt wird. Daher denke ich dass die Invertierung nicht richtig läuft

Gruß Ralf

steff393, 2022-09-04:
Du hast den aktuellen Stand von gestern abend verwendet, oder?
Die Bedeutung von "meter" bei /inverter ist für alle Wechselrichter gleich: Pos. = Einspeisung, Neg. = Bezug
Fronius liefert es zwar erst anders rum, aber unmittelbar nach dem Einlesen wird es ja sofort negiert (Zeile 92 in inverter.cpp).

PV-Leistung: 6023W
Zählerwert: -5492W (negativ, also Bezug aus dem Netz)
=> Damit wäre house = 11515W

PV-Leistung: 468W
Zählerwert: 159W (positiv, also Einspeisung ins Netz)
=> Damit wäre house = 309W

Insofern wären die Werte erst mal für mich stimmig. Wenn das aber nicht zu der realen Situation bei dir passt, kann es vielleicht sein, dass eure Smart Meters sich da unterschiedlich verhalten?

andy5macht, 2022-09-04:
Habe es ebenfalls getestet.
Der Betrag passt. Allerdings stimmt das Vorzeichen nicht.
Bezug(+)/Einsp.(-) -25 W => sollte ohne "-" sein.

anselra, 2022-09-04:
> Du hast den aktuellen Stand von gestern abend verwendet, oder? Die Bedeutung von "meter" bei /inverter ist für alle Wechselrichter gleich: Pos. = Einspeisung, Neg. = Bezug Fronius liefert es zwar [...]

Ja habe den aktuellen stand genommen.

PV-Leistung: 6023W Zählerwert: -5492W (negativ, also Bezug aus dem Netz) => Damit wäre house = 11515W
hier wäre richtig gewesen 6023W Erzeugung - 5492W Netzeinspeisung = 531 W Verbrauch 

Aktuell wird bei mir über /inverter 
{"inverter":{"isConnected":"1"},"power":{"AC_Total":"0","house":"64870","inverter":"0","inverter_scale":"-2","meter":"6663","meter_scale":"-1"}}

Aktuell ist es über fronius:
0W Erzeugung - 688W Netzbezug = 688 W Verbrauch

andy5macht, 2022-09-06:
Ich habe nochmal nachgeforscht, warum die invertierung nicht funktioniert.
Die Invertierung funktioniert, wenn diese nach dem Befehl: `mb.task();  // Common local Modbus task` ausgeführt wird.
Vermutlich hat das Damit zu tun, dass die Variable im Modbus als **u**int16 verwendet wird.

steff393, 2022-09-19:
Hallo Ralf,
hat zwar mit dem Thema nichts zu tun, aber wenn du nur eine Wallbox hat, dann sollte cfgMqttLp nicht [3,1] sein, sondern nur [3]. Sorry, das Beispiel im Wiki war missverständlich dazu, werd ich ändern.

andy5macht, 2022-09-20:
Hallo Ralf,
die Config hat einen weiteren Parameter bekommen. Der aber nicht zwangsweise verwendet werden muss und außerdem nichts mit dem Wechselrichter zu tun hat. Ich verstehe aktuell noch nicht, warum der WBEC sich nicht mit deinem Wechselrichter verbinden soll...ich werde aber deine Config heute abend bei mir testen, dann sehen wir weiter.

andy5macht, 2022-09-21:
@anselra
von welcher Version bist du denn gekommen?
Mit deiner Config geht es wirklich nicht - was aber logisch ist... ich habe `"cfgPvActive":1` zu deiner Config hinzugefügt und dann verbindet er auch mit dem Fronius.

anselra, 2022-09-22:
@andy5macht 
danke, bei mir läuft es mit deiner conf auch. 
Sobald ich aber Mqtt dazu nehme geht es nicht mehr. Ich probiere noch rum, da ich dies ja schonmal am laufen hatte, wie weiter oben beschrieben mit dem Berechnungsproblem und dies ohne den Eintrag "cfgPvActive":1 . Wenn ich es im Wiki richtig verstehe hat dieser wert auch nur Auswirkungen auf die Funktion Überschussladen und nicht ob Werte vom WR geholt werden

steff393, 2023-05-25:
Hi zusammen, weiß jemand, wie man die Modbus-Adresse vom Smart-Meter rausfinden kann, wenn sie nicht 240 ist?
Neuere Wechselrichter (GEN24) haben wohl 200.
Kann man sich die im Wechselrichter anzeigen lassen?

anselra, 2023-05-28:
@steff393 ich habe mir alle Einträge auf dem Interface angeschaut, habe unter dem User Service nicht gefunden

steff393, 2023-09-14:
Danke trotzdem!
Hat jemand der Fronius-Nutzer schon mal beobachtet, dass die Werte nicht zur Anzeige in der Fronius App passen?

anselra, 2023-09-26:
> Danke trotzdem! Hat jemand der Fronius-Nutzer schon mal bemerkt, dass die Werte nicht zur Anzeige in der Fronius App passen?

@steff393 ich hatte es am Anfang mal verfolgt und festgestellt, dass ein gewisser zeitlicher Versatz gibt zwischen der WR Anzeige und der Fronius APP. Die Werte die über die IP des WR kommen passen bei mir zu der Anzeige am WR, hoffe das hilft.

closed 2024-03-24 by steff393