steff393/wbec #26: Wattangabe nicht stimmig?

frad1980, 2022-03-22:
Hallo, wenn ich die Wattwerte von meinem Smart Energy meter kontrollieren, dann sehe ich eine Diskrepanz. In wbec bekomme ich mehr Watt (Leistung) angezeigt, als mein Energymeter aktuell verbraucht...?

[image]

Muss da noch etwas angepasst werden an der Config?

steff393, 2022-03-22:
Gute Frage!
Das W in wbec ist eigentlich nicht richtig. Der interne Werte in der Heidelberg wird in VA angeben. Diesen übernehme ich 1:1.
Könnte ein Effekt von Wirkleistung & Blindleistung sein. Aber da kenne ich mich zu wenig aus und ich hätte vermutet, dass die Abweichung deutlich geringer sein müsste.
Grundsätzlich wären 5,5kW bei 8A schon plausibel (entspricht 11kW/16A), wenn das Auto den max. Strom zieht.

DrGalvany, 2022-03-27:
Hallo,
bei meinem Fahrzeug stimmt die Anzeige von wbec.
Meines lädt aber nur einphasig.
Welches Fahrzeug hast du daran?
Lädt es ein, zwei oder dreiphasig?
Schaue auch mal direkt im KSEM nach, da die Anzeige des Penticors evtl. hier abweicht.

steff393, 2022-04-17:
Scheint sich erübrigt zu haben

closed 2022-04-17 by steff393