steff393/wbec #27: ESP innerhalb der Box auf Heidelberg auflegen

maxpd1, 2022-03-22:
Hi,

überlege gerade wie ich die Box am besten anbinde.
Es existiert noch kein Switch und dieser wäre auch vermutlich 20 Meter von der Box entfernt.

An der Box selber liegt (noch) kein separater Strom. 

Manche haben das WLAN Modul extern positioniert. 

Spricht was gegen die Verwendung einer Phase mit Wago Klemmen auf diesen Converter und parallel weiter in die Heidelberg?

[image]

und von dem Converter an den Wemos?

In der Box selber ist für all dies genug Platz und liese sich auch dort in einer zusätzlichen Box verstauen. Oder rauchen die Converter gerne ab und zerstören mir die gesamte Heidelberg?

Im Sommer, der prallen Sonne hinter der Aluplatte, wirds nicht zu heiß in der Box für Wemos und Co?

Wemos D1 Mini kompatibel oder muss es eine NodeMCU sein?

Wo bekomme ich diesen Aufsatz her?

[image]

Gruß
maxpd

steff393, 2022-03-22:
Hallo maxpd, schick mir am besten kurz eine Mail an wbec393@gmail.com wg. dem Aufsatz

steff393, 2022-03-22:
Hallo maxpd,
technisch sollte das machbar sein. Aber auf die Risiken hast du ja hingewiesen, sehe ich ähnlich. 
Ein weiteres Gegenargument ist, dass diese Schichten (Gehäuse, Heidelberg-Kunststoff, Heidelberg-Blech, Ladekabel) schon einen Einfluss auf die WLAN-Qualität haben werden. Wie viel lässt sich schwer abschätzen.
Daher würde ich (falls möglich) eher ein längeres Modbus-Kabel verlegen.

maxpd1, 2022-03-22:
Du meinst damit Kabel aus deinem wbec Upgrade in den Heidelberg Input? 
Du sprichst bei Modbus-Kabel dabei aber nicht von einem Spezialkabel, sondern von einem Stück Jumperkabel, Netzwerkkabel oder auch simples Stromkabel?

steff393, 2022-03-22:
Heidelberg empfiehlt in der Anleitung glaub ich ein CAT-Kabel (Netzwerkkabel). Alternativ sollte auch J-Y(St)Y z.B. 2x2x0,8 gehen. Verdrillt und geschirmt sollte es sein. Ich würde sagen es kommt auf die Länge an. Je kürzer desto weniger kritisch ist das Kabel. Theoretisch sollte Modbus ja bis 700m gehen. Bei <1m geht bestimmt auch ein Klingeldraht o.ä. (ist ja dann notfalls auch schnell ausgetauscht)

andy5macht, 2022-03-22:
Wemos D1 Mini ist kompatibel. Habe ich im Einsatz. Pins sind 1:1 verwendbar.

HoSt81, 2022-03-23:
Ich habe die 3,3V direkt von der oberen Platine abgegriffen. Den Wemos aber extern in einer Abzweigdose montiert. Der Anschluss erfolgt dabei über die beiden linken Busklemmen, welche dann natürlich (an der Durchkontaktierung) aufgetrennt wurden
[image]
[image]
.
Von der Stomaufnahme sollte das auf dem 3V3 Rail her noch passen - Trotzdem auf eigene Gefahr!

maxpd1, 2022-03-23:
> Wemos D1 Mini ist kompatibel. Habe ich im Einsatz. Pins sind 1:1 verwendbar.

danke für deine Bestätigung. Mich wundert es nur, denn die GPIO Bezeichnungen weichen ja schon stark ab:

[image]

> Ich habe die 3,3V direkt von der oberen Platine abgegriffen. Den Wemos aber extern in einer Abzweigdose montiert. Der Anschluss erfolgt dabei über die beiden linken Busklemmen, welche dann natürlich [...]

Danke für deinen Hinweis!
Naja, löten beim 700€ Objekt will ich vermeiden ^^
Warum hast du dennoch ausgelagert?
Verstehe ich richtig, dass du unten links die Pins (Output) jetzt als Strom nimmst?

HoSt81, 2022-03-23:
> Danke für deinen Hinweis!
> Naja, löten beim 700€ Objekt will ich vermeiden ^^
> Warum hast du dennoch ausgelagert?
> Verstehe ich richtig, dass du unten links die Pins (Output) jetzt als Strom nimmst?

Ich wollte halt ein zusätzliches NT / die einhergehenden Verluste vermeiden.
Ausgelagert habe ich wegen des mäßigen Empfangs in der Garage.
Ja, die pins links sind jetzt 3V3 und GND. 
Die Anschlüsse sind ja standardmäßig direkt miteinander Verbunden (IN-OUT).  Wo man den Bus anschließt, ist eigentlich egal...

andy5macht, 2022-03-23:
> 
Ich habe die Pins mittels pinouts verglichen, die Pins zum ESP8266 sind identisch und auch die Bezeichnung D0-D8 ist identisch und wenn du es immernoch nicht glaubst...es funktioniert sogar ;)
[image]

maxpd1, 2022-03-23:
Ok wenn nur gpio 0 2 4 und 5 betrachtet werden. Der gesamte Rest weicht ab.
Eine direkte Verbindung von gpio zu Heidelberg ist nicht möglich?

andy5macht, 2022-03-23:
Nicht ganz, diese pins werden verwendet und sind alle identisch mit dem Wemos D1 mini
#define PIN_DI              5   // GPIO5, NodeMCU pin D1
#define PIN_RO              2   // GPIO2, NodeMCU pin D4
#define PIN_DE_RE           4   // GPIO4, NodeMCU pin D2
#define PIN_RST             0   // GPIO0, NodeMCU pin D3 
#define PIN_SS             15   // GPIO15,NodeMCU pin D8

Direkte verbindung ist nicht möglich, da der RS485 Bus Differenziell ist. Das kann mittels GPIO pins nicht realisiert werden.

steff393, 2022-03-23:
@HoSt81: Raffinierte Lösung, vor allem der Ansatz mit den Klemmen! 👍 
Hatte da auch schon gesucht, aber nichts gefunden. Es geht also doch. Aber mir wär das Risiko zu groß gewesen ;-)

closed 2022-03-23 by steff393

cola32, 2023-01-22:
kann mir jemand erklären welche Pins eines esp8266 wo an der Heidelberg angeschlossen werden müssen?

steff393, 2023-01-22:
Es gibt wbec jetzt als fertige Module mit Anleitung: https://steff393.github.io/wbec-site/

cola32, 2023-01-22:
Finde ich auch sehr interessant!
Was soll das Ganze kosten?

Ich würde allerdings einen esp8266 mit RS485 Modul bevorzugen, da ich an der Box bereits 5V anliegen habe.
Möchte mir ungern noch ein 12V Netzteil einbauen.
Ja, und ein paar esp8266 habe ich auch noch rumliegen.

steff393, 2023-01-22:
Schick mir am besten eine Mail an wbec393@gmail.com

weissjuergen, 2023-08-19:
> Hi,
>
> überlege gerade wie ich die Box am besten anbinde. Es existiert noch kein Switch und dieser wäre auch vermutlich 20 Meter von der Box entfernt.
>
> An der Box selber liegt (noch) kein separater [...]

Hast du jetzt rausgefunden woher du den Aufsatz bekommst ?

Mich würde das auch interessieren, ebenso wie die Modbus Leitungen von der Heidelberg daran angeschlossen werden.

Gruß

steff393, 2023-08-19:
s.o., schick mir am besten eine Mail an o.g. Adresse. 
Die Aufsteckmodule gibt es nicht mehr, aber fertige Geräte (inkl. Anleitung): https://steff393.github.io/wbec-site/