steff393/wbec #34: Ethernet Version

moddmylife, 2022-05-07:
Good morning.
First, thank you for brilliant work!
I have some issues with my wireless network. If it fails, wbec stopped working and the Heidelberg energy control goes into bus error. 
Nothing that depends on wbec, only on my wireless network and my house made of steel (I think)...

Is it possible to include Ethernet modules like the W5500 or ENC28J60 to control everything over wired network? Maybe PoE is also possible.

steff393, 2022-05-07:
Hi,
that's a typical question. Many people choose the following alternative:
[image]

Of course it's no 'real' Ethernet, but I'd say it has no significant disadvantages compared to Ethernet.

moddmylife, 2022-05-10:
Good morning and thanks for quick response!
Thats a good point, but in my case I have Ethernet Distribution next to the UV where the Cable from the Wallbox came in. Best way to connect is in my case to put the ESP inside the Wallbox itself.

DrGalvany, 2022-05-15:
Hello steff393,
have just seen the picture of the connection of the wallbox here, and I shocked.
Have in memory to have connected my cables on the right side.
So, immediately checked again and I was relieved.
The picture is visually the other way around, but correctly described.

Gruß
Micha

steff393, 2022-05-15:
Yes, you're right IN and OUT are mirrored on the picture, sorry.
But it shouldn't matter, as IN A and OUT A are directly connected with each other on the PCB (same for B) :-)

closed 2022-05-18 by steff393