steff393/wbec #36: Wallbox läuft nicht mehr

elektronikfreak, 2022-05-18:
Hallo, seit heute Morgen hat meine Wallbox folgendes Fehlerbild:

Sechsmaliges Blinken weiß, Pause, dreimaliges Blin‐
ken blau (90 % an, 10 % aus), Pause
Mögliche Störungsursache: Über- oder Unterspan‐
nung der Versorgungsspannung.
Beim Betrieb im Lastmanagement bedeutet diese
Blinksequenz, dass ein Kommunikationsfehler zwi‐
schen externer Steuerung und der Wallbox oder zwi‐
schen Leader-Wallbox und der Wallbox besteht.
● Bei Über- oder Unterspannung Sie müssen nicht
eingreifen.
● Bei Kommunikationsfehler muss der Monteur die
korrekte Ausführung der Kommunikationsleitung
überprüfen.
Nach einem Selbsttest und behobener Störung leuch‐
tet die Frontbeleuchtung weiß

Wenn ich den Schieberegeler vom WEBC nach rechts verschiebe springt dieser immer wieder auf 0A.

Hat das Problem schonmal jemeand gehabt und kann mir sagen was ich machen muß ?

Danke

steff393, 2022-05-18:
Das ist das typische Fehlerbild bei Kommunikationsproblemen (oder auch bei aktiviertem Remote Lock, s. Wiki) und kann sehr sehr viele mögliche Ursachen haben: Störung/Unterbrechung der Modbusleitung, falsches Kabel, falsch gesetzte DIP-Schalter, Abschlusswiderstand, ...
Dass es aber "plötzlich" passiert ist eher untypisch.
Seit wann hast du wbec am laufen? Welche Ausgabe kommt bei /json?

elektronikfreak, 2022-05-18:
Hallo und vielen Dank für die Antwort, 

Ich habe über das bootlog rausbekommen das die ID 1 nicht gefunden werden konnte und bin auf dann auf die Suche gegangen. Erstmal den S4 in der Wallbox auf no follower gestellt damit das Auto wieder lädt und dann habe ich das Kabel durchgemessen. (Ich habe eine Modbusleitung von der Garage bis in den Keller.) Die zwei Adern und alle anderen im Kabel hatten keine durchgang mehr. Unglaublich und dann dämmerte es mir die Mäuse in der Garage waren es. Sie haben das Ganze Kabel durchgefressen sammt Schirm um 3:23 heute Nacht durchgefressen. Nun geht wieder alles.

Danke
Gruß

closed 2022-05-18 by elektronikfreak