steff393/wbec #38: Ladung wird unterbrochen, da WBec Controller aus dem STA-Mode fliegt

ChristianNebendahl, 2022-06-07:
Hallo Steff,
erstmals vielen Dank für Deine super Arbeit des WBECs. Super Lösung!
Ich habe das Problem, dass das WBEC Interface aus dem Client-Mode fliegt und das das Laden meines Autos darauf hin unterbrochen wird. Den Fehlerzustand kann man an der Heidelberg Control gut sehen, da diese dann abwechselnd 6x weiß und dann 3x blau blinkt (Kommunikationsfehler).  Dies passiert leider häufig, so dass die Gesamtlösung OpenWB mit WBec und Heidelberg Control unzuverlässig ist. Im Fehlerzustand finde ich dann auf meiner Fritzbox7590 das WEB als AP wieder. Wenn ich dann nur das WBec durch einen PowerOnReset neu starte, funktioniert alles wieder tadellos. 
Ich habe das WBec schon dichter an meine Fritzbox platziert, um Sendeunterbrechungen durch ein zu schwaches WLan zu reduzieren. Ich weiß nicht, ob eventuell nur temporär mein WLan (als potentielle Fehlerquelle) gestört ist und daraufhin das WBec des STA Mode verlässt und in den AP geht (und verbleibt). 

Christian

ChristianNebendahl, 2022-06-07:
PS: Das WBec Interface befindet sich direkt neben der Heidelberg Control (Kabellänge max. 60cm). Ich habe nur eine Heidelberg Control, die als Slave mit der ID=1 konfiguriert ist (S5 = 0000 und S4=0001 (slave address, Bus-ID)). Ebenso ist der 120 Ohm Abschlusswiderstand eingeschaltet (S6.2=1 (Bus Abschluss aktiv))

steff393, 2022-06-07:
Hallo Christian,
danke für das Lob ;-) 
Das kenne ich so bisher nicht. Eine Möglichkeit wäre, den folgenden Parameter auf 0 zu stellen (siehe auch Issue #12): 
wbec 0.4.2 will have an optional parameter cfgWifiSleepMode:
"cfgWifiSleepMode":0 => WIFI_NONE_SLEEP
"cfgWifiSleepMode":1 => WIFI_LIGHT_SLEEP
"cfgWifiSleepMode":2 => WIFI_MODEM_SLEEP
If it has a different value or is not set at all, then there's no change (i.e. WiFi.setSleepMode() is then not called).

Ich weiß nicht warum, aber seit ein paar Monaten gibt es häufiger WLAN-Probleme, die sich dadurch lösen lassen. Vorher war das nicht der Fall. Ich werd vermutlich in der nächsten Version den Wert per Default auf 0 stellen.

closed 2022-06-27 by steff393

ChristianNebendahl, 2022-06-28:
Hi, kurze Rückmeldung: Seit 3 Wochen läuft das wbec Interface bei mir problemlos, nachdem ich "cfgWifiSleepMode":2 => WIFI_MODEM_SLEEP (im CPP) und cfgMbCycleTime von 10s auf 3s in cfg.json gesetzt habe (siehe #12). 
Welche Einstellung letzten Endes entscheidend ist, hab ich nicht mehr ausgetestet, aber es geht jetzt!