steff393/wbec #39: Build fails: rtcvars.h: No such file or directory

VanKurt, 2022-06-29:
I just cloned this repository and opened it with VS Code PlatformIO.
When compiling I get the following error:

src/pvAlgo.cpp:11:10: fatal error: rtcvars.h: No such file or directory

PS: Here's the full compilation output:

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (3.2.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- modbus-esp8266 @ 4.1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- Ethernet @ 2.0.0
|   |   |-- SPI @ 1.0
|-- ESP Async WebServer @ 1.2.3
|   |-- ESPAsyncTCP @ 1.2.2
|   |-- Hash @ 1.0
|   |-- ArduinoJson @ 6.19.4
|   |-- AsyncElegantOTA @ 2.2.7
|   |   |-- ESP8266WiFi @ 1.0
|   |   |-- ESPAsyncTCP @ 1.2.2
|   |-- ESP8266WiFi @ 1.0
|-- ArduinoJson @ 6.19.4
|-- PubSubClient @ 2.8.0
|-- NTPClient @ 3.2.1
|-- MFRC522 @ 1.4.10
|   |-- SPI @ 1.0
|-- RTCVars @ 0.1.1
|-- WebSockets @ 2.3.7
|   |-- ESP8266WiFi @ 1.0
|   |-- ESPAsyncTCP @ 1.2.2
|   |-- Ethernet @ 2.0.0
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|   |-- AsyncElegantOTA @ 2.2.7
|   |   |-- ESP8266WiFi @ 1.0
|   |   |-- ESPAsyncTCP @ 1.2.2
|   |-- ESP8266WebServer @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|-- AsyncElegantOTA @ 2.2.7
|   |-- ESP8266WiFi @ 1.0
|   |-- ESPAsyncTCP @ 1.2.2
|-- Bounce2 @ 2.71.0
|-- LittleFS @ 0.1.0
|-- ESP8266mDNS @ 1.2
|   |-- ESP8266WiFi @ 1.0
|-- ArduinoOTA @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266mDNS @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|-- WiFiManager-asyncwebserver
|   |-- DNSServer @ 1.1.1
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WebServer @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- ESP Async WebServer @ 1.2.3
|   |   |-- ESPAsyncTCP @ 1.2.2
|   |   |-- Hash @ 1.0
|   |   |-- ArduinoJson @ 6.19.4
|   |   |-- AsyncElegantOTA @ 2.2.7
|   |   |   |-- ESP8266WiFi @ 1.0
|   |   |   |-- ESPAsyncTCP @ 1.2.2
|   |   |-- ESP8266WiFi @ 1.0
|-- EspSoftwareSerial @ 6.12.7
|-- ESP8266WiFi @ 1.0
|-- ESP8266HTTPClient @ 1.2
|   |-- ESP8266WiFi @ 1.0
|-- SPI @ 1.0
|-- ESPAsyncTCP @ 1.2.2
Building in release mode
Compiling .pio/build/nodemcuv2/src/button.cpp.o
Compiling .pio/build/nodemcuv2/src/globalConfig.cpp.o
Compiling .pio/build/nodemcuv2/src/goEmulator.cpp.o
Compiling .pio/build/nodemcuv2/src/inverter.cpp.o
Compiling .pio/build/nodemcuv2/src/loadManager.cpp.o
Compiling .pio/build/nodemcuv2/src/logger.cpp.o
Compiling .pio/build/nodemcuv2/src/main.cpp.o
Compiling .pio/build/nodemcuv2/src/mbComm.cpp.o
Compiling .pio/build/nodemcuv2/src/mqtt.cpp.o
Compiling .pio/build/nodemcuv2/src/phaseCtrl.cpp.o
Compiling .pio/build/nodemcuv2/src/powerfox.cpp.o
Compiling .pio/build/nodemcuv2/src/pvAlgo.cpp.o
Compiling .pio/build/nodemcuv2/src/rfid.cpp.o
src/pvAlgo.cpp:11:10: fatal error: rtcvars.h: No such file or directory

*****************************************************************
* Looking for rtcvars.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:rtcvars.h"
* Web  > https://registry.platformio.org/search?q=header:rtcvars.h
*
*****************************************************************

   11 | #include <rtcvars.h>
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio/build/nodemcuv2/src/pvAlgo.cpp.o] Error 1
================================================================== [FAILED] Took 3.92 seconds ==================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

steff393, 2022-06-29:
The file is from here: https://github.com/highno/rtcvars
But it should be resolved automatically, if PlatformIO is correctly installed (because the libaray is included in the platformio.ini)

VanKurt, 2022-06-30:
Hi Stefan, I think I found the cause of the problem:

In your code the include looks like this:

`#include <rtcvars.h>`

But the file name is actually **RTCVars.h**. Thus the include needs to be changed to

`#include <RTCVars.h>`

On Windows this does not matter (it does not care about casing). But since I'm building on Manjaro Linux, the casing is important.

steff393, 2022-06-30:
Good to know. Yes, this should be changed. If you want you can raise a PR, if not I can also modify it at the next chance.

VanKurt, 2022-06-30:
I have created the pull request.

closed 2022-06-30 by VanKurt