steff393/wbec #42: Energy since installation wurde zurückgesetzt

cgiudici, 2022-07-01:
Hi! Die Register 17/18 beinhalten ja die Energy since installation. Die sollte nach meinem Verständnis nicht durch Stromausfall etc. zurückgesetzt werden? Meine ist jetzt zwischendurch wieder auf 0 zurück, also der Wert "energyI" im JSON zeigt wieder niedrigere Werte an als vorher. Strom war auch mal weg in der Zwischenzeit. Hatte das jemand auch schon? Ist das ein Bug in der EnergyControl WB?
Ich ging davon aus, dass ich den Wert als Zähler für die Doku nutzen kann.

seppelon, 2022-07-01:
Ich kann bestätigen dass es ein komisches Verhalten gibt.
Für mich scheint es so, als ob im Betrieb mit Modbus der Zähler nicht im Eeprom der Box abgelegt wird.
Nach Netzausfall steht der per mqtt veröffentlichte Wert jedenfalls immer wieder bei 37kWh, dass ist die Menge die ich testweise geladen habe bevor ich die Steuerung mit Wbec angeschlossen habe. Ich habe mir allerdings bisher nicht die Mühe gemacht zu schauen was wbec genau ausliest und published, es gibt in der Wallbox ja einen Zähler für since install und einen since power on.

steff393, 2022-07-02:
Ja, ich habe davon auch schon vereinzelt gehört. Aber wbec kann nur den Wert lesen, den die Heidelberg ausgibt.

Bei Timeout nullt wbec auch die Werte (aber nur lokal in wbec), aber ich glaube nicht, dass das hier der Fall ist.

closed 2022-08-02 by steff393