steff393/wbec #44: Nicht erreichbar nach flash

Deanw1975, 2022-07-30:
Servus,

ich habe einen NodeMCU mit der 4.0.4 Version geflasht.
Nach der Einrichtung des WLAN finde ich diesen auch in meinem Router, kann mich aber nicht auf seine IP anmelden.
Wenn ich die Seite http://192.168.178.XXX/ aufrufe kommt nichts.
VG
Dean

steff393, 2022-07-30:
Hallo Dean,
probier bitte mal, ob andere Seiten, z.B. /json, /edit oder /heap erreichbar sind. 
Wenn ja, dann musst du einfach noch auf der Seite /edit die web.html und web.js aus dem "data"-Ordner hochladen.

Deanw1975, 2022-07-30:
Hi Steffan,

Ja da geht. Danke.

VG
Dean

steff393, 2022-07-31:
Hallo Dean, das Filesystem ist ein eigener Speicherbereich. Wenn ich den zum Binary hinzufügen würde, dann würde ein Update der Software auch die cfg.json überschreiben (und man müsste sich alles wieder neu einrichten). Ich hatte früher mal zwei separate Binaries, aber das hatte auch Verwirrung erzeugt.

closed 2022-07-31 by steff393