steff393/wbec #48: PV Überschuss Solaredge Geräte-ID

alexanderv92, 2022-09-19:
Hallo,

bei mir ist die Geräte-ID vom Solaredge Wechselrichter nicht 1 sondern 2. In Homeassistant muss ich die auf 2 setzen, damit ich die Werte bekomme. 

Das müssten im Code diese zwei Werte sein: 
#define INVERTER_SE_ADDRESS 
#define SMARTMETER_SE_ADDRESS

Könntest du diese als config-Variable einfügen? 

Dankeschön!

steff393, 2022-09-19:
Hallo Alex,

ok, danke für den Tipp. Das war mir noch nicht klar, dass es da unterschiedliche Einstellungen gibt.
Grundsätzlich sollte es möglich sein, das konfigurierbar zu machen. Ich weiß nur noch nicht, wie genau es am besten ist. Eigentlich war die Überlegung, dass man einfach den cfgInverterType umstellt und dann gleich die richtigen Einstellungen hat. Wenn mal alles einzeln einstellen kann wird es flexibler (was scheinbar eben nötig ist), aber auch komplizierter. 

Ich überleg mir mal was.

steff393, 2022-10-01:
Hab jetzt mal eine Lösung eingebaut, die beides abdecken sollte.
Die Auswahl der Wechselrichters geht weiterhin über `cfgInverterType`, wodurch die Standwerte festgelegt werden. 
Der Port und die Adressen können aber bei Bedarf mittels folgender Parameter überschrieben werden:
`cfgInverterPort` Port des Wechselrichters für Modbus-TCP
`cfgInverterAddr` Modbus-Adresse des Wechselrichters
`cfgInvSmartAddr` Modbus-Adresse des Smart-Meters

closed 2022-10-01 by steff393