steff393/wbec #53: Verbindungsproblem

Heiner1967, 2022-12-29:
Hi, ich habe 2 WB verbunden und nutze die intenre Steuerung. Das funktioniert soweit.

Beim anbinden von wbec:

veraendere ich die Konfig in der Leaderbox 
>>wird zum Follower
>>Systemleistung wird auf 0 gestellt da nur beim Leader konfiguriert
>>die BUS-ID wird auf 2 eingestellte (die andere Box war ja schon auf 1 als erster Follower)
>> der Abschlusswiderstand wird deaktiviert

Ist das richtig so? wbec wird ja zum Leader, oder?

Dann habe ich am D1mini einen MAX 3485 der aber im Vergleich zum Max 485 zwar die richtigeren 3,3V statt 5V fuer den ESP nutzt, aber das Ding hat RX und TX statt RO und DI und keinen RE/DE. Da ja RE/DE die Bereitschaft zum Senden/Empfangen bereit stellen soll und RO ja der Empfang ist habe ich hier RX angeschlossen und statt DI als Sender TX verbunden . Dau hab ich noch zwischen A und B die 120 Ohm Abschlusswiderstand eingefuegt.

Hab ich das alles so richtig verstanden:

Danke fuer ein Feedback

steff393, 2022-12-29:
Hallo Heiner,

das schaut eigentlich alles so richtig aus, wie du es beschreibst. Sollte eigentlich funktionieren. Evtl. noch TX/RX vertauschen (kenne jetzt deinen Treiber nicht). Wenn du noch Fragen hast, dann meld dich bitte per wbec393@gmail.com.

Viele Grüße
Stefan

closed 2022-12-29 by steff393

Heiner1967, 2022-12-30:
der RX/TX dreher war es, Danke fuer den Hinweis