steff393/wbec #54: Überschussleistung vom Stromzähler aus einem MQTT Topic lesen

emmrichd, 2022-12-30:
Hallo,

wie wäre es, die Überschussleistung der PV Anlage aus einem MQTT Topic zu lesen?
Damit wäre es möglich, die TASMOTA und SMLReader Leseköpfe für moderne Messeinrichtungen/Stromzähler zu lesen.

Viele Grüße, Dieter

steff393, 2022-12-30:
Ja, gute Idee, das sollte machbar sein.
Ich würde das Topic dann
`wbec/pv/setWatt` (Einheit 1W, +=Bezug, -=Einspeisung)
nennen.

Clanchef, 2022-12-31:
Das würde mir auch gefallen ... und wenn das dann auch noch mit 2 Heidelberg Wallboxen funktionieren würde :-)

emmrichd, 2022-12-31:
Hallo,

Super, dass Du die Idee aufnimmst.
Es wäre jedoch besser, man könnte das Topic definieren, weil man es bei
Tasmota und SMLReader nicht komplett frei definieren kann.
Viele Grüße, Dieter

steff393, 2023-02-11:
Weiß jemand wie das Standard-Topic bei Tasmota oder SMLReader heißt und wie die Einheit/Vorzeichen sind?
Dann würd ich das als Default gleich so vorsehen.

sja, 2023-02-11:
`13:42:59.491 MQT: tele/smartmeter/STATE = {"Time":"2023-02-11T13:42:59","Uptime":"34T16:50:48","UptimeSec":2998248,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":14,"Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":11,"Mode":"11n","RSSI":52,"Signal":-74,"LinkCount":11,"Downtime":"0T00:01:03"}}`
`13:45:44.492 MQT: tele/smartmeter/SENSOR = {"Time":"2023-02-11T13:45:44","Main":{"power":249,"from_grid":12961.441,"to_grid":9076.483}}`
Für OBIS 1.8.0 from_grid (12961kWh) und 2.8.0 to_grid (9076kWh).

hawa-lc4, 2023-02-11:
Je nach Zähler wird ein mehr oder weniger großer Datensatz geliefert.
Mein eBz DD3 liefert etwas mehr, besonders interessant sind hier aber die Werte zu 16.7.0 denn dieser Wert sagt aus ob Bezug oder Einspeisung vorliegt.
Ich habe diesen Zähler in die WBEC Software bereits mit eingebunden und übergebe die Daten per MQTT an openWB, läuft gut, und bin jetzt dabei das auf github zu dokumentieren (work in progress).

Näheres zu den Codes findet man im Netz, auch auf volkszaehler.org (https://wiki.volkszaehler.org/software/obis)

/EBZ5DD32R10DTB_107

1-0:0.0.0*255(1EBZxxxxxxxxxxxxx)
1-0:96.1.0*255(1EBZxxxxxxxxxxxxx)
1-0:1.8.0*255(005399.38571842*kWh)
1-0:1.8.1*255(000001.004*kWh)
1-0:1.8.2*255(005398.381*kWh)
1-0:2.8.0*255(000114.39640785*kWh)
1-0:16.7.0*255(000033.53*W)           <<hier!
1-0:36.7.0*255(000104.54*W)
1-0:56.7.0*255(-000092.91*W)
1-0:76.7.0*255(000021.90*W)
1-0:32.7.0*255(231.1*V)
1-0:52.7.0*255(235.0*V)
1-0:72.7.0*255(235.3*V)
1-0:96.5.0*255(001C2104)
0-0:96.8.0*255(0369E8A2)

!

emmrichd, 2023-02-11:
Für SMLReader ist es:
iot/smartmeter/sensor/1/obis/1-0:16.7.0/255/value
Leistung in Watt.
Der Wert hat ein Vorzeichen, minus für Einspeisung ins Netz.

steff393, 2023-02-11:
Ich habe es jetzt so umgesetzt (ab v0.4.7):
Es gibt 2 neue Parameter:
`cfgMqttWattTopic`: MQTT Topic, das den Wert für Bezug(+)/Einspeisung(-) enthält (max. 59 Zeichen)
`cfgMqttWattJson`: Ein Text, der dazu dient den gesuchten Wert in z.B. einem JSON-String zu finden (max. 14 Zeichen)

Der Wert nach dem cfgMqttWattJson wird mittels atol() in eine Zahl umgewandelt: 
- ein Minus wird korrekt interpretiert
- Nachkommastellen werden abgeschnitten
- Einheiten (z.B. W) ebenso

Wichtig noch: Ein JSON-String wird nicht korrekt deserialisiert, sondern nur von vorne bis hinten durchsucht. Der erste Wert der passt wird verwendet.
Damit sollten sich hoffentlich viele Varianten abdecken lassen. 

Default:
```
"cfgMqttWattTopic":"wbec/pv/setWatt"
"cfgMqttWattJson":""
```

SMLReader (s.o. von @emmrichd):
```
"cfgMqttWattTopic":"iot/smartmeter/sensor/1/obis/1-0:16.7.0/255/value"
"cfgMqttWattJson":""  <- hier nicht nötig
```

Tasmota (s.o. von @sja):
```
"cfgMqttWattTopic":"tele/smartmeter/SENSOR"
"cfgMqttWattJson":"power\":"   <- wichtig: Dem Anführungszeichen muss ein Backslash \ vorangestellt werden!
```

SMLReader (Variante von @hawa-lc4):
```
"cfgMqttWattTopic":"/EBZ5DD32R10DTB_107"
"cfgMqttWattJson":"1-0:16.7.0255("
```

steff393, 2023-02-11:
Leicht Off-Topic (frage für einen Freund): Kennt jemand einen fertigen Zählerkopf, der für den EMH ED300L Stromzähler geeignet ist und die Werte wie oben definiert übergeben könnte?

emmrichd, 2023-02-11:
Ja, ich schreibe dir persönlich.

Clanchef, 2023-02-20:
@steff393: Das geht nicht wenn man 2 Heidelberg Wallboxen an einer wbec dran hat, oder?

Bei mir sieht das so aus:
Topic: stat/tasmota/STATUS8
Value: {"StatusSNS":{"Time":"2023-02-20T14:34:56","SML":{"Verbrauch_T1":0.5883216,"Verbrauch_T2":13873.0200602,"Verbrauch_Summe":13873.6083819,"Einspeisung_Summe":30095.8603625,"Watt_L1":-759.00,"Watt_L2":-801.53,"Watt_L3":-844.11,"Watt_Summe":-2404.65,"Volt_L1":0.0,"Volt_L2":0.0,"Volt_L3":0.0}}}
--> "cfgMqttWattTopic":"stat/tasmota/STATUS8","cfgMqttWattJson":"StatusSNS.SML.Watt_Summe"
stimmt das?

steff393, 2023-02-20:
Die Überschussregelung von wbec geht generell nur für die 1. Wallbox. 
An das Thema den Strom dann auch noch irgendwie gerecht auf mehrere Wallboxen zu verteilen hab ich mich bisher noch nicht rangewagt ;-)

Dein cfgMqttWattJson ist zu lang (max. 14 Zeichen). Folgendes würde aber ja auch reichen (13 Zeichen):
`"cfgMqttWattJson ":"Watt_Summe\":"`
Der Text wird einfach nur von vorne bis hinten durchsucht (`"t_Summe\":"` würde theoretisch auch reichen, um die Stelle eindeutig zu finden). Ich seh schon, vielleicht sollte ich die Länge noch ein bisschen erhöhen ;-)

Clanchef, 2023-02-21:
Hallo Stefan, vielen Dank für die schnelle Antwort. 
mit 2 Wallboxen scheint das nicht zu funktionieren. Ich bekomme keine PV Leistung bzw. Watt_Summe bei Bezug/Einspeisung angezeigt.
Bzgl. Lastmanagement kann der evcc eigentlich nichts besser. Sollte an beiden Wallboxen geladen werden können regelt auch evcc den zur Verfügung stehenden Strom nicht "gerecht". Im Zweifel starten beide Fahrzeuge mit dem Laden, um danach festzustellen dass der Strom für beide gleichzeitig nicht reicht, so dass eine Wallbox pausieren muss. Meistens priorisiere ich das im evcc dann manuell, indem ich ein Fahrzeug auf "PV" stehen lasse, das zweite auf "Aus".

Ernie1806, 2023-03-25:
Hab gestern die 0.4.8 installiert und nun wird die `wbec/pv/setWatt` immer wieder auf 0 überschrieben. Könntest du das mal gegen checken? Danke

csh-de, 2023-04-30:
wo muss man das eintragen? In die cfg.json?

leider klappt es nicht...

meine cfg.json:
{"cfgApPass":"meinpasswort","cfgCntWb":1, "cfgMqttIp":"192.168.178.153","cfgMqttLp":[1],"cfgMqttUser":"name","cfgMqttPass":"userpasswort","cfgMqttWattTopic":"wbec/pv/setWatt","cfgMqttWattJson":""}

Die Daten sende ich per Note Red an wbec/pv/setWatt, laut mqtt explorer kommen die Werte an aber wbec nimmts nicht??
[https://ibb.co/85BgC5M](url)
https://ibb.co/85BgC5M

emmrichd, 2023-07-10:
Hallo,

die serverlose Übertragung mit einem Tasmota Lesekopf am Stromzähler zu wbec geht, das kann ich bestätigen.
Danke für diese Möglichkeit!
Gibt es eine Möglichkeit, die Pull-Frequenz zu ändern?
Viele Grüße, Dieter

emmrichd, 2023-08-26:
Hallo,

ich besitze noch einen Shelly 3EM, auch der kann via HTTP request die aktuelle Leistung ausgeben:
`http://192.168.180.191/status`
`{"wifi_sta":{"connected":true,"ssid":"LB30","ip":"192.168.180.191","rssi":-75},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"18:32","unixtime":1693067541,"serial":xxxx,"has_update":false,"mac":"xxxxx","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":126.84,"pf":0.56,"current":0.95,"voltage":241.04,"is_valid":true,"total":1579698.6,"total_returned":319024.5},{"power":144.12,"pf":0.75,"current":0.80,"voltage":242.05,"is_valid":true,"total":1457138.6,"total_returned":0.0},{"power":-82.76,"pf":-0.20,"current":1.70,"voltage":242.41,"is_valid":true,"total":1402685.9,"total_returned":1031754.7}],"total_power":188.20,"emeter_n":{"current":0.00,"ixsum":1.93,"mismatch":false,"is_valid":false},"fs_mounted":true,"v_data":1,"ct_calst":0,"update":{"status":"idle","has_update":false,"new_version":"20230510-083155/v1.13.1-gda6f9f2","old_version":"20230510-083155/v1.13.1-gda6f9f2","beta_version":"20230817-211859/v1.14.0-rc2-gf0fd560"},"ram_total":49920,"ram_free":30304,"fs_size":233681,"fs_free":154867,"uptime":4358199}`

Damit müsste doch:
`"cfgPvHttpIp":"192.168.180.191","cfgPvHttpPath":"/status","cfgPvHttpJson":"total_power"`
funktionieren? Geht irgendwie nicht.
Viele Grüße, Dieter

steff393, 2023-08-26:
Hallo Dieter,
ja, eigentlich schon. Genau, das \": beim total_power, das anfangs fehlte, ist wichtig. Es dauert auch ein bisschen (ca. 1 Minute) bis die Werte zum ersten Mal abgefragt werden.
Alternativ geht mit dem Shelly 3EM (nicht mit dem Pro) auch `cfgShellyIp`. Die anderen 3 Parameter kannst du dafür wieder entfernen.

emmrichd, 2023-08-26:
Danke, muss ich wbec nach Änderungen selbst neu starten? Wenn ja, wie?

steff393, 2023-08-26:
Ja, einfach Power off/on oder über http://wbec.local/reset

emmrichd, 2023-08-26:
Offtopic:, ist das das gleiche "Reset" wie auf der Config-Seite? Habe ich nie gedrückt. Reset ist bei mir, Konfiguration löschen, und nicht Restart. Reset für Restart finde ich nicht eindeutig. 

Mit der Shelly IP geht es. Komisch, dass der andere Weg nicht ging.

steff393, 2023-09-02:
Ja, das ist alles das gleiche:
- http://wbec.local/reset
- Reset auf der Config-Seite
- Reset-Knopf
Sorry ;-)

closed 2024-03-24 by steff393