steff393/wbec #55: V0.4.6 - WifiManager - Abstutz

Schnup89, 2023-01-04:
Hallo und vielen Dank für dein tolles Projekt!

Ich habe folgenden Fehler entdeckt:
Mit der Version 0.4.6 stürzt der ESP im WifiManager nach dem Flashen ab.

Den Fehler kann ich reproduzieren in dem ich per PyFlasher die V0.4.6 mit der "wipe" Option flashe.
Sobald ich mein Handy (android) mit dem wbec-wifi verbinde stürzt der ESP ab und startet neu.
Somit ist es nicht möglich den ESP im Wlan zu integrieren.
Gleicher ESP, gleiche Konstellation mit V0.4.5 funktioniert ohne Probleme.

Workaround:
Mit der V0.4.5 flashen, Wifi-Parameter setzen und dann Upgrade auf 0.4.6 ohne wipe.

Serial-Log bei Absturz:

```
Using 'COM7' as serial port.
Writing at 0x000d29ea... (100 %)Wrote 868032 bytes (591580 compressed) at 0x00000000 in 14.0 seconds (effective 496.0 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[21:17:18]
[21:17:18]
[21:17:18]Starting wbec ;-)
[21:17:19]01:00:00: CFG : Failed to open config file... Creating default config...
[21:17:19]Successful!
[21:17:19]01:00:00: CFG : cfgWbecVersion: v0.4.6
[21:17:19]01:00:00: CFG : cfgBuildDate: Dec 27 2022 18:52:35
[21:17:19]01:00:00: CFG : cfgCntWb: 1
[21:17:19]*WM: 
[21:17:19]*WM: AutoConnect
[21:17:19]*WM: Connecting as wifi client...
[21:17:19]*WM: Status:
[21:17:19]*WM: 0
[21:17:19]*WM: No saved credentials
[21:17:19]*WM: Waiting for connection result with time out
[21:17:29]*WM: Connection timed out
[21:17:29]*WM: Connection result: 
[21:17:29]*WM: 0
[21:17:29]*WM: 
[21:17:29]*WM: Configuring access point... 
[21:17:29]*WM: wbec
[21:17:29]*WM: wbec1234
[21:17:31]*WM: AP IP address: 
[21:17:31]*WM: 192.168.4.1
[21:17:31]*WM: HTTP server started
[21:17:45]*WM: Request redirected to captive portal
[21:17:45]*WM: Handle root
[21:17:45]*WM: Request redirected to captive portal
[21:17:45]
[21:17:45]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[21:17:45]
[21:17:45]Exception (28):
[21:17:45]epc1=0x4022b3bd epc2=0x00000000 epc3=0x00000000 excvaddr=0x0c7a000c depc=0x00000000
[21:17:45]
[21:17:45]>>>stack>>>
[21:17:45]
[21:17:45]ctx: cont
[21:17:45]sp: 3fff5080 end: 3fff55f0 offset: 0190
[21:17:45]3fff5210:  00000020 3fff52b4 3fff52b4 0000003b  
[21:17:45]3fff5220:  00000194 3fff52b4 3fff5884 4022b635  
[21:17:45]3fff5230:  00000000 312e312f 00000000 00000000  
[21:17:45]3fff5240:  312e312f 00000000 00000000 00000005  
[21:17:45]3fff5250:  00000000 00000000 312e312f 00000000  
[21:17:45]3fff5260:  3fff5314 00000020 00000014 00000014  
[21:17:45]3fff5270:  00000067 00000070 3fff5934 00000194  
[21:17:45]3fff5280:  3fff52f0 3fff5884 0000003b 4022b82a  
[21:17:45]3fff5290:  402c6af8 00000020 3fff531c 00000014  
[21:17:45]3fff52a0:  00000067 00000001 3fff5934 40231567  
[21:17:45]3fff52b0:  402c6af8 3fff704c 000d000f 89000020  
[21:17:45]3fff52c0:  3fff5314 0000000a 3fff5314 402315d8  
[21:17:45]3fff52d0:  3fff7524 3ffe90f2 3fff5314 3ffe90f2  
[21:17:45]3fff52e0:  3fff7524 00000194 3fff5884 4022b8fb  
[21:17:45]3fff52f0:  4023e3e4 00000000 000003e8 40101193  
[21:17:45]3fff5300:  00000000 3fff7524 0000003b 00000001  
[21:17:45]3fff5310:  00000000 74786574 616c702f 0a006e69  
[21:17:45]3fff5320:  0014001f 00000000 3fff534c 00000000  
[21:17:45]3fff5330:  3ffea310 00000000 3fff5884 4022c900  
[21:17:45]3fff5340:  00000000 65686361 00000000 00000000  
[21:17:45]3fff5350:  000e000f 00000000 00000000 000f000f  
[21:17:45]3fff5360:  00000000 3fff54e0 00000000 00000030  
[21:17:45]3fff5370:  3fff7524 003b003f 80000000 00000000  
[21:17:45]3fff5380:  3fff54e0 3fff58cc 3fff5884 4023d5b8  
[21:17:45]3fff5390:  00000000 00000000 3ffe8900 401001d9  
[21:17:45]3fff53a0:  00000000 000c000f 00000000 4022b956  
[21:17:45]3fff53b0:  00000000 00000000 00000000 69746365  
[21:17:45]3fff53c0:  00000000 4bc6a7f0 401009e5 00000000  
[21:17:45]3fff53d0:  00006a3c 00000000 00000000 00000001  
[21:17:45]3fff53e0:  00000000 00000000 3fff5884 00000000  
[21:17:46]3fff53f0:  3fff54e0 3fff58a8 3fff5884 4022d1a6  
[21:17:46]3fff5400:  4023e434 00000000 00001388 4023abeb  
[21:17:46]3fff5410:  00000000 00000000 3fff7414 00000000  
[21:17:46]3fff5420:  3fff54e0 3fff282c 3fff54d0 4022d2f3  
[21:17:46]3fff5430:  402ade97 00000001 00000001 402adf1e  
[21:17:46]3fff5440:  3fff61f4 00000000 00000000 3fff61f4  
[21:17:46]3fff5450:  00000000 00000000 3fff5480 402315d8  
[21:17:46]3fff5460:  3ffe8ca4 3ffe8ca4 3fff5480 3fff559b  
[21:17:46]3fff5470:  00000000 3fff282c 3fff54d0 4022d495  
[21:17:46]3fff5480:  00000000 3fff290c 00000000 00000000  
[21:17:46]3fff5490:  00000000 00000000 3fff54ec 3fff2b30  
[21:17:46]3fff54a0:  3fff555c 00000000 00000020 401014d3  
[21:17:46]3fff54b0:  3fffdad0 00000000 3fff54d0 3fff2b30  
[21:17:46]3fff54c0:  3fffdad0 00000000 3fff29d8 4020b9cc  
[21:17:46]3fff54d0:  3fff5844 3fff5884 3fff559b 3fff555c  
[21:17:46]3fff54e0:  00000000 feefeffe 00000000 00000000  
[21:17:46]3fff54f0:  feefeffe 00000000 00000000 00002710  
[21:17:46]3fff5500:  00006a3b 4023ec70 00000000 4023ec70  
[21:17:46]3fff5510:  00000000 4023ec70 00000000 4023ec70  
[21:17:46]3fff5520:  00000000 4023ec70 00000000 4023ec70  
[21:17:46]3fff5530:  00000000 00000000 ffffffff fe000001  
[21:17:46]3fff5540:  3ffe8ca4 00000000 fe010035 00000000  
[21:17:46]3fff5550:  00000000 0000000a 3fff61c4 63656277  
[21:17:46]3fff5560:  34333231 feefef00 feefeffe feefeffe  
[21:17:46]3fff5570:  feefeffe feefeffe feefeffe feefeffe  
[21:17:46]3fff5580:  feefeffe feefeffe feefeffe feefeffe  
[21:17:46]3fff5590:  feefeffe feefeffe 77efeffe 00636562  
[21:17:46]3fff55a0:  feefeffe feefeffe feefeffe feefeffe  
[21:17:46]3fff55b0:  feefeffe feefeffe feefeffe feefeffe  
[21:17:46]3fff55c0:  feefeffe feefeffe feefeffe feefeffe  
[21:17:46]3fff55d0:  feefeffe 00000000 3fff2b1c 40232b04  
[21:17:46]3fff55e0:  feefeffe feefeffe 3ffe8908 40101915  
[21:17:46]<<<stack<<<
[21:17:46]
[21:17:46]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[21:17:46]
[21:17:46] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[21:17:46]
[21:17:46]load 0x4010f000, len 3460, room 16 
[21:17:46]tail 4
[21:17:46]chksum 0xcc
[21:17:46]load 0x3fff20b8, len 40, room 4 
[21:17:46]tail 4
[21:17:46]chksum 0xc9
[21:17:46]csum 0xc9
[21:17:46]v000d3ec0
[21:17:46]~ld
[21:17:46]
[21:17:46]
[21:17:46]Starting wbec ;-)
[21:17:46]01:00:00: CFG : cfgWbecVersion: v0.4.6
[21:17:46]01:00:00: CFG : cfgBuildDate: Dec 27 2022 18:52:35
[21:17:46]01:00:00: CFG : cfgCntWb: 1
[21:17:46]*WM: 
[21:17:46]*WM: AutoConnect
[21:17:46]*WM: Connecting as wifi client...
[21:17:46]*WM: Status:
[21:17:46]*WM: 0
[21:17:46]*WM: No saved credentials
[21:17:46]*WM: Waiting for connection result with time out

```

steff393, 2023-01-21:
Ich habe jetzt mal die 0.4.6 nochmal mit den gleichen Sourcen aus dem Tag neu gebaut. Die Binaries unterscheiden sich leicht. Ich kann allerdings nicht genau bewerten wodurch die Unterschiede verursacht sind. 
Mit der neuen Datei von heute sehe ich den Fehler nicht mehr. Mit der alten kann ich ihn klar nachvollziehen. Vermutlich ist irgendetwas im Build-Prozess schiefgegangen.
Die Datei im Release habe ich ausgetauscht.

Schnup89, 2023-01-24:
Danke funktioniert! 👍