steff393/wbec #57: Standard-Ladestrom nach Neustart oder bei neuem Ladevorgang

Horst1989, 2023-01-31:
Hallo,

bei mir läuft wbec seit ein paar Wochen, zusammen mit einem BMW i3.
Leider muss ich bei jedem neuen Ladevorgang den Ladestrom wieder einstellen, standardmäßig setzt wbec die Wallbox auf 0A. 

Gibt es eine Einstellung wo ich den "default-Ladestrom" einstellen kann?

Vielen Dank!

steff393, 2023-01-31:
Hallo,
ist das bei anderen auch so? Wäre mir noch nicht so aufgefallen.

Viele Grüße
Stefan

AexKR, 2023-02-09:
Hallo, 

bei mir ist es ähnlich: ich habe dann MQTT abgeschaltet (ggfs. Zufall?) und seitem läuft es stabil. Aber nur so lange der Strom da ist. Sobald ist trenne und ein paar Tage später wieder in Betrieb nehme, muss einmalig der Laderegler verschoben werden. Das Verhalten hatte ich mit 0.45 und auch 0.46.

Viele Grüße,
Alex

Horst1989, 2023-02-13:
Hallo,

ich glaube der Hinweis mit Abschalten von MQTT hat geholfen, vielen Dank! :)

Hatte es aktiv um über EVCC ansteuern zu können aber das nie in Betrieb genommen, anscheinend wartet aber wbec immer auf einen Starttrigger...

closed 2023-02-13 by steff393