steff393/wbec #62: Comm-Failure BusID 1, aber Wallbox meint keinen Kommunikationsfehler zu haben

geek-at, 2023-02-23:
Ich habe die Wallbox heute angeschlossen und sie ist nicht in dem modbus error blink-state, also meint die Wallbox, dass sie eine erfolgreiche Modbus Verbindung hat, aber wbec auf dem d1 mini erhält keine Verbindung.

MQTT log sagt mir `Comm-Failure BusID 1`

Meine config:

```json
{
    "cfgApSsid": "wbec",
    "cfgApPass": "wbec1234",
    "cfgCntWb": 1,
    "cfgMqttIp": "192.168.1.51",
    "cfgMqttLp": []
}
```

JSON output der API sieht so aus:

```json
{
  "wbec": {
    "version": "v0.4.7",
    "bldDate": "Feb 11 2023 16:02:46",
    "timeNow": "21:52:58"
  },
  "box": [
    {
      "busId": 1,
      "version": "0",
      "chgStat": 0,
      "currL1": 0,
      "currL2": 0,
      "currL3": 0,
      "pcbTemp": 0,
      "voltL1": 0,
      "voltL2": 0,
      "voltL3": 0,
      "extLock": 0,
      "power": 0,
      "energyP": 0,
      "energyI": 0,
      "energyC": 0,
      "currMax": 0,
      "currMin": 0,
      "logStr": "",
      "wdTmOut": 0,
      "standby": 0,
      "remLock": 0,
      "currLim": 0,
      "currFs": 0,
      "lmReq": 131,
      "lmLim": 255,
      "resCode": "e4"
    }
  ],
  "modbus": {
    "state": {
      "lastTm": 34989499,
      "millis": 35000912
    }
  },
  "rfid": {
    "enabled": false,
    "release": false,
    "lastId": ""
  },
  "pv": {
    "mode": 0,
    "watt": 0
  },
  "wifi": {
    "mac": "2C:3A:E8:XX:XX:XX",
    "rssi": -82,
    "signal": 36,
    "channel": 11
  }
}
```

Weiß jemand, was das sein könnte?

hawa-lc4, 2023-02-24:
hallo geek-at,

ohne jetzt die Situation und deinen Hardwareaufbau genauer zu kennen wären meine ersten Gedanken zur Überprüfung:
- hat die Wallbox wie beschrieben die Bus-ID 1 erhalten? Wenn nicht fühlt sie sich nicht angesprochen, der Bus kann trotzdem für sie in Ordnung sein.
- wenn die wbec Hardware selbst aufgebaut ist, prüfen ob die MCU mit dem RS485 Adapter korrekt kommuniziert (Kabel, Pin-Belegung, etc. pp.).
- wenn auch weniger wahrscheinlich: die Bus-Terminierung in der Wallbox und am RS485 Adapter nochmal kontrollieren. Manche Adapter haben den 120 Ohm Widerstand automatisch aktiv, bei anderen muß man erst eine Lötbrücke setzen.
- ebenso weniger wahrscheinlich: die beiden Drähte 'A' und 'B' sollten nicht vertauscht sein.
- eventuell das Kabel zu lang oder nicht abgeschirmt? Läuft das Kabel eventuell an einem starken "Störsender" vorbei?

Die Wallbox kann nur die grundsätzliche Existenz und Funktion des RS485-Bus erkennen, wenn keiner mit ihr "redet" oder nur fehlerhafte Zeichen kommen wird sie das vermutlich nur wenig stören.
Ich kenne diese Situation bei mir nur wenn die Verbindung echt fehlt, sonst läuft hier die Kommunikation des wbec Moduls sehr stabil. Eine kleine Änderung habe ich allerdings gegenüber dem Original: bei mir läuft der RS485 Adapter nicht über den Software-Serial sondern über den UART0, also den Hardware-Serial.

hth.

geek-at, 2023-02-24:
Super, danke für die Pointer, ich werde das alles nochmal durchgehen

geek-at, 2023-02-25:
Hab jetzt die DIPs nochmal überprüft, sehen eigentlich gut aus

[image]

Hab den ganzen NodeMCU getauscht gegen einen anderen (auch D1 Mini), frisch geflasht mit `wbec_firmware_0_4_7.bin` und Pins doppelt gecheckt, die passen, sowie die Kabelpolarität aber sicherheitshalber auch ein anderes Kabel genommen.

Mein RS485 Adapter hat schon den 120Ohm Widerstand eignebaut, also hab ich meinen wieder raus genommen.

Leider immer noch `COMM-Failure BusID 1`

@hawa-lc4 du meintest, dass dus über den HW Serial bus laufen hast. Welchen RS485 Adapter hast du da genau?

Ich werde testen, ob es vielleicht mit einer älteren Firmware funktioniert.

andy5macht, 2023-02-25:
Versorgst du den MAX485 mit 5V oder mit 3,3V? Die Pegel für die Wallbox sollten 5V sein, also sollte der MAX485 mit 5V versorgt werden. optimalerweise mit pegelwandler dazwischen, ich habe es ohne pegelwandler aufgebaut, funktioniert bisher ohne probleme. Aber das hat ja hawa-lc4 schon geschrieben.

helix0712, 2023-02-26:
Hallo, ich hänge mich hier mal mit rein, da ich das gleiche Problem habe, allerdings mit 2 Boxen.
Beide hängen schon länger mittels Modbus (CAT6 Netzwerkkabel) an einem USB RS485 Adapter und dann an einem Linux Rechner.
Ich wollte jetzt auf WBEC umstellen und habe dazu einen D1 Mini und an dessen 5V Ausgang den RS485 Adapter gehängt (einen mit DE/RE) Pins. Mittlerweile habe ich 3 verschiedene Aufbauten (NodeMCU, 2x D1 Mini). WBEC hat Version 0.47. Ich bekomme ebenfalls immer:
11:27:14: MB  : Comm-Failure BusID 1
bzw.
11:27:13: MB  : Comm-Failure BusID 2
im MQTT log angezeigt.

Was mich halt wundert, ist zum einen der Status "e4" aber gleichzeitig scheinen alle Parameter korrekt ausgelesen zu werden..(siehe Screenshots) - wie bei geek-at auch.
Verkabelung habe ich schon mehrfach geprüft.

[image]
[image]

Vielleicht finden wir es gemeinsam heraus...

helix0712, 2023-02-26:
...und was ich gerade feststelle, da das E-Auto in der Garage angeschlossen wurde - es funktioniert trotzdem.
Der Fehler erscheint weiter im Log, aber die Ladesteuerung mittels evcc/mqtt funktioniert einwandfrei. Umschalten zwischen 6A/16A hat anstandslos funktioniert. Evtl.- liegt hier 'nur' ein kosmetisches Problem vor?

Gruß,

steff393, 2023-02-26:
Ja genau, manchmal kommt keine Antwort, oder sie ist gestört. Hab ich bei mir auch ab und an. 
Da die Werte aber alle eh 10s aktualisiert werden ist das kein echtes Problem, eher eine "Info". Man kann da auch mit den Parametern noch ein bisschen "nachtunen", ist aber eigentlich nicht nötig.

closed 2023-02-26 by steff393

geek-at, 2023-02-27:
Okay es wird immer merkwürdiger. Ich habe einen ESP32 programmiert, der über den selben RS485 converter mit der Heidelberg spricht und ich bekomme Daten heraus wie zB Temperatur und Spannung. Aber die Box blinkt im "no communication" modus. Sehr spannend.

myrdrks, 2023-06-05:
@geek-at das gleiche Problem habe ich auch. Hast du eine Lösung dafür?