steff393/wbec #68: ESP32-WROOM-32D Devkit V4 Error

hackicodi, 2023-03-16:
Hi,  
hier der Fehlerlog:

13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
[... 359 lines cut ...]
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344
wrong chip id 0x8344

steff393, 2023-03-16:
Kriterien für vernünftige Issues nicht erfüllt -> Closed

DcFamas, 2023-05-11:
Der ESP32 wird einfach nicht unterstützt. Du musst ein ESP8266 nutzen. Sofern du aber dies nichteinmal siehst, ist der Selbstbau nichts für dich, sorry. :(