steff393/wbec #70: Laden mit wbec nicht mehr möglich

mmoeller53, 2023-03-21:
Hallo, hab den Controller fertig gekauft und eingebaut, alles funktionierte erstmal.
Erste ernüchterung der Zoe mag erst ei 12,5A anfangen zu laden, OK. 
Somit bringt mir überschuss laden nicht viel, dachte ich lässt du es drinne kannste was steuern und hast es für die Zukunft.

Dann wollte meine Mutter laden, musste ich erst jedes Mal per Handy wbec auf 16A stellen (keine Ahnung ob man confog mässig das machen kann immer an, gerne auch per http knopf)

Also wbec per Go e api dauerhaft auf 16A gestellt, Auto läd

Nun läd aber seit 3 Tagen keiner der beiden zoes mehr weder der neue noch der alte,
Wenn ich wbec Aufrufe bekomme ich keine Anzeige ob Auto verbunden ist oder nicht, ob laden darf oder nicht, steht nur nen strich drinnen, ebenfalls bei den Ampere.

Stell ich es um auf 16A passiert weiterhin nichts.

Falls ihr Ideen habt immer her damit, anosnten muss ich es nachher Mal ausbauen um zu laden

steff393, 2023-03-21:
Hallo, läuft ganz unten im Bild die Zeit?
Bekommst du auf http://wbec.local/json eine Antwort? 

Löst es sich durch einen Reset (druck auf den kleinen Taster seitlich am Gehäuse oder Strom aus/an)?

mmoeller53, 2023-03-24:
Nach dem reset läuft es nun erstmal wieder wie gehab,
Werde es jetzt Mal beobachten, danke schonmal für die Hilfe

closed 2023-03-24 by steff393