steff393/wbec #71: Timeout beim Überschussladen

istmichseegal, 2023-04-04:
Autos mögen es nicht so besonders wenn sie häufig ein- und ausgeschaltet werden. Wie wäre es mit einem konfigurierbaren Timout sodass eine Freischaltung bei PV Überschuss mindestens z.B. 15min dauert.

steff393, 2023-05-20:
Habe dafür jetzt gerade den Parameter cfgPvMinTime eingebaut:
uint8_t  cfgPvMinTime;                // PV charging: Minimum activation time (in minutes), 0 to disable

Interessant wäre aber noch ein Beleg für "Autos mögen es nicht so besonders wenn sie häufig ein- und ausgeschaltet werden." 
Ist das wirklich so?

istmichseegal, 2023-05-20:
Klar "mögen" ist natürlich eine vermenschlichung der Maschine. Aber man muss immer bedenken, dass bei jedem einschalten entsprechende Systeme erst starten, Relais intern einschalten und erst dann der Ladevorgang beginnt. Mein i3 braucht z.B. immer ca. 30s bis der Ladevorgang beginnt nach der Freigabe. Relais haben auch nur eine gewisse Menge an Schaltspielen (Lebensdauer). Deswegen denke ich, dass es sinnvoll ist immer eine gewisse Zeit zu Laden bevor wieder abgeschaltet wird unabhängig vom Überschuss. Bei wechselnder Bewölkung kann dies sonst nämlich zu ständigem ein- bzw. Ausschalten führen

closed 2023-05-20 by istmichseegal