steff393/wbec #73: Fronius GEN24: Modbus ID

Chilled, 2023-04-28:
Die Modus Geräte ID der Gen24 Geräte ist die 200 bei den Snapin Invertern die 240. Hier wäre eine Konfiguration wünschenswert
Siehe Seite 13 [https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2649.pdf](https://www.google.com/url?sa=i&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=0CAQQw7AJahcKEwiAx4nSjc3-AhUAAAAAHQAAAAAQAw&url=https%3A%2F%2Fwww.fronius.com%2F~%2Fdownloads%2FSolar%2520Energy%2FOperating%2520Instructions%2F42%2C0410%2C2649.pdf&psig=AOvVaw0eqedETugKdfuyb5yCQyhZ&ust=1682789527437825)

ReWaKa, 2023-04-30:
Habe wohl das selbe Problem: es werden keine Smartmeter-Messwerte gefunden/angezeigt.
Beim Fronius Symo GEN24 6.0 wird als Zähleradresse 200 für das Fronius-Smartmeter verwendet.
Gehe ich richtig davon aus, dass das der Grund ist?
Mit http://<IP>/inverter wird folgendes ausgegeben:
inverter |  
isConnected | "1"
power |  
AC_Total | "0"
house | "4685"
inverter | "4685"
inverter_scale | "0"
meter | "0"
meter_scale | "0"

Könnte man die Zähleradresse parmetrierbar machen?

Chilled, 2023-04-30:
Ok.... Ich glaub ich habs hinbekommen. @steff393 hat die Funktion eingebaut den Port sowie die SmartMeter- und Inverter-ID in der cfg.json anzugeben. Folgende Werte können über die cfg.json angepasst werden.

cfgInverterPort
cfgInverterAddr
cfgInvSmartAddr

Sollte im Wiki die PV Überschussladen erwähnt werden. Bei den CFG Parametern ist es dokumentiert https://github.com/steff393/wbec/wiki/Übersicht-der-Parameter

ReWaKa, 2023-04-30:
@Chilled: Danke für den Hinweis - hatte nicht in die Parameterliste gschaut
@steff393: Danke! Nicht nur für diese Lösung, sondern fürs das Super-Projekt!

steff393, 2023-04-30:
Sorry für die späte Antwort. Ja genau, das sind die richtigen Parameter dafür.

closed 2023-04-30 by steff393