steff393/wbec #75: Not able to connect WBEC to openWB2 standalone via MQTT

TacitRainbow, 2023-05-13:
WBEC works fine, local webclient displays the correct values, configuration in WBEC seems to be okay 
{"cfgApPass":"wbec1234","cfgCntWb":1,"cfgMqttIp":"192.168.178.xx","cfgMqttLp":[15]}

...but I cannot connect it to the new openWB2 standalone box. There is a precofigured pulldown for a MQTT-chargepoint, but it doesn't offer further configuration choices. Also the topics to be sent and received seem to have different formats (openWB/chargepoint vs. openWB/lp). Can anybody offer an idea or an approach for a newbie (like me) how to fix this issue? THANKS

steff393, 2023-05-13:
Seems that I need to adapt the software to these new names. Can you provide me the topics to be sent and received?
Or is it really only the "lp" that needs to be changed to "chargepoint"?

TacitRainbow, 2023-05-13:
Hallo Steffan - klar. Scheint wirklich nur LP vs. chargepoint zu sein.

 

Viele Grüße

Dirk

 

Einstellungen für Ladepunkt MQTT-Ladepunkt

 

bei mir hat der MQTT-Ladepunkt die ID 16 - daher:

	von openWB gesendete Topics

	
		openWB/chargepoint/16/set/current
		Sollwert des Ladestroms, mit Nachkommastellen (Float), positiv (laden) oder negativ (entladen, falls vom Ladepunkt unterstützt).
		Beispiel: 12.34
		openWB/chargepoint/16/set/phases_to_use
		Sollwert für die Anzahl Phasen als Ganzzahl (Integer). Mögliche Werte: 1 oder 3
		Beispiel: 1
	
	
	von openWB zwingend erwartete Topics
	
		openWB/set/chargepoint/16/get/currents
		Aktuelle Phasenströme des Ladepunktes, Array mit drei Zahlen (Float), positiv (laden) oder negativ (entladen, falls vom Ladepunkt unterstützt).
		Beispiel: [6.78,6.89,6.94]
		openWB/set/chargepoint/16/get/power
		Leistung in Watt, mit Nachkommastellen (Float), positiv (laden) oder negativ (entladen, falls vom Ladepunkt unterstützt).
		Beispiel: 123.45
		openWB/set/chargepoint/16/get/plug_state
		Zeigt an, ob aktuell ein Fahrzeug angesteckt ist, Wahrheitswert (Bool).
		Beispiel: true
		openWB/set/chargepoint/16/get/charge_state
		Zeigt an, ob ein Ladevorgang aktiv ist, Wahrheitswert (Bool).
		Beispiel: true
		openWB/set/chargepoint/16/get/phases_in_use
		Anzahl der Phasen, mit denen geladen wird, Ganzzahl (Integer). Mögliche Werte: 0 bis 3
		Beispiel: 3
	
	
	optionale Topics
	
		openWB/set/chargepoint/16/get/imported
		Geladene Energie in Wh, mit Nachkommastellen (Float), nur positiv
		Wird dieses Topic nicht vom Ladepunkt gesendet, so wird intern ein Zählerstand anhand der Leistung simuliert.
		Beispiel: 123.45
		openWB/set/chargepoint/16/get/exported
		Entladene Energie in Wh, mit Nachkommastellen (Float), nur positiv
		Wird dieses Topic nicht vom Ladepunkt gesendet, so wird intern ein Zählerstand anhand der Leistung simuliert.
		Beispiel: 123.45
		openWB/set/chargepoint/16/get/voltages
		Aktuelle Phasenspannungen des Ladepunktes, Array mit drei Zahlen (Float), nur positiv.
		Die Spannungen werden nicht verarbeitet, sondern lediglich im Status angezeigt.
		Beispiel: [230.12,231.08,232.54]
		openWB/set/chargepoint/16/get/power_factors
		Aktuelle Leistungsfaktoren des Ladepunktes, Array mit drei Zahlen (Float), Wertebereich von -1 bis 1.
		Die Leistungsfaktoren werden nicht verarbeitet, sondern lediglich im Status angezeigt.
		Beispiel: [-0.87,0.96,1.0]
		openWB/set/chargepoint/16/get/rfid_tag
		Eine Zeichenfolge, die einen eingelesenen Tag für openWB repräsentiert.
		Beispiel: "123654"
	
	

 

endet: Samstag, 13. Mai 2023 um 12:12 Uhr

steff393, 2023-05-20:
Ok, seems to require more adjustments than just the "lp" <->"chargepoint" :-(
Do you know the meaning of the /chargepoint/**16**/
Is this is ID of the wallbox, like previously the load point number?

TacitRainbow, 2023-05-20:
Yup - 16 is the ID of the 'chargepoint' aka 'Ladepunkt' in previous versions. 

Another thing I am not sure if it needs to be checked out. Currently the openWB controls charging currency in 1A steps. This will be changed to finer steps in the future (not implemented yet). Hence I wonder if the multiplier in Mqtt.cpp line#45 might not need to be adjusted in due time. 

Sorry for the trouble - the openWB v2 is still in beta, so a few issues might magically disappear in due time anyway.

steff393, 2023-05-20:
Ok, should work now.
exported and power_factors are not available, but are anyway optional

closed 2023-05-26 by steff393

TacitRainbow, 2023-06-23:
works - as soon as OpenWB v.2 assigns a 1-digit ID to the chargepoint (might be necessary to delete existing chargepoints in the OpenWB-configuration; if new chargepoints are still assigned incrementally and are above 9, configure two chargepoints and delete the first one... seems to be a workaround)

alwayssimifuzi, 2023-09-02:
@steff393 How do I need to set cfgMqttWattTopic if I use two wallboxes with one wbec Premium? 
I'm new to OpenWB and just installed OpenWB 2.0

steff393, 2023-09-02:
Hi,
for openWB you do not need cfgMqttWattTopic at all.

All you need to configure is here: https://github.com/steff393/wbec/wiki/Configuration-Example:-openWB-via-MQTT
Normally only cfgMqttIp and cfgMqttLp

h202-ing, 2023-10-23:
Hi,
I configured it as described but it's not working here. It seems like the MQTT-Connection is working, openWB can read all topics from the wbec. But if I enable the charging in openWB (2.0) wbec does just nothing.

I checked the MQTT-communication with MQTT-Explorer --> openWB / system / chargepoint / 5 / set / current = 6
So the topic for the current is set but it seems like wbec doesn't recognize it.

I updated the wbec software to the current version and applied the "common.js"-fix. Same story as discribed.
Is there a chance to see what wbec receives from the openWB mqtt topic?

steff393, 2023-10-23:
Yes, you're right. I discovered yesterday in discussion with another user, that the openWB/chargepoint/X/set/current topic is not subscribed by wbec. I assume that openWB used up to now both topics simultaneously, and now only the new one.
I'll add this in the next release.

https://github.com/steff393/wbec/blob/638b5a39b79a4a859c88c32bb1672deb73f4dfe0/src/mqtt.cpp#L177-L182

h202-ing, 2023-10-31:
I played a bit around with the mqtt topics. If I replace 
https://github.com/steff393/wbec/blob/638b5a39b79a4a859c88c32bb1672deb73f4dfe0/src/mqtt.cpp#L177-L182

with this
```c++
snprintf_P(topic, sizeof(topic), PSTR("openWB/chargepoint/%d/set/current"), cfgMqttLp[i]);
client.subscribe(topic);
//snprintf_P(topic, sizeof(topic), PSTR("openWB/lp/%d/AConfigured"), cfgMqttLp[i]);
//client.subscribe(topic);
//snprintf_P(topic, sizeof(topic), PSTR("wbec/lp/%d/enable"), cfgMqttLp[i]);
//client.subscribe(topic);
//snprintf_P(topic, sizeof(topic), PSTR("wbec/lp/%d/maxcurrent"), cfgMqttLp[i]);
//client.subscribe(topic);
``` 

it seems like that the current in wbec is set to the correct value. I'm not sure if this is the only topic that is needed for the whole functionality but in a first testing it works fine.

I comment out the topics with "wbec/lp" because I'm not sure what they're doing. The wbec publish doesn't contain those topics (at least I didn't see them in the publish) so I think that they're needless.

hopefully this will help you a bit and maybe saves some time.

If there are other openWB topics needed I'm happy to provide support.

ArWi90, 2024-04-28:
Hi all,
have the additional MQTT Topics for OpenWB 2.0 ("chargepoint" instead of "lp") made it into the latest release?

I used WBEC with OpenWB 1.9 and it worked. I recently updated to OpenWB v2 and I cannot get the MQTT communication up and running.
I am using WBEC v0.5.2.

When I check the MQTT Topics in ioBroker, I don't see them getting updated.
[image]

[image]

The ID in OpenWB is 9, unfortunately continuously incrementing with every new try.

Any idea? Thank you!

TacitRainbow, 2024-04-30:
I had the same problem increment problem for LPs/chargepoints with openWB (I am using the stand-alone, version 2.1.4-Beta.2, release from April 29th) I simply added additional IDs for chargepoint until they ran over and started with the lowest, free single digit - then deleted all the not needed ones)

As per MQTT - ( am using the stand-alone, version 2.1.4-Beta.2 of OpenWB, release from April 29th and WBEC v1.5.2 for Pro) - I had to delete the 'old connection'  and set up a new one with the requisite IP-adress, then it worked perfectly fine- Otherwise - and I am not a real pro - your configuration looks fine.

ArWi90, 2024-05-03:
> I had the same problem increment problem for LPs/chargepoints with openWB 
> As per MQTT - ( am using the stand-alone, version 2.1.4-Beta.2 of OpenWB, release from April 29th and WBEC v1.5.2 for Pro)

1. To solve the incrementing ID issue in OpenWB 2, you just need to delete a the unnecessary components and restart the system. They will be gone.
2. Longer investigations of the source code and comparison to the comments in this thread + release notes showed me, that the changes were only implemented in pro and other versions - which you are using. Lucky you.
@steff393 Is it planned to recompile the standard version as well? I have a system up and running and just miss the correct readout of the MQTT topic defining the set current...
Otherwise I need to switch back to OpenWB 1.9.

I would be very thankful and could donate some money in case you create a new release. Thank you

ArWi90, 2024-05-27:
Hi @steff393,
did you already find time to take a look into this issue? 
Any support is appreciated!

Br. Arne

ArWi90, 2025-01-15:
Hi @steff393.
Are there any further steps planned to fix this issue?
Thank you.