steff393/wbec #77: Heidelberg Energy Control über EVCC / WBEC 1 Phasig laden? MQTT Daten offline?

Locutus-wolf359, 2023-05-26:
Moin,

ich würde gerne mit der Heidelberg PV Überschuß laden nutzen. Das WBEC Lan Modul habe ich verkabelt und eingerichtet. So alleine funktioniert es auch. Den Ladestrom kann ich auch reduzieren.

Unsere Anlage har nur nicht immer genügend Leistung für 3P und 4KW.

Könnte WBEC im Zusammenspiel mit EVCC die Phasen von 3 auf 1 reduzieren?

Aktuell bekomme ich WBEC nur als Go-e Wallbox zum laufen. Damit scheint es nicht zu gehen.

Ich versuche nun gerade WBEC über MQTT im EVCC einzubinden, hierfür habe ich einen RockPi als Brooker eingerichtet.

EVCC erhält auch Daten, aber der Status wird als Offline / outdated angezeigt, das Auto wird als nicht verbunden angezeigt. Habt ihr vielleicht einen Tipp?

`wbec
site
error = charger: wbec/lp/1/status outdated: 10s
gridEnergy = 58.7
battery
= 1
gridPower = 2
batteryPower = 344
tariffEffectiveCo2 = 0
pv
= 1
batteryCapacity = 0
batterySoc = 87
greenShare = 1
pvPower = 0
homePower = 357
tariffEffectivePrice = 0.086
loadpoints
1
chargeDuration = 0
sessionEnergy = 0
sessionPrice = <nil>
sessionSolarPercentage = 0
chargedEnergy = 0
chargePower = 0
mode = pv
sessionCo2PerKWh = <nil>
sessionPricePerKWh = <nil>
chargeCurrent = 0
updated = 1685129904
status = offline
$SYS`

Inhalt cfg.json
`{"cfgApPass":"wbec1234","cfgCntWb":1,"cfgWifiOff":1"cfgMqttIp":"192.168.1.42","cfgMqttLp":[1],"cfgMqttUser":"mqtt-evcc","cfgMqttPass":"Password"}` 

Danke und Gruß Christian

zivillian, 2023-05-26:
> Unsere Anlage har nur nicht immer genügend Leistung für 3P und 4KW.
> Könnte WBEC im Zusammenspiel mit EVCC die Phasen von 3 auf 1 reduzieren?

Leider nein. Die Heidelberg kann zwar entweder mit einer Phase oder mit 3 Phasen angeschlossen werden, aber selbst nicht umschalten (und wbec kann das somit auch nicht steuern).

> Ich versuche nun gerade WBEC über MQTT im EVCC einzubinden, hierfür habe ich einen RockPi als Brooker eingerichtet.

Wenn du sowieso evcc nutzen möchtest, stellt sich mir die Frage, [wofür du noch wbec brauchst](https://docs.evcc.io/docs/devices/chargers#heidelberg-energy-control-)

Um deine eigentliche Frage zu beantworten: Ich hatte die gleiche Herausforderung und habe mir [eine Umschaltung gebaut](https://github.com/zivillian/heidelberg-1p3p), die ([vielleicht irgendwann auch von evcc gesteuert](https://github.com/evcc-io/evcc/discussions/6168)) zwischen 1p und 3p umschalten kann.

Locutus-wolf359, 2023-05-27:
Danke für die Tipps. Werde ich mir ansehen.

> Wenn du sowieso evcc nutzen möchtest, stellt sich mir die Frage, [wofür du noch wbec brauchst](https://docs.evcc.io/docs/devices/chargers#heidelberg-energy-control-)

Der Plan war eigentlich die App Wallbox-Steuerung mit dem WBEC Adapter zu nutzen - ohne viel gebastel. Das hat nur nicht so wie gedacht funktioniert. Da die App bei meinem WR auch den aus dem Speicher entnommen Strom als Solar Ertrag gemessen hat. Abends beims Kochen war also 3.5kw Solar Ertrag da. ;)

Mit einem RS485 Adapter habe ich es noch nicht sauber zum laufen bekommen. 

Mit dem WBEC und dem Go-E Profil in EVCC funktioniert es bis auf die Phasen Umschaltung ganz gut. Momentan steuere ich das dann manuell mit Min+PV an.

hawa-lc4, 2023-05-27:
Deine Frage:  "Könnte WBEC im Zusammenspiel mit EVCC die Phasen von 3 auf 1 reduzieren?"
Wurde von zivillian bereits beantwortet, aber es ist nur die halbe Antwort.

In der referenzierten Dokumentation heißt es wörtlich: "Diese Funktion ist leider mit der Heidelberg Energy Control **_alleine_** nicht möglich."
Das Wort "alleine" ist entscheidend, man muß also etwas nachhelfen.

Ich kenne EVCC nicht aber ich habe im Wesentlichen das Gleiche mit openWB realisiert.
Auch dieses Management System kann eine Phasenumschaltung  der WB anfordern und wbec kann damit auch umgehen.
Die Umschaltung selbst mache ich mit einem 2-poligen Hilfsschütz und einer kleinen Box basierend auf Tasmota.
Beim Schütz unbedingt auf ausreichend Schaltkapazität achten. Mit diesem schalte ich die Phasen L2 und L3 ab oder zu, je nach Bedarf.
Bitte auch unbedingt sicherstellen daß das EV mit der Phasenumschaltung umgehen kann. Es gibt Fahrzeuge da führt das zu Schäden an der Elektronik; der Zoe soll so ein Kandidat sein. Mein Skoda citigo-e hat sich bisher nicht beschwert.

Zur Frage von zivillian "..., wofür du noch wbec brauchst"
Bei EVCC hat mich das pay-Modell gestört und openWB unterstützt die Energy Control nativ nicht.
Mit wbec als Kopplungsglied ist das aber prima.

Derzeit plane ich meine wbec Box mit der Tasmota Box zu vereinen und das Schütz über wbec direkt anzusteuern, das nur so zur Info.

steff393, 2023-05-28:
Die Modbus-Abfrage der Wallbox findet normalerweise alle 10s statt. Direkt danach werden die Werte per MQTT geschickt. Vielleicht hilft es, "cfgMbCycleTime":5 zu setzen, wenn die 10s für EVCC zu langsam sind.

Locutus-wolf359, 2023-05-29:
Danke für Eure Antworten.

Momentan komme ich mit Min+PV ganz gut zu Recht. 

Hat das irgendwelche Nachteile, wenn ich die Heidelberg als GO-E einbinde? Hardwareschäden oder ähnliches? Sonst würde ich es jetzt erstmal so laufen lassen.

Mit MQTT brauche ich dann ja auch nicht weiter probieren, ich dachte das dann die 1/3P Umschaltung funktionieren würde.

Werde mir die nächsten Monate mal ansehen wie es läuft und sonst auf eine PV fähige WB umsteigen.

steff393, 2023-05-29:
Nein, das hat keine Nachteile. MQTT habe ich nur zusätzlich eingebaut, damit man auch 2 oder mehr Heidelberg an EVCC betreiben kann.

closed 2023-05-29 by steff393

andy5macht, 2023-05-30:
Ich realisiere das Umschalten von 3 auf 1 Phase aktuell manuell.
Im Herbst/Winter drehe ich zwei Sicherungen raus und zur Garage geht dann nur noch 1 Phase, welche die Wallbox versorgt.
Das blöde ist, dass ich den Parameter: cfgPvPhFactor immer von 69 auf 23 Umstellen muss und im Frühjahr umgekehrt.
Könnte man das nicht automatisch anhand der Spannungen pro Phase, welche die Wallbox misst, umstellen?
Dann bräuchte ich nur die Sicherungen Aus/Einschalten und trotzdem kann ich per PV mit der maximalen Leistung die eben Möglich bzw. Überschuss ist.

Eine Umsetzung würde mich freuen. Ob ich dann irgendwann die Automatische Phasenumschaltung mache überlege ich mir dann noch. (z.b. Nur wenn kein Auto angesteckt ist oder so ähnlich würde ich das Realisieren.

steff393, 2023-05-30:
Hi,
das sollte über folgenden Code in pvAlgo() gehen:
```
// Phase factor auto-detection - based on wallbox connection
	if (cfgPvPhFactor == 0) {
		if (content[BOXID][6] > LIMIT_230V && 
				content[BOXID][7] < LIMIT_0V && 
				content[BOXID][8] < LIMIT_0V) {
			phaseFactor = 23;
		} else {
			phaseFactor = 69;
		}
	} else {
		phaseFactor = cfgPvPhFactor;
	}

	// Phase factor auto-detection - based on car connection
	if (cfgPvPhFactor == 1) {
		if (content[BOXID][2] > 0 && 
				content[BOXID][3] == 0 && 
				content[BOXID][4] == 0) {
			phaseFactor = 23;
		} else if (content[BOXID][2] > 0 && 
							 content[BOXID][3] > 0 && 
							 content[BOXID][4] == 0) {
			phaseFactor = 46;
		} else {
			phaseFactor = 69;
		}
	} else {
		phaseFactor = cfgPvPhFactor;
	}
```
Und unten in der Formel dann cfgPvPhFactor durch phaseFactor ersetzen. Hab aber bisher noch nicht ausreichend Testergebnisse.

andy5macht, 2023-06-02:
Cool,  super Idee. Werde ich mir bei Gelegenheit anschauen bzw. ausprobieren.