steff393/wbec #79: RFID Karte funktioniert nicht mehr

DcFamas, 2023-06-13:
Hallo zusammen,
ich habe den genannten RFID Kartenleser an meiner WBEC Installation hängen. Zu Beginn hatte ich hier keine Probleme, nun jedoch wird die Karte erkannt und ist auch i.O., das Laden wird jedoch nicht freigegeben. Unter /json sehe ich, dass die Karte i.O. ist. Leider lassen sich alle Seiten öffnen, bis auf den /edit Reiter. Dieser zeigt leider einfach keine Sachen an, also nur den Balken oben. Selbst das Menu fehlt leider. Über die Android App kann ich das Laden derzeit noch manuell aktivieren.

Muss ich hier ggf. den ESP neu flashen?

Viele Grüße :)

steff393, 2023-06-14:
Hallo,
bisher ist mir noch kein Fall bekannt, wo das /edit nicht mehr ging. Vor allem weil das nur per SW-Änderung "entfernbar/deaktivierbar" wäre. Kann es vielleicht am Browser liegen (Javascript, etc.)?

DcFamas, 2023-06-14:
Hey,
leider nein. Habs mit dem Handy und dem Laptop probiert. Das Gerät hat derzeit aber noch keinen richtigen Internetzugang. /status und alles andere läuft i.O. 

Grüße

steff393, 2023-06-14:
Ah, ja. Dieser Editor (ACE) wird vom Browser aus dem Internet nachgeladen. Wenn das Gerät, mit dem du /edit aufrufst also keinen Internetzugang hat, dann geht es nicht.
https://ace.c9.io/

DcFamas, 2023-06-15:
Moin, danke für deine schnelle Antwort. Wäre es grundsätzlich möglich dies zu ändern? Das Gerät hat in der Garage eigentlich kein Internet. Das /edit nicht funktionsfähig ist, ohne einen Internetzugang, finde ich persönlich noch i.O.

Leider gibt er ja dann auch ohne Internetzugriff nicht die Ladekarte frei, obwohl sie eigentlich unter /json als success steht. Ich teste heute Nachmittag mal mit einer SIM-Karte in dem Hotspot. :)

Viele Grüße

steff393, 2023-06-15:
Hallo DcFamas,
das dürfte eigentlich nichts miteinander zu tun haben.
wbec selber braucht kein Internet (gut, ein Zeitserver sollte schon erreichbar sein). Das Endgerät mit dem du /edit aufrufst sollte aber Internetzugriff haben, da dessen Browser dann von hier das Javascript für den Editor holt:
`<script charset=utf-8 id=ace src=https://cdnjs.cloudflare.com/ajax/libs/ace/1.2.6/ace.js></script>`

Das ist aber wie gesagt eine Kommunikation zwischen Browser und o.g. Adresse. wbec liefert danach nur noch den Inhalt der Dateien an den Browser.

DcFamas, 2023-06-16:
Hab es gestern ausprobiert. Dadurch, dass der Hotspot keine SIM-Karte hat, hat das verbundene Smartphone ja auch keinen Internetzugriff. Mit SIM-Karte im Hotspot ist der Aufruf /edit wieder möglich. Grundsätzlich ludt die Webseite viel schneller. Leider hatte ich gestern die Ladekarte nicht zur Verfügung, ich berichte dennoch heute noch.

Wenn das nicht hilft, neu flashen?

DcFamas, 2023-06-23:
Tatsächlich funktioniert die RFID Karte nun. Ohne die Internetverbindung ist wbec relativ träge. Mit Internetverbindung ist alles i.O. und schnell. Liegt es ggf. daran? :/

steff393, 2023-06-24:
Der eigentliche Ablauf (Kartenerkennung -> Freischaltung der Wallbox) sollte unabhängig vom Internet sein. Auch für /json sollte Internet keine Rolle spielen. Für /edit kommt aber eben der Editor aus dem Internet.

closed 2023-06-29 by steff393