steff393/wbec #85: Charge Log has incorrect data

salamander-in-steinachtal, 2023-07-05:
The highly useful charge log (many thanks), is returning an odd and constant value of 4294864.989, the charge times are absolutely correct.

{"wbec":{"version":"v1.4.9","bldDate":"May 28 2023 21:45:14","timeNow":"10:57:23"},"box":[{"busId":1,"version":"108","chgStat":4,"currL1":0,"currL2":0,"currL3":0,"pcbTemp":268,"voltL1":236,"voltL2":240,"voltL3":240,"extLock":1,"power":0,"energyP":7.21,"energyI":102.307,"energyC":0,"currMax":16,"currMin":6,"logStr":"00.779.2909/01 022900 2147 HD_EnergyControl","wdTmOut":60000,"standby":4,"remLock":1,"currLim":0,"currFs":0,"lmReq":0,"lmLim":255,"resCode":"0"}],"modbus":{"state":{"lastTm":2245270202,"millis":2245272613}},"rfid":{"enabled":false,"release":false,"lastId":""},"pv":{"mode":0,"watt":0},"wifi":{"mac":"78:21:84:A6:EA:28","rssi":-64,"signal":72,"channel":11}}

web.html --> Energiezähler 102.307 kWh (Correct)

log.html --> (Incorrect)
Ladeübersicht Start	Ende Geladene kWh Ladedauer hh:mm	Wallbox
04.07.2023 | 05:43	05.07.2023 | 11:03	0.000	29:20	1
03.07.2023 | 22:05	04.07.2023 | 05:42	4294864.989	7:37	1
03.07.2023 | 06:09	03.07.2023 | 22:05	4294864.989	15:55	1
03.07.2023 | 00:33	03.07.2023 | 06:09	4294864.989	5:35	1
02.07.2023 | 13:18	03.07.2023 | 00:31	4294871.249	11:13	1
02.07.2023 | 08:10	02.07.2023 | 13:17	4294872.199	5:07	1
14.06.2023 | 07:06	14.06.2023 | 11:01	0.000	3:54	1
12.06.2023 | 12:20	14.06.2023 | 07:06	4294872.199	42:46	1
12.06.2023 | 10:25	12.06.2023 | 12:15	4294872.199	1:49	1
11.06.2023 | 12:40	12.06.2023 | 10:20	4294872.199	21:39	1

steff393, 2023-07-09:
I can currently not reproduce the behavior.
If the strange values would be interpreted as signed instead of unsigned they would be shown as:
4294864,989 => -102,307 kWh --> fits to correct value of Energiezähler
4294871,249 =>  -96,047 kWh
4294872,199 =>  -95,097 kWh
4294872,199 =>  -95,097 kWh

wbec calculates energyI (=Energy since installation) of ChargeEnd - energyI of ChargeStart
So it seems that somehow energyI at ChargeEnd was seen as 0.

On the other hand, it can be seen, that the endTime of one entry is often the startTime of the next entry.
Does it help to delete the chargeLog-0.txt file? Can be done in /edit by right-click on the file.

salamander-in-steinachtal, 2023-07-09:
It’s got weirder over the past days … going backwards - and the last entry looks like it could be right, perhaps this happens under strange start condition.

I’ll delete the log as you suggest and update. 

04.07.2023 | 05:43	09.07.2023 | 19:16	0.201	133:33	1
03.07.2023 | 22:05	04.07.2023 | 05:42	4294864.989	7:37	1
03.07.2023 | 06:09	03.07.2023 | 22:05	4294864.989	15:55	1
03.07.2023 | 00:33	03.07.2023 | 06:09	4294864.989	5:35	1
02.07.2023 | 13:18	03.07.2023 | 00:31	4294871.249	11:13	1
02.07.2023 | 08:10	02.07.2023 | 13:17	4294872.199	5:07	1
14.06.2023 | 07:06	14.06.2023 | 11:01	0.000	3:54	1
12.06.2023 | 12:20	14.06.2023 | 07:06	4294872.199	42:46	1
12.06.2023 | 10:25	12.06.2023 | 12:15	4294872.199	1:49	1
11.06.2023 | 12:40	12.06.2023 | 10:20	4294872.199	21:39	1

salamander-in-steinachtal, 2023-07-14:
Sorry, still doing it... 

Start	Ende	Geladene
kWh	Ladedauer
hh:mm	Wallbox
12.07.2023 | 00:26	14.07.2023 | 13:29	0.000	61:03	1
11.07.2023 | 04:05	12.07.2023 | 00:25	4294864.788	20:20	1
09.07.2023 | 21:03	11.07.2023 | 04:05	4294864.788	31:01	1

Note that the car is not used a lot, and plugged in all the time

steff393, 2023-07-20:
Kannst du bitte mal prüfen, ob wbec zu diesen Zeitpunkten einen Reset hatte? 
In http://wbec.local/json sieht man unter "millis" wie viele Millisekunden wbec schon läuft. Wäre interessant, ob dies zurückgerechnet genau zu den Zeitpunkten mit den seltsamen Werten passt.

salamander-in-steinachtal, 2023-07-20:
{"wbec":{"version":"v1.4.9","bldDate":"May 28 2023 21:45:14","timeNow":"00:39:53"},"box":[{"busId":1,"version":"108","chgStat":4,"currL1":0,"currL2":0,"currL3":0,"pcbTemp":275,"voltL1":242,"voltL2":241,"voltL3":240,"extLock":1,"power":0,"energyP":7.411,"energyI":102.508,"energyC":0,"currMax":16,"currMin":6,"logStr":"00.779.2909/01 022900 2147 HD_EnergyControl","wdTmOut":60000,"standby":4,"remLock":1,"currLim":0,"currFs":0,"lmReq":0,"lmLim":255,"resCode":"0"}],"modbus":{"state":{"lastTm":3590614516,"millis":3590614831}},"rfid":{"enabled":false,"release":false,"lastId":""},"pv":{"mode":0,"watt":0},"wifi":{"mac":"78:21:84:A6:EA:28","rssi":-73,"signal":54,"channel":11}}

salamander-in-steinachtal, 2023-07-20:
Ladeübersicht

Start	Ende	Geladene
kWh	Ladedauer
hh:mm	Wallbox
09.07.2023 | 21:03	11.07.2023 | 04:05	4294864.788	31:01	1
11.07.2023 | 04:05	12.07.2023 | 00:25	4294864.788	20:20	1
12.07.2023 | 00:26	15.07.2023 | 22:12	4294864.788	93:45	1
15.07.2023 | 22:14	18.07.2023 | 09:34	4294864.788	59:19	1
18.07.2023 | 10:35	18.07.2023 | 10:59	0.000	0:23	1
18.07.2023 | 19:31	20.07.2023 | 18:31	4294864.788	47:00	1
20.07.2023 | 18:32	21.07.2023 | 00:41	0.000	6:09	1

salamander-in-steinachtal, 2023-07-20:
Current time 00:44

kalwados, 2023-07-21:
Ich habe hier ein vergleichbares Problem seit dem Update auf die V0.4.9 (vorher 0.4.5) an einer Heidelberg Energy Control; derzeit noch ohne PV-Anlage.

In der JSON-Datei steht dann u.A:
`"energyP":0,"energyI":0,"energyC":4292920.576,..."resCode":"e4"`

Nach einen Reset der Wallbox (Sicherung aus und wieder ein) ist für ein paar Minuten alles in Ordnung:
`"energyP":0,"energyI":2046.686,"energyC":0,..."resCode":0`
15 Minuten später ist der Fehler wieder da.

Wenn ich dann einen Reset des wbec-Controllers durchführe, stehen nur noch Null-Werte in der JSON-Datei. 

Ich hänge hier mal drei JSON-Dateien an:
- OK.json -> nach Neustart der Wallbox
- Error.json -> ca. 15 Minuten später
- Reset.json -> nach Reset von wbec.

Hoffentlich kannst Du mit diesen Information die Fehlerursache eingrenzen; gerne liefere ich weitere Daten nach!

Gruß aus dem Tecklenburger Land
Achim

[OK.json.txt](https://github.com/steff393/wbec/files/12131699/OK.json.txt)
[Error.json.txt](https://github.com/steff393/wbec/files/12131701/Error.json.txt)
[Reset.json.txt](https://github.com/steff393/wbec/files/12132990/Reset.json.txt)

steff393, 2023-08-26:
@kalwados: Du hast den Standby aktiviert, oder?
In der nächsten Version kommt eine Änderung, die damit besser umgehen kann. Aktuell würde ich ihn eher deaktivieren.

steff393, 2023-08-26:
@salamander-in-steinachtal 
Kannst du bitte noch posten, was bei http://wbec.local/chargelog?id=0&len=10 zurückgegeben wird?
Und bitte auch noch die Datei chargeLog-0.txt hier uploaden?

kalwados, 2023-08-26:
Ja, habe Standby kurz nach meiner Meldung wieder deaktiviert, läuft bislang ohne Probleme.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

salamander-in-steinachtal, 2023-08-27:
> @salamander-in-steinachtal
> Kannst du bitte noch posten, was bei http://wbec.local/chargelog?id=0&len=10 zurückgegeben wird?
> Und bitte auch noch die Datei chargeLog-0.txt hier uploaden?

{"line":[{"timestamp":1689452095,"duration":213575,"energy":4294864788,"user":0,"box":0},{"timestamp":1689669334,"duration":1424,"energy":0,"user":0,"box":0},{"timestamp":1689701505,"duration":169205,"energy":4294864788,"user":0,"box":0},{"timestamp":1689870749,"duration":183411,"energy":4294864788,"user":0,"box":0},{"timestamp":1690055015,"duration":208175,"energy":4294864788,"user":0,"box":0},{"timestamp":1690263216,"duration":15121,"energy":4294860175,"user":0,"box":0},{"timestamp":1690278349,"duration":271,"energy":4294860175,"user":0,"box":0},{"timestamp":1690278833,"duration":874,"energy":0,"user":0,"box":0},{"timestamp":1690908198,"duration":20846,"energy":4294860175,"user":0,"box":0},{"timestamp":1690929055,"duration":63001,"energy":5898,"user":0,"box":0}]}

steff393, 2023-08-29:
Danke, ich kann's mittlerweile nachstellen. Wird in der nächsten Version korrigiert!

closed 2023-08-29 by steff393

salamander-in-steinachtal, 2023-08-30:
Genius - many thanks!