steff393/wbec #89: Verbesserungsvorschlag: Das Array "cfgMqttWattJson" erweitern oder konfigurierbar machen. 30 wird oft knapp werden.

alocinb, 2023-07-17:
https://github.com/steff393/wbec/blame/b702a9e48b1000780891ef55d1aaa81b2c3bad4c/src/globalConfig.cpp#L31

Gruß
NIBO

alocinb, 2023-07-18:
na ja da muss ich nicht lang suchen, schaue in dein Programm mqtt.cpp Zeile 134 // Example: {"Time":"2022-12-10T17:25:46","Main":{"power":-123,"from_grid":441.231,"to_grid":9578.253}}

der Example von dir ist schon 91 Zeichen lang.

Gruß
NIBO

alocinb, 2023-07-18:
sorry, habe ich vergessen zu erwähnen, wenn das cfgMqttWattJson  Array kleiner ist als das Mqtt Json empfangener Wert, startet der ESP immer wieder neu.
Danach hilft nur der MQTT Broker zu stoppen, und der Wert in cfgMqttWattJson zu entfernen.

alocinb, 2023-07-18:
Sorry, Sorry, der Fehler liegt an eine andere Stelle.
Ich hatte ein MQTT Topic wo fast 240 Zeichen lang wahr, und dadurch ist der ESP immer abgestürzt.
Danach habe ich das "cfgMqttWattJson" erweitert auf 240 und gleichzeitig, weil ich nicht der ganze Topic im Terminal Log gesehen habe, die Zeile 39 in mqtt.cpp von LOGN auf LOGEXT getauscht.
Und das wahr die Lösung, die LOGN Funktion kann nur 100 Zeichen ausgeben und LOGEXT kann 600, dadurch ist wahrscheinlich der ESP nicht mehr abgestürzt, und ich dachte das liegt am cfgMqttWattJson bzw cfgMqttWattTopic.
LOGEXT könnte hier helfen.
Sorry für die falschen annahmen von mir.

Gruß
NIBO

steff393, 2023-07-20:
Ah, ok, ja das muss ich mir dann tatsächlich nochmal anschauen.

closed 2023-08-26 by steff393