steff393/wbec #92: Die neue Webgui zeigt Update Option auf wbecLan

hpoeppel, 2023-07-20:
Die neue Webgui zeigt auf dem wbecLan einen Update Button, der aber nicht funktioniert. 

Fehlermeldung ist "FileNotFound" in der Data.zip ist auch keine Datei dafür enthalten.

Installierte Version ist 2.4.9

alocinb, 2023-07-20:
Hallo, siehtst du über Edit ob es eine Datei gibt = cfg.json ?

steff393, 2023-07-20:
Ja, das ist ein bekanntes Verhalten: wbecLan kann nur mittels curl-Befehl aktualisiert werden (s. Seite 10 in der Anleitung). Daher führt dieser Link bei wbecLan ins Leere.
Der Einfachheit halber (um keine unterschiedlichen Dateien für das WebInterface zu haben) habe ich aber keine Spezialbehandlung eingefügt. Dies wird auch so bleiben.

closed 2023-07-20 by steff393