steff393/wbec #96: Kommunikationsfehler

myrdrks, 2023-08-29:
Hallo, 
ich habe die WB an einer NodeMCU. Alles korrekt angeschlossen, achtfach geprüft. Alle Pins sind korrekt, 12-fach geprüft.
In der Weboberfläche und auch im JSON stehen korrekte Daten (Energiezähler 128,777 kWh zB). Bei verbunden (bzw. chgStat) steht immer 10 und die Box leuchtet "Kommunikationsfehler" 6x Weiß, 3x Blau.
Das JSON sieht so aus:
`{
wbec: {
version: "v0.4.9",
bldDate: "Aug 29 2023 10:59:55",
timeNow: "12:47:04"
},
box: [
{
busId: 1,
version: "108",
chgStat: 10,
currL1: 0,
currL2: 0,
currL3: 0,
pcbTemp: 250,
voltL1: 234,
voltL2: 237,
voltL3: 234,
extLock: 1,
power: 0,
energyP: 0,
energyI: 128.777,
energyC: 0,
currMax: 16,
currMin: 6,
logStr: "00.779.2885/04 030915 2219 HD_EnergyControl",
wdTmOut: 60000,
standby: 4,
remLock: 0,
currLim: 160,
currFs: 0,
lmReq: 0,
lmLim: 255,
resCode: "0",
failCnt: 0
}
],
modbus: {
state: {
lastTm: 2009114,
millis: 2015796
}
},
rfid: {
enabled: false,
release: false,
lastId: ""
},
pv: {
mode: 0,
watt: 0
},
wifi: {
mac: "50:02:91:79:4A:D3",
rssi: -50,
signal: 100,
channel: 11
}
}`
Was läuft hier verkehrt?

steff393, 2023-08-29:
Hallo @myrdrks,
ich vermute du hast die Wallbox per Remote lock gesperrt, oder? Das würde das Verhalten erklären:
https://github.com/steff393/wbec/wiki/Modbus-Register

Viele Grüße
Stefan

myrdrks, 2023-08-29:
Ganz genau danach sieht es aus. Aber ich habe nie selbst remote lock aktiviert. Kann man über das Webinterface Register setzen?

steff393, 2023-08-29:
Ja, in diesem Fall mit http://wbec.local/json?remLock=1

myrdrks, 2023-08-29:
Das war es! Danke dir für deinen Support!

closed 2023-08-29 by steff393