steff393/wbec #97: Ladestrom Einstellung vs. Realität

marc107, 2023-09-01:
Hallo zusammen,

zunächst einmal vielen Dank @steff393 für die Übersendung der Platine in 2022 und die tolle Arbeit im Projekt. ;-)

Nach einigen Startproblemen mit der Verbindung zur Heidelberg konnte ich nunmehr eine erfolgreiche Verbindung aufbauen.
Wbec läuft in nachfolgender Config:

```
{"wbec":{"version":"v0.4.9",
"bldDate":"May 28 2023 18:43:20",
"timeNow":"17:49:16"},
"box":[{"busId":1,
"version":"108",
"chgStat":7,
"currL1":32,
"currL2":31,
"currL3":32,
"pcbTemp":299,
"voltL1":238,
"voltL2":237,
"voltL3":236,
"extLock":1,
"power":2315,
"energyP":1.087,
"energyI":586.821,
"energyC":1.087,
"currMax":16,
"currMin":6,
"logStr":"00.779.2909/02 025013 2150 HD_EnergyControl",
"wdTmOut":60000,
"standby":4,
"remLock":1,
"currLim":160,
"currFs":0,
"lmReq":160,
"lmLim":255,
"resCode":"0"}],
"modbus":{"state":{"lastTm":2768867,"millis":2771081}},
"rfid":{"enabled":false,
"release":false,
"lastId":""},
"pv":{"mode":0,
"watt":0},
"wifi":{"mac":"44:17:93:0E:xx:53",
"rssi":-78,
"signal":44,
"channel":6}}
```

Stelle ich nunmehr den Ladestrom in wbec auf 6A, so wird laut MQTT von wbec nacfolgendes ausgegeben:
```
power = 1394
currL1 = 1.9
currL2 = 1.9
currL3 = 1.9
```
Erhöhe ich in wbec den Ladestrom auf 16A, so ergibt sich folgende Ausgabe via MQTT:
```
power = 2316
currL1 = 3.3
currL2 = 3.1
currL3 = 13.2
```

Ich stelle mir die Frage, warum der Strom nicht auf 16 A erhöht wird. Der Akku des Fiat 500e hatte zum Zeitpunkt des Ladens ca. 45%.

Inwiefern es relevant ist vermag ich nicht zu beurteilen, aber ich gebe die Werte via MQTT an evcc weiter.
Die evcc.yaml ist wie folgt definiert:
```
# mqtt message broker
mqtt:
  broker: 192.168.178.xx:1883
  # topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

#interval: 60s

meters:
- type: custom
  name: Stromzaehler
  power:
    source: mqtt
    topic: fhem/Strom
    timeout: 30s
    scale: 1
- name: wbec-LP1-Meter
  type: custom
  power:
    source: mqtt
    topic: wbec/lp/1/power
  energy:
    source: mqtt
    topic: wbec/lp/1/energy
  currents:
    - source: mqtt
      topic: wbec/lp/1/currL1
    - source: mqtt
      topic: wbec/lp/1/currL2
    - source: mqtt
      topic: wbec/lp/1/currL3

- type: custom
  name: PV_gesamt
  power:
    source: mqtt
    topic: fhem/PV_gesamt
    timeout: 30s
    scale: 1

chargers:
- name: wbec-LP1
  type: custom
  status:
    source: mqtt
    topic: wbec/lp/1/status
  enabled:
    source: mqtt
    topic: wbec/lp/1/enabled
  enable:
    source: mqtt
    topic: wbec/lp/1/enable
  maxcurrent:
    source: mqtt
    topic: wbec/lp/1/maxcurrent

site:
  title: Heidelberg
  meters:
    grid: Stromzaehler
    pv: PV_gesamt

loadpoints:
- title: Out # display name for UI
  charger: wbec-LP1 # Wallbox Referenz
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true 
  mode: now
```

In evcc werden wiederum völlig andere und falsche Werte angegeben:

[image]

Die Leistung in evcc ändert sich nicht, egal ob im wbec 6A oder 16A eingestellt sind.

Hat jemand einen Erklärungsansatz dafür?

Viele Grüße

Marc

steff393, 2023-09-01:
Hallo Marc,
die Wallbox legt nur den maximal erlaubten Ladestrom fest. Das Auto entscheidet anschließend, wie viel Strom es wirklich zieht.

Es gibt zum einen die Leistungsmessung in der Wallbox ("power"). Andererseits kann evcc die Leistung auch aus Strom und Spannung berechnen. Diese Werte können schon voneinander abweichen (v.a. da evcc die realen Spannungen ja gar nicht sieht, s.o., und vermutlich einen Ersatzwert annimmt). Welcher Wert jetzt aber näher an der Realität ist weiß ich leider nicht.

marc107, 2023-09-01:
Danke für deine schnelle Rückmeldung!

> die Wallbox legt nur den maximal erlaubten Ladestrom fest. Das Auto entscheidet anschließend, wie viel Strom es wirklich zieht.
Ich muss nachher mal im Auto nachsehen. Nicht, dass dort eine Begrenzung von einer Mitfahrerin eingestellt wurde... ;-)

Die Frage der Abweichungen erscheint mir wohl besser im evcc-Forum untergebracht. Konntest du diese auch beobachten?

marc107, 2023-09-01:
Das liest sich ähnlich:
https://github.com/evcc-io/evcc/discussions/5928

JTL42, 2023-09-03:
Hallo Marc,

der Fiat 500e hat 5 Ladestufen, die in ca. 20% Schritten die Ladeleistung festlegen. Bei Stufe 3 beträgt diese also 60%. Du kannst WBEC so konfigurieren, dass er das kompensiert, indem Du in cfgPvPhFactor einen Wert von 44 anstelle 69 einträgst. Bei anderen Ladestufen entsprechend mehr oder weniger. Das ganze hat sogar einen entscheidenden Vorteil. Du brauchst normalerweise 6A um das Laden zu starten, d.h. beim PV Laden würde es etwa mindestens 4,2KW benötigen. Bei Ladestufe 3 kannst Du aber bereits bei 2,4KW laden, weil der Fiat nicht die gesamte Leistung zieht. 

Schau also mal auf die eingestellte Ladeleistung und melde Dich bei weiteren Fragen. Habe ebenfalls einen 500e 😊

Grüße 
JT

marc107, 2023-09-03:
Hi JT,

vielen Dank für die Tipps!
Ich konnte mich am WE ein wenig mit dem Thema auseinandersetzen und musste dabei feststellen, dass im Auto Ladestufe 1 eingestellt ist. In meinem Fall lädt der Fiat dann mit 3x1,9A == 1,4kW. Das finde ich super, da ich nur eine sehr kleine PV-Anlage mit 2kWp besitze und das Auto nur sehr wenig (ca. 3500km/a) bewegt wird.

Ein wenig schade finde ich, dass in evcc der falsche Wert für die Ladeleistung angezeigt wird, da ich noch nicht genau weiß, welchen Einfuss das Ganze auf das Überschussladen nimmt.

Den Ladepunkt habe ich in evcc wie folgt definiert:

```
loadpoints:
- title: Out # display name for UI
  charger: wbec-LP1 # Wallbox Referenz
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true 
  mode: pv
  enable:
    threshold: -750 # evcc startet das Laden, wenn 2 Minuten lang mindestens 750 W Einspeiseleistung (Überschuss) vorhanden sind
    delay: 5m
  disable:
    threshold: 750 # evcc beendet das Laden, wenn 15 Minuten lang mehr als 750 W Netzstrom bezogen wurde
    delay: 15m # Ausschalteverzögerung, Laufzeit dabei mindestens 15 Minuten
```

Ich befürchte, dass die von evcc angenommene Ladeleistung von 3,7kWh nunmehr dazu führt, dass er das Überschussladen unterbricht, da er ggf. denkt, dass er mehr als 750W aus dem Netz bezieht. Wobei der Screenshot dargestellte Wert für den Netzbezug i.H.v. 2,5kW ja stimmt, nur die Ladeleistung halt nicht.
Da es die nächsten Tage schönes Wetter geben wird, sollte das ja herauszubekommen sein.
Wobei mich der optische Mangel irgendwie nervt... ;-)

JTL42, 2023-09-04:
Hi Marc,

ich hatte EVCC nur mal kurzzeitig verwendet, da mir die Oberfläche besser gefallen hatte, bin aber mittlerweile zu der Weboberfläche von WBEC zurückgewechselt. EVCC hat leider nicht zurverlässig den Batteriestand des Fiat 500 abrufen können und es steuert nicht so feingranular wie WBEC. In meiner Config hatte ich allerdings die Werte von der PV Anlage und dem Stromzähler aus den WBEC mqtt Werten genommen:

- name: pv
  type: custom
  power:
    source: mqtt
    topic: wbec/inverter/pwrInv
- name: grid
  type: custom
  power:
    source: mqtt
    topic: wbec/pv/watt

Mit Deinen Ladeeinstellung wirst Du wahrscheinlich nicht glücklich werden. Der Fiat 500e benötigt beim Laden für die Boardelektronik etwa 500 Watt, d.h. bei Ladestufe 1 geht fast die Hälfte der Leistung verloren, d.h. reines Überschussladen ist nicht sehr effizient. Auf Ladestufe 1 hast Du einen Regelbereich von ungefähr 0,8 KW (bei 6 A) bis 2,2 KW (bei 16 A) +/- Ladeelektronik. Deine 2 KWp Anlage dürfte aktuell wahrscheinlich maximal 1,5 KW zur Mittagszeit liefern und ca. 8 KWh am Tag erzeugen. Davon also maximal 4 KWh, die in der Batterie angkommen könnten und davon musst Du noch Eigenverbrauch im Haus abziehen und die Zeit bis Deine Anlage mal mindestens 1 KW produziert, damit das Laden überhaupt beginnt.

Wenn Du also sowieso noch zusätzlich aus dem Netz beim Laden beziehen musst, dann ist es tatsächlich effizienter mit höherer Ladeleistung aber dafür in einer kürzeren Zeit zu laden, damit Du die 500 Watt für die Elektronik nicht zu lange hast.

Die 750 Watt Grenze für den Netzbezug in Deiner Konfiguration hast Du in der Regel sehr schnell erreicht und das führt dazu, dass die Ladung sehr häufig an und wieder ausgeschaltet wird, das ist für den Akku und die Elektronik eher ungünstig.

Folgendes könnte bei Dir funktionieren. Such Dir ein Zeitfenster raus, bei dem der PV Strom an einem guten Tag über 800 Watt liegt, vermutlich 10 - 16 Uhr. Das wäre der präferierte Zeitraum. Stelle die Ladestufe im Fiat etwas größer ein, also 2 oder 3, abhängig davon dann cfgPvPhFactor in WBEC. Zusätzlich cfgPvLimStart auf den A Wert, bei dem bei der gewählten Ladestufe die 800 Watt erreicht werden, cfgPvLim Stop auf etwa die Hälfte.  Zusätzlich cfgPvMinTime auf 120 Minuten.

Wenn Du als Lademodus jetzt "PV" wählst, erreichst Du folgendes: Das Laden des Autos erfolgt erst, wenn Du 800 Watt von der PV bekommst (also zumindest ein halbwegs sonniger Tagesstart). Abhängig von der Ladestufe wird er ber dann mit höherer Leistung laden und den Rest aus dem Netz beziehen. Durch cfgPvMinTime läuft er mindestens 2 Stunden durch, selbst wenn die Sonne zwischendurch mal weg ist und er regelt die Ladeleistung immer ein wenig nach, falls mehr oder weniger Sonne verfügbar ist. Also einfach mal die Parameter ein wenig ausprobieren, ggfs. auch mit cfgPvOffset einfach einen absoluten Wert anstelle eines prozentualen Faktors bei cfgPvPhFactor einstellen.

Das Ganze erlaubt es Dir das Auto einfach immer an der Wallbox angesteckt zu lassen und nur zu laden, wenn von der PV der Schwellwert erreicht wird. Ist zumindest etwas smarter als feste Zeiten zu programmieren. 

Hoffe das hilft, oder gibt zumindest ein paar Ansätze für Lösungen.

Viele Grüße
JT

closed 2023-09-16 by steff393

marc107, 2023-09-21:
Hallo JT,

erstmal sorry für die späte Rückmeldung, aber mir ist die Benachrichtigung über eine neue Antwort in meinen Mails durchgegangen... :-(

Tatsächlich war die schicke GUI ausschlaggebend für EVCC, somit ging es uns (zunächst) beiden so.

Die Ansätze von dir sind in der Tat sehr wertvoll, vielen Dank dafür! 
Die Information mit den 500W Leistung für die Bordelektronik war mir neu, das ist in der Tat ein nicht zu unterschätzender Faktor.

Da ich deine Antwort erst jetzt gelesen habe und aufgrund der Tatsache, dass die Sonne in der letzten Zeit sehr viel geschienen hat, konnte ich Erfahrungen mit meinen ungünstigen Einstellungen sammeln. Nur aufgrund der Tatsache, dass der 500e ein Zweitwagen ist und entsprechend wenig bewegt wird, kann man mit dieser langsamen Ladeeinstellung überhaupt "arbeiten. Es kommt tatsächlich nicht viel an, aber für mein Nutzungsszenario reicht es gerade so aus.
Laut EVCC nutze habe ich aktuell ca. 55% Sonne in mein Auto getankt. Für die Größe der Anlage eigentlich kein so schlechter Wert, deine Annahmen zu den Leistungswerten waren recht treffend (1,3kW im Peak, 8-9 kWh pro Tag).

Bei der nächsten Schönwetterperiode werde ich deine vorgeschlagenen Einstellungen testen! :-)

Beste Grüße

Marc