steff393/wbec #99: Web Interface x.5.0

steff393, 2023-09-10:
Es gibt vereinzelte Berichte, dass im Web Interface keine aktuellen Daten angezeigt werden (nur Striche "-") und auch die Navigation über die Buttons nicht funktioniert.
Dies deutet darauf hin, dass die JavaScript-Skripte nicht vollständig geladen wurden, bzw. abbrechen. 

Aktuell ist das Verhalten bei mir noch nicht nachvollziehbar.

Die Neuerung die damit zu tun haben könnte ist, dass einige Javascript-Funktionen in Datei common.js verschoben wurden. Die HTML-Seiten müssen jetzt 2 .js-Dateien laden: Die common.js und die für die jeweilige Seite.

Bislang sehe ich als Workaround nur, auf die Dateien aus dem Release x.4.9 zurückzugehen:
https://github.com/steff393/wbec/releases/download/v0.4.9/data.zip

steff393, 2023-09-10:
Mögliche Ursache: Die folgenden neuen Dateien wurden nicht per "Durchsuchen.." + "Upload" hochgeladen:
- common.js
- time.html
- time,js

Diese sind jedoch zwingend erforderlich sobald die übrigen Dateien aktualisiert wurden.

hbpv, 2023-09-16:
Hi

habe die Dateien drin, dennoch alle Felder leer. Komischerweise tauchten kurz nach dem anstecken des Autos die Werte auf, inzwischen wieder weg. PV  frontend fehlt auch komplett.

common.js etc sind alle da..

Auto lädt mit dem Max was ich grade zu Beginn noch einstellen konnte, jetzt wiegesagt alles auf 0/-

viele Grüße 

[image]

[image]

hbpv, 2023-09-16:
Update… wartet man auf der leeren Seite bis zu 30 Sekunden, kommen die Werte. Es scheint einfach ewig zu laden… (der js oder wer auch immer im Hintergrund)… 
[image]

ThoHilde, 2023-09-17:
Ich hatte ebenfalls das Problem, dass unter 0.5.0 nur Striche '-' angezeigt wurden. Zurück auf die Dateien aus dem 0.4.9 Data-Ordner hat geholfen. Die Seiten bauen sich instant auf und es werden Werte angezeigt.
Mit den 0.5.0 Dateien wurde mir, zumindest am PC, angezeigt, dass die Wallbox offline sei. Ebenso wurde ein auswählbares Feld mit Wallbox1 oberhalb des Ladestroms angezeigt. Das ist nun nicht mehr der Fall und war auch nicht am Smartphone so. Keine Ahnung, ob das damit ggf. zusammenhängen kann.

steff393, 2023-09-17:
Die Anzeige, dass die Wallbox offline sei, ist auch ein Indicator dafür, dass die beiden JavaScript-Dateien noch nicht geladen wurden. Diese Anzeige wird erst durch das Script "versteckt".

Man kann die "defer"-Attribute noch aus allen .html-Dateien löschen, da sie bei `type="module"` unnötig sind:
ALT:
```
  <script src="common.js" type="module" defer></script>
  <script src="web.js"    type="module" defer></script>
```
NEU:
```
  <script src="common.js" type="module"></script>
  <script src="web.js"    type="module"></script>
```

Die aktuellen Dateien im data-Ordner habe ich heute umgestellt, so dass auch folgender Workaround möglich wäre:
- HTML-Dateien: Löschen der Zeile `<script src="common.js" type="module"></script>` und des `type="module"` in der Zeile darunter
- JS-Dateien: Ersetzen von `import {<Funktionsname>} from './common.js';` durch die jeweiligen Funktionen (Copy & Paste) aus der common.js
- Die Datei common.js kann anschließend auch gelöscht werden (optional, per Rechtsklick)

Die betroffenen HTML/JS-Dateien sind cfg, log, time und web.
Somit sollte zumindest jeder mit relativ einfachen Anpassungen in der Lage sein, die 0.5.0er Funktionen zu nutzen.

wexman, 2023-09-19:
Ich habe das Problem auch, probiert unter Edge / Firefox, Windows & Linux (Mint) & Android

ThoHilde, 2023-09-19:
Bei der 0.5.0 mit den Data-Files 0.4.9 hatte ich mehrer Ladeabbrüche. Bin nun ebenfalls komplett auf die 0.4.9 zurück. Bisher scheint alles zu funktionieren.

jochenpetry, 2023-10-01:
Bei mir lädt die Weboberfläche bei 0.5.0 auch gar nicht, nur teilweise bzw. nur nach etlichen Neuaufrufen. Verschiedene Browser, WLAN-Kanäle, ESP reset...hat alles nix gebracht. Der Zugriff auf .../edit bzw. .../update ist schnell, so dass ein Dow grade auf 0.4.9 noch over the air möglich ist

istmichseegal, 2023-10-04:
Ich habe dasselbe Problem, aber ausschlieslich auf dem Smartphone (Firefox, Android). Auf der Rechner (Firefox, Windows) geht es sofort und immer.

steff393, 2023-10-13:
Mittlerweile sind mehr als 300 Leute auf die x.5.0 umgestiegen (wenn man mal nach den Downloadzahlen geht). 
Gibt es jemand, der 
- der Version 1.5.0 oder 2.5.0 nutzt **und**
- den Workaround aus meinem letzten Kommentar bereits angewandt hat (data_new.zip & Verschieben der Funktionen aus der common.js in die Einzeldateien) **und**
- immer noch längere Ladezeiten bei einer der .html-Seiten feststellt?

Wenn ja, dann wäre eine Aufzeichnung mit den Entwicklerwerkzeugen der Browser vielleicht noch eine Möglichkeit, um der Ursache auf die Spur zu kommen:
- Firefox: F12 --> Netzwerkanalyse
- Edge: F12 --> Netzwerk

Velines, 2023-10-15:
Ich habe das Problem, nachdem ich heute mal aktualisiert habe. wbec-0.5.0 mit Files data-new.zip.
OS ist macOS 12.7, Browser in diesem Fall Chrome. Unter Firefox hängt es genauso, Safari funktioniert aber.

[wbec.fritz.box.har.zip](https://github.com/steff393/wbec/files/12909742/wbec.fritz.box.har.zip)

MichaelH1981, 2023-10-16:
Ich habe den Workaround heute angewendet und wbec läuft mit der data_new.zip jetzt, bis auf die Zeitfunktionen unter Safari, sowie unter Edge. Vor dem Workaround habe ich keine funktionierende Weboberfläche zustande gebracht. 
--> In der time.js fehlte die Funktion setClass, obwohl die ursprüngliche Importfunktion diese nicht angefordert hat. 
Leider sind meine Fähigkeiten in diesem Bereich extrem eingerostet...

Nach dem Hinzufügen der Funktion lässt sich die Oberfläche in der time.html bedienen, allerdings kommen immer 404 Not Found Error beim Aufrufen von wbec.local/time?id=0 und damit läuft die ganze Funktion nicht.

steff393, 2023-11-26:
Habe mittlerweile von einem Nutzer einen Log bekommen, danke dafür!
Es sieht so aus, als wären es zu viele Anfragen an den ESP in kurzer Zeit: web.html, web.js, web.css und dann auch noch die common.js.
Daher habe ich die common.js wieder entfernt und stattdessen die Funktionen in die einzelnen Module kopiert.

Die Dateien sind im Ordner 'data' aktualisiert und auch hier abgelegt: 
https://github.com/steff393/wbec/releases/download/v0.5.0/data_fix.zip

Wie gehabt muss man die Dateien in der data_fix.zip mittels /edit einzeln auf wbec laden. Die Datei common.js kann unter /edit auch mittels Rechtsklick->Delete gelöscht werden.

closed 2024-03-24 by steff393

VedMod, 2025-02-11:
> Die Dateien sind im Ordner 'data' aktualisiert und auch hier abgelegt: https://github.com/steff393/wbec/releases/download/v0.5.1/data.zip
> 

Gibt es die auch für die aktuelle x.5.3? Bzw. kann man dafür nicht ein script schreiben, dass dies automatisch für jede neue Version tut?

BSS-HA, 2025-03-08:
Servus zusammen, aktuell habe ich auch das problem bei 2.5.4 das mein Safari, Chrom, Firefox usw die Seite von WBEC nicht laden kann bzw keine Werte angezeigt werden. Auch die Buttons funktionieren nicht. Über den Ping Befehl erreiche ich den WBEC. Config wenn man der Darstellung glauben kann ist leer. Warum auch immer. Gibt es Lösungen zu diesem Thema?

BSS-HA, 2025-08-07:
Bei mir hat bis heute auch noch nichts geändert. Über Safari geht es garnicht. Firefox nur sehr sehr eingeschränkt. Etwas nervig

abuesen, 2025-08-08:
Bei mir hat es auf Mobilgeräten noch nie funktioniert. Eventuell habe ich den Fehler gerade gefunden.

Mir scheint der sendBuffer zu klein. Dadurch wird das JSON abgeschnitten, nicht vollständig geladen oder ungültig. Gerade safari und chrome (mobile) sind sehr empfindlich, was content-länge und Abschluss der Verbindung betrifft. Deshalb sollte man ausführliche Content-Header setzen. Da habe ich aber noch nicht nach geschaut.

In der [webSocket.cpp](https://github.com/steff393/wbec/blob/master/src/webSocket.cpp) wird  `#define JSON_LEN 256` gesetzt. Später dann im loop
```
StaticJsonDocument<JSON_LEN> data;
...
char response[JSON_LEN];
serializeJson(data, response, JSON_LEN);
webSocket.broadcastTXT(response);
```

Mein JSON ist jedoch für eine Box bei 809 byte.

Zum Versuch mal `#define JSON_LEN 1024` oder auch 2048 nehmen. Dann müsste es schon besser werden.

steff393, 2025-08-13:
Guter Hinweis, danke!
Allerdings eine kleine Korrektur: Für die webSocket-Anbindung wird nicht das vollständige http://wbec.local/json gesendet, sondern nur eine kleinere Variante (und auch immer nur für eine einzelne Wallbox):
https://github.com/steff393/wbec/blob/674b324565eb7efe57048f5213fd5df3e2d5bac9/src/webSocket.cpp#L66-L75

Habe diese gerade mit dem [offiziellen Rechner](https://arduinojson.org/v5/assistant/) mit Extremwerten geprüft und komme damit auf max. 293 Byte. Also in der Tat zuviel. Leider.

Ein `#define JSON_LEN      512` sollte daher reichen.

Nun ist es aber noch so: Bei den Fertigmodulen musste ich bereits im März 2024 im Zuge der Einführung von §14a EnWG zusätzliche Werte in das JSON einfügen und habe damit (ohne es genauer auszurechnen) den Puffer auf 512 erhöht. In den Versionen ab 1.5.2 und 2.5.2 sind ist also bereits JSON_LEN=512 umgesetzt. Dort ist die maximale Länge (laut Rechner) 347 Byte => Passt.

Damit könnte für die Fertigmodule, **die älter als x.5.2 sind**, ein Update auf die aktuellste Version helfen (falls der Buffer die alleinige Ursache ist).
Im Github-Code werde ich die 512 bei nächster Gelegenheit einbauen.

Was mich an der Theorie aktuell noch ein wenig zweifeln lässt ist, dass es mit JSON_LEN 256 lt. dem Rechner eigentlich niemals funktionieren dürfte (auch nicht bei minimal kurzen Dateninhalten, z.B. 0) => ich muss das nochmal genauer testen

abuesen, 2025-08-17:
So, nun habe ich 2.5.4 drauf. Damit funktioniert es öfter als mit 2.5.2, aber im Grunde nicht benutzbar. Der Zustand ist auf mobilgeräten seit mindestens knapp 2 Jahren so, so lange wie ich den wbecLAN habe. Deshalb nutzt ich den wbec inzwischen nur noch als gateway zu evcc. 

Es liegt wirklich vermutlich nicht am json selbst, denn die Navigation klappt dann auch nicht und ich hatte gerade mehrfach den Effekt, dass auch kein css geladen wurde. Screenshot kommt gleich.
Interessant: auch ohne css wurden gerade ab und an Werte angezeigt.

Ich finde weiterhin, für smartphone und tablet (Android) ist es kaputt.

abuesen, 2025-08-17:
Hier noch die Fotos. Alle mit Firefox. Das ohne CSS kommt selten vor, mit CSS ohne Werte ist normal.

[image]
[image]
[image]

BSS-HA, 2025-09-16:
Ich hab es jetzt über HomeAssist gelöst. Zumindest für mich. Ich arbeite mit der Web Oberfläche garnicht mehr. Alles wird in HomeAssist dargestellt was in der Weboberfläche vorher war. Das läuft zuverlässig.