steff393/wbec #101: Probleme beim Flashen eines Lolin D1 Mini Pro

moddmylife, 2023-09-11:
Hallo,
wenn ich versuche über Plattformio einen D1 Mini Pro zu flashen (egal ob 0.5.0 oder 0.4.9), Startet dieser nach dem Flashen nicht. 
Nach dem errechnen und anpassen der Monitoring-Baudrate bekomme ich folgende anzeige:

rst cause 3, boot mode (3,6)
[image]

was könnte der grund sein? Env und Board habe ich angepasst.

`Processing d1_mini_pro (platform: espressif8266@4.0.1; board: d1_mini_pro; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_pro.html
PLATFORM: Espressif 8266 (4.0.1) > WeMos D1 mini Pro
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- modbus-esp8266 @ 4.1.0
|-- ESP Async WebServer @ 1.2.3
|-- ArduinoJson @ 6.21.3
|-- PubSubClient @ 2.8.0
|-- NTPClient @ 3.2.1
|-- MFRC522 @ 1.4.10
|-- RTCVars @ 0.1.1
|-- WebSockets @ 2.4.1
|-- AsyncElegantOTA @ 2.2.7
|-- Bounce2 @ 2.71.0
|-- LittleFS @ 0.1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoOTA @ 1.0
|-- WiFiManager-asyncwebserver
|-- EspSoftwareSerial @ 6.12.7
|-- ESP8266WiFi @ 1.0
|-- ESP8266HTTPClient @ 1.2
|-- SPI @ 1.0
|-- ESPAsyncTCP @ 1.2.2
Building in release mode
Compiling .pio\build\d1_mini_pro\src\button.cpp.o
Compiling .pio\build\d1_mini_pro\src\globalConfig.cpp.o
Compiling .pio\build\d1_mini_pro\src\goEmulator.cpp.o
Compiling .pio\build\d1_mini_pro\src\inverter.cpp.o
Compiling .pio\build\d1_mini_pro\src\loadManager.cpp.o
Compiling .pio\build\d1_mini_pro\src\logger.cpp.o
Compiling .pio\build\d1_mini_pro\src\main.cpp.o
Compiling .pio\build\d1_mini_pro\src\mbComm.cpp.o
Compiling .pio\build\d1_mini_pro\src\mqtt.cpp.o
Compiling .pio\build\d1_mini_pro\src\phaseCtrl.cpp.o
Compiling .pio\build\d1_mini_pro\src\powerfox.cpp.o
Compiling .pio\build\d1_mini_pro\src\pvAlgo.cpp.o
Compiling .pio\build\d1_mini_pro\src\rfid.cpp.o
Compiling .pio\build\d1_mini_pro\src\shelly.cpp.o
Compiling .pio\build\d1_mini_pro\src\webServer.cpp.o
Compiling .pio\build\d1_mini_pro\src\webSocket.cpp.o
Generating LD script .pio\build\d1_mini_pro\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\d1_mini_pro\libe8c\ESP8266WiFi\BearSSLHelpers.cpp.o
Compiling .pio\build\d1_mini_pro\libe8c\ESP8266WiFi\CertStoreBearSSL.cpp.o
Compiling .pio\build\d1_mini_pro\libe8c\ESP8266WiFi\ESP8266WiFi.cpp.o
In file included from src\webServer.cpp:4:
.pio\libdeps\d1_mini_pro\AsyncElegantOTA\src/AsyncElegantOTA.h:4:2: warning: #warning AsyncElegantOTA.loop(); is deprecated, please remove it from loop() if defined. This function will be removed in a future release. [-Wcpp]
    4 | #warning AsyncElegantOTA.loop(); is deprecated, please remove it from loop() if defined. This function will be removed in a future release.
      |  ^~~~~~~
Compiling .pio\build\d1_mini_pro\libe8c\ESP8266WiFi\ESP8266WiFiAP.cpp.o
[... 189 lines cut ...]
Compiling .pio\build\d1_mini_pro\FrameworkArduino\stdlib_noniso.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\time.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\uart.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\umm_malloc\umm_info.c.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\umm_malloc\umm_integrity.c.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\umm_malloc\umm_local.c.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\umm_malloc\umm_malloc.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\umm_malloc\umm_poison.c.o
C:\Users\MKA\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\mmu_iram.cpp:26:34: note: '#pragma message: ICACHE size 16K'
   26 | #pragma message("ICACHE size 16K")
      |                                  ^
Archiving .pio\build\d1_mini_pro\libFrameworkArduino.a
Linking .pio\build\d1_mini_pro\firmware.elf
Retrieving maximum program size .pio\build\d1_mini_pro\firmware.elf
Checking size .pio\build\d1_mini_pro\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=======   ]  66.9% (used 54784 bytes from 81920 bytes)
Flash: [========  ]  83.1% (used 867852 bytes from 1044464 bytes)
Building .pio\build\d1_mini_pro\firmware.bin
Creating BIN file ".pio\build\d1_mini_pro\firmware.bin" using "C:\Users\MKA\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and ".pio\build\d1_mini_pro\firmware.elf"
========================================================================================= [SUCCESS] Took 81.88 seconds =========================================================================================
 *  Terminal will be reused by tasks, press any key to close it. 
`

Irgendwelche Ideen?

steff393, 2023-09-16:
wbec ist nicht für den D1 Mini Pro ausgelegt.

closed 2023-09-16 by steff393

joschro, 2024-09-26:
die wbec Firmware läuft bei mir auf einem Wemos D1 Mini Pro anstandslos. Versuch's mal mit der fertigen Firmware von Stefan und z.B. dem esptool:
esptool --port /dev/ttyUSB0 --baud 115200 write_flash --flash_size=4MB -fm dio 0 wbec_firmware_0_5_3.bin
(unter Linux bei mir, evt. ist bei Dir der Port ein anderer)