steff393/wbec #105: Abstürze mit 0.5.0

maxkr2, 2023-10-07:
Hallo Stefan,
ich habe mit der aktuellen Version das Problem, dass der ESP mehrmals am Tag abstürtzt und teilweise die Kommunikation mit der (aktuell einzigen) Wallbox erst wieder funktioniert, wenn ich den ESP von der Spannungsversorgung trenne und erneut verbinde. Nicht jeder Absturz führt aber zum Verlust der Kommunkation, aber im HomeAssistant fällt es mit auf, dass er den kWhCounter selbst ohne Ladevorgang aufaddiert (da er zwischendrin 0 ist).

Habe erst einmal die 0.4.9 installiert.

Vielleicht hast du eine Idee.

Danke & viele Grüße
Max

hbpv, 2023-10-08:
Same here!
Teilweise lassen sich Werte nicht nach HA ziehen, was du zu „nicht verfügbar“ Stati in HA führt, was wiederum automatisierungen in HA in Probleme bringt. Beispielsweise frage ich den PV Mode/Lademodus ab und setze dann zb den Speicher DOD auf 2%, um das entladen des Speichers zu verhindern. Kommt dort aber kein/falscher Wert, hängen diese Automatisierungen.

Das ganze scheint unabhängig von dem anderem Problem zu sein, das ich Ladezeiten von 10-15 Sek. + vom webinterface seit 5.0 habe, was anstrengend ist. UND: der Energiezähler wurde mir genullt ( von plausiblen 3xxx kWh) und wird nun jedes Mal genullt statt aufaddiert.

Liebe Grüße

maxkr2, 2023-10-13:
Muss leider feststellen, dass das Downgrade auf 0.4.9 und das kurrzeitige Ausschalten um Mitternacht keine Verbesserung gebracht hat: Heute werden mir wieder 19MW von der Wallbox angezeigt, da diese 16 Mal 0 und dann wieder 1198KWh übertragen hat.
@hbpv - wie hast du genullt - im HA oder am WBEC?

steff393, 2023-10-13:
Hi,
woran macht ihr den Absturz fest? Sieht man wirklich, dass der Wert von millis (=Millisekunden seit Reset) entsprechend niedrig ist?
@hbpv: Welche Version nutzt du?

Was steht bei euch unter /json bei "standby"?

hbpv, 2023-10-13:
> Hi, woran macht ihr den Absturz fest? Sieht man wirklich, dass der Wert von millis (=Millisekunden seit Reset) entsprechend niedrig ist? @hbpv: Welche Version nutzt du?
> 
> Was steht bei euch unter /json bei "standby"?

Hi !

Ich nutze 0.5.0 mit Pro (Wifi). Standby in json=0.
Siehe Video etwa 10 Sekunden Ladezeit. Bei jedem reload, nicht nur initial…

hbpv, 2023-10-13:
> Muss leider feststellen, dass das Downgrade auf 0.4.9 und das kurrzeitige Ausschalten um Mitternacht keine Verbesserung gebracht hat: Heute werden mir wieder 19MW von der Wallbox angezeigt, da diese 16 Mal 0 und dann wieder 1198KWh übertragen hat. @hbpv - wie hast du genullt - im HA oder am WBEC?

Ich hatte kurz nach Update auf 0.5.0 seitens der  sowas, siehe Foto, letzte oberste Ladung ;-))). Nach dem nächsten /reset war alles auf 0 :-(
[image]

maxkr2, 2023-10-13:
@hbpv Alles auf 0 wollte ich vermeiden.. 
Bei mir ist Standby 4, und vor etwa 75min hat sich der Wert im HA erhöht, millis steht bei 1518832.

steff393, 2023-10-13:
Hi Leute,
sorry, aber das geht mir hier zu durcheinander. Um vernünftig zu einer Lösung zu kommen brauche ich von euch mehr Informationen:
- cfg.json
- Die Ausgabe von http://wbec.local/json
- Euer Setup (nutzt ihr z.B. andere Funktionen wie PV-Laden, Powerfox, ...)
- Wie ist die Anbindung an Home Assistant (MQTT, HTTP, ...)?
- Wie lange seht ihr das Verhalten? Schon immer? Seit dem Update?
- Löst ihr einen http://wbec.local/reset aus (dann ist logisch, dass viele Werte erst mal 0 sind)?

=> Schickt mir das am besten an wbec393@gmail.com

closed 2023-10-13 by steff393