steff393/wbec #106: wbec EnergieZähler Wert nach Stromausfall weg

csh-de, 2023-10-15:
wbec EnergieZähler Wert nach Stromausfall weg.

lese den Wert immer aus und speichere ihn in eine Datenbank, über Grafana sehe ich dann alles, nach einem Stromausfall ist der Wert von 844kwh auf 315kwh gefallen, ist der Wert aus der Wallbox oder speichert die wbec Software den Wert?
Kann man da Eingreifen? Offset oder ähnliches? Warum speichert sich der Wert nicht nach Stromausfall?

Danke und sonst läuft sie perfekt!!

steff393, 2023-10-15:
Hi,
die Wallbox speichert den Wert nur beim Übergang in den Standby dauerhaft ab.
wbec liest nur den Wert aus der Wallbox.
Man kann einen Wert cfgEnergyOffset **abziehen**. Allerdings darf cfgEnergyOffset derzeit **nur positiv** sein. Somit hilft das in dieser Situation nicht.

Es gäbe aber einen mehr oder weniger komplizierten Umweg:
Man kann bewusst einen Overflow erzeugen, indem man folgende Darstellung negativer Zahlen nutzt:
-1 W = 65535
-2 W = 65534
-3 W = 65533
...
-32768 W = 32768
=> Somit kann z.B. mit "cfgEnergyOffset":65533 erreichen, dass 3 W dazuaddiert werden.

closed 2023-10-20 by steff393

csh-de, 2023-11-01:
ah ok,

ist es möglich die Box einmal die Woche in den Standby zu fahren?
Wie kommt sie da wieder raus? wenn man ein Auto anschließt?

Grüße

------------------------------------------------------------------------------------------------------ 

From: steff393 
Sent: Sunday, October 15, 2023 11:34 AM
To: steff393/wbec 
Cc: csh-de ; Author 
Subject: Re: [steff393/wbec] wbec EnergieZähler Wert nach Stromausfall weg (Issue #106)

Hi,
die Wallbox speichert den Wert nur beim Übergang in den Standby dauerhaft ab.
wbec liest nur den Wert aus der Wallbox.
Man kann einen Wert cfgEnergyOffset abziehen. Allerdings darf cfgEnergyOffset derzeit nur positiv sein. Somit hilft das in dieser Situation nicht.

Es gäbe aber einen mehr oder weniger komplizierten Umweg:
Man kann bewusst einen Overflow erzeugen, indem man folgende Darstellung negativer Zahlen nutzt:
-1 W = 65535
-2 W = 65534
-3 W = 65533
...
-32768 W = 32768
=> Somit kann z.B. mit "cfgEnergyOffset":65533 erreichen, dass 3 W dazuaddiert werden.