steff393/wbec #107: WBEC wechselt Ladestärke über EVCC  nicht zuverlässig

apos-t, 2023-10-19:
Hallo,

ich habe folgendes Problem, wenn ich als letzten Lademodus in EVCC FAST hatte also 16A und beim nächsten Ladevorgang  im Modus PV oder min. + PV die Ladung beginne - mit normalerweise 6A. Beginnt WBEC die Ladung mit 16A und es findet auch keine Reduzierung der Ladestärke statt. Abhilfe schafft nur in EVCC in den Fast Modus zu wechseln und anschließend wieder in PV bzw. Min. + PV. EVCC sendet nach meinen Logs sauber die 6A über MQTT beim Start des Ladevorgangs, WBEC scheint die Einstellung zu Beginn des Ladevorgangs jedoch zu verschlucken. 

Einstellungen sehen wie folgt aus:
{"cfgApPass":"XXXXXXX","cfgCntWb":1, "cfgLoopDelay":10, "cfgMqttIp":"192.168.X.X","cfgMqttLp":[1], "cfgStandby":0, "cfgNtpServer":"fritz.box" }

Viele Dank

steff393, 2023-10-20:
Hi, kannst du mir bitte das Log an wbec393@gmail.com schicken?
Ein Hinweis zum Standby. Wenn man diese Funktion nutzt (standby=0), und die Wallbox im Standby-Schlaf ist, dann gehen Kommandos von EVCC ins Leere. wbec kann die Wallbox nicht per Modbus aufwecken.

apos-t, 2023-10-23:
Hi,

Log ist raus -  Log ist vom zweiten Durchlauf - Ablauf beide mal gleich. In evcc von off auf Fast 16A den Ladevorgang gestartet. Abstand ca. 5 min, Ladevorgang startet jedoch immer mit 6A. EVCC schickt in dem Fall nur einmal das Kommando 16A. Sobald ein Ladevorgang aktiv ist und anschließend die Ladestärke wechselt z.B. bei PV oder Min+PV funktioniert die Ladestromanspassung ohne Probleme nur beim das jeweils erste Kommando wird scheinbar nicht verarbeitet

steff393, 2023-11-07:
Lösung:
Callback für wbec/lp/x/maxcurrent
Zwischen den folgenden beiden Zeilen fehlt ein `maxcurrent[i] = (uint8_t)val`
https://github.com/steff393/wbec/blob/638b5a39b79a4a859c88c32bb1672deb73f4dfe0/src/mqtt.cpp#L99-L100

--> Umsetzung im nächsten Release

commit a9ecafcb (2023-11-26): Set maxcurrent, when it was received from evcc (fix #107)
closed 2023-11-26 by steff393