steff393/wbec #110: Wallbox will nicht mehr mit Wbec sprechen

BlackandGold1, 2023-12-12:
Seltsam von heut auf morgen geht nichts mehr: Wbec erreichtbar aber ich kann nichts mehr schalten. Hab die Wallbox nur mit dem currLim=160 angeschaltet b.z.w mit 0 aus. jetzt geht nichts mehr: weiß einer Rat?

Bootlog: 01:00:00: CFG : cfgWbecVersion: v0.5.0
01:00:00: CFG : cfgBuildDate: Sep  1 2023 19:54:44
01:00:00: CFG : cfgCntWb: 1
01:00:03: MB  : HwVersion: 15
01:00:03: RFID: Disabled (rfid.txt not found)
12:25:12: MB  : Comm-Failure BusID 1
12:25:23: MB  : Comm-Failure BusID 1
12:25:35: MB  : Comm-Failure BusID 1
12:25:46: MB  : Comm-Failure BusID 1
12:25:58: MB  : Comm-Failure BusID 1
12:26:09: MB  : Comm-Failure BusID 1
12:26:11: WEBS: {"wbec":{"version":"v0.5.0","bldDate":"Sep  1 2023 19:54:44","timeNow":"12:26:11"},"box":[{"busId":1,"version":"0","chgStat":0,"currL1":0,"currL2":0,"currL3":0,"pcbTemp":0,"voltL1":0,"voltL2":0,"voltL3":0,"extLock":0,"power":0,"energyP":0,"energyI":0,"energyC":0,"currMax":0,"currMin":0,"logStr":"","wdTmOut":0,"standby":0,"remLock":0,"currLim":0,"currFs":0,"lmReq":0,"lmLim":255,"resCode":"e4","failCnt":6}],"modbus":{"state":{"lastTm":62591,"millis":64410}},"rfid":{"enabled":false,"release":false,"lastId":""},"pv":{"mode":0,"watt":0},"wifi":{"mac":"E8:DB:84:C5:FF:69","rssi":-79,"signal":42,"channel":13}}
12:26:20: MB  : Comm-Failure BusID 1
12:26:31: MB  : Comm-Failure BusID 1
12:26:43: MB  : Comm-Failure BusID 1
12:26:54: MB  : Comm-Failure BusID 1
12:26:54: MB  : Timeout BusID 1
12:27:06: MB  : Comm-Failure BusID 1
12:27:18: MB  : Comm-Failure BusID 1
12:27:30: MB  : Comm-Failure BusID 1
12:27:41: MB  : Comm-Failure BusID 1
12:27:53: MB  : Comm-Failure BusID 1
12:28:04: MB  : Comm-Failure BusID 1
12:28:16: MB  : Comm-Failure BusID 1
12:28:27: MB  : Comm-Failure BusID 1
12:28:39: MB  : Comm-Failure BusID 1
12:28:50: MB  : Comm-Failure BusID 1
12:29:02: MB  : Comm-Failure BusID 1
12:29:13: MB  : Comm-Failure BusID 1
12:29:25: MB  : Comm-Failure BusID 1
12:29:36: MB  : Comm-Failure BusID 1
12:29:48: MB  : Comm-Failure BusID 1
12:29:59: MB  : Comm-Failure BusID 1
12:30:10: MB  : Comm-Failure BusID 1
12:30:22: MB  : Comm-Failure BusID 1
12:30:33: MB  : Comm-Failure BusID 1
12:30:45: MB  : Comm-Failure BusID 1

Json:
{
    "wbec": {
        "version": "v0.5.0",
        "bldDate": "Sep  1 2023 19:54:44",
        "timeNow": "12:36:27"
    },
    "box": [
        {
            "busId": 1,
            "version": "0",
            "chgStat": 0,
            "currL1": 0,
            "currL2": 0,
            "currL3": 0,
            "pcbTemp": 0,
            "voltL1": 0,
            "voltL2": 0,
            "voltL3": 0,
            "extLock": 0,
            "power": 0,
            "energyP": 0,
            "energyI": 0,
            "energyC": 0,
            "currMax": 0,
            "currMin": 0,
            "logStr": "",
            "wdTmOut": 0,
            "standby": 0,
            "remLock": 0,
            "currLim": 0,
            "currFs": 0,
            "lmReq": 0,
            "lmLim": 255,
            "resCode": "e4",
            "failCnt": 59
        }
    ],
    "modbus": {
        "state": {
            "lastTm": 672580,
            "millis": 680993
        }
    },
    "rfid": {
        "enabled": false,
        "release": false,
        "lastId": ""
    },
    "pv": {
        "mode": 0,
        "watt": 0
    },
    "wifi": {
        "mac": "E8:DB:84:C5:FF:69",
        "rssi": -78,
        "signal": 44,
        "channel": 13
    }
}

cfg:{
    "cfgApSsid": "",
    "cfgApPass": "",
    "cfgCntWb": 1,
    "cfgMbCycleTime": 10,
    "cfgMbDelay": 100,
    "cfgMbTimeout": 120000,
    "cfgStandby": 4
}

hawa-lc4, 2023-12-12:
Das weiß/blaue blinken der Wallbox deutet daraufhin das auch diese einen Störung am Bus erkennt.
Die WB hast du schon mal aus- und angeschaltet (und dazwischen auch mal gut 10 Sekunden warten)?
Eine Möglichkeit ist natürlich das tatsächlich am Bus etwas kaputt gegangen ist:
 - a) die Bus-Leitung; klemm die Drähte an beiden Seiten mal ab und schließe sie auf einer Seite zusammen, dann mit einem Meßgerät den Widerstand messen; der sollte dann für die beiden Drähte nicht zu hoch sein. Genaue Werte kann ich dir hier nicht sagen denn da müssen die Länge der Leitung und der Leiterquerschnitt einigermaßen genau bekannt sein.
 - b) je nach Länge der Bus-Leitung und deren Verlegung/Schirmung kann es vorkommen das bei starken Änderungen des Ladestromes Störungen am Bus entstehen die auch mal den Bustreiber-Baustein (meist ein MAX485) zerstören. Dann geht gar nichts mehr; das allerdings zu identifizieren erfordert schon etwas Elektronik Kenntnisse.
 - c) die 120 Ohm Abschlußwiderstände sind auf beiden Seiten aktiviert? Wenn nicht kann es gut sein das der Bus eine ganze Zeit problemlos läuft und irgendwann von jetzt auf gleich aufgibt.

BlackandGold1, 2023-12-12:
Ja gefühlt 1000 mal an und aus. Keine Chance. Hab geschirmtes Cat 6e kabel genommen und das läuft so seit 09/21.  Zu  c) was für Widerstände?

hawa-lc4, 2023-12-12:
Ein Cat 6E Kabel ist schon mal ein guter Anfang; wie lang ist es und wo ist es verlegt?
Hast du die beiden Adern eines geschirmten Paares genommen oder Adern aus unterschiedlichen Paaren?
Wurde eventuell auch an der Hardware etwas gemacht, so das eventuell die Adern von "A" und "B" vertauscht wurden?
(Meistens wird es direkt parallel zur Stromversorgungsleitung und dann über längere Strecken verlegt, perfekt für Störungen)
Die Abschlußwiderstände werden in der Wallbox mit dem Schalter S6/2 (auf ON) aktiviert und am RS485 Modul der wbec kann ich es dir nicht sagen, ich weiß nichts über deine Hardware; berichte bitte mal in Stichpunkten darüber.
Da du aber die Version 0.5.0 erwähnt hast schließe ich aus der ersten "0" das es sich um eine Eigenbaulösung handelt.
Der RS485 Bus muß an beiden Enden, jeweils am letzten Gerät (und ich gehe davon aus das es auch hier nur eine point-to-point Verbindung zwischen WB und wbec ist) mit 120 Ohm abgeschlossen sein. Je nach verwendetem RS485 Modul im wbec ist das unterschiedlich einzustellen.
Wenn  elektronisch etwas kaputt gegangen ist kann das sowohl in der WB als auch im Modul des wbec sein, nach meiner Erfahrung aber sehr wahrscheinlich das Letztere.
Wie weit gehen deine Elektronik-Kenntnisse?

hawa-lc4, 2023-12-12:
In der WB aktiviert S6/2 den Abschlußwiderstand, das ist also korrekt und S4/4 auf ON stellt die Bus-ID der WB auf die 1, das ist ebenfalls korrekt. Alle anderen Schalter von S4 sind off.
Da der wbec von Steff ist gehe ich davon aus das auch hier die Bus-Terminierung korrekt eingerichtet ist und bei einer Kabellänge von 30 cm sind Störungen durch die Stromleitung seehr unwahrscheinlich.
Ich kenne das "Innenleben" der Box von steff nicht und kann hier nicht weiter helfen, k.A. ob steff hier Austauschmodule anbietet.
Soweit ich meine Heidelberg WB kenne stört sich die nicht daran wenn gar keine RS485/ModBus Verbindung besteht, nur wenn auf der Fehler auftreten.
Blinkt die WB bei dir munter weiter wenn du die beiden Adern vom Bus komplett abklemmst, egal ob an der WB oder am wbec?

BlackandGold1, 2023-12-12:
[image]
 Ja wenn ich beide Kabel abmachen blinkt sie. Wenn ich die Dip Schalter zurücksetzen und sie neu Starte kann ich auch normal laden.

BlackandGold1, 2023-12-12:
Blaues Klebeband war nur drum für die Markierung  A. CfG Standby war auf 4 hat aber nix gebracht. Ich Messe mal.

BlackandGold1, 2023-12-13:
Hab jetzt noch nen zweiten Webc getestet und  andere Kabel.. genau das gleiche Problem.  Hab jetzt mal ne neue Wallbox bestellt.. bin gespannt.

BlackandGold1, 2023-12-13:
Fehler gefunden. mit anderen ESP geht es.

closed 2023-12-15 by BlackandGold1