steff393/wbec #112: Auto läd nicht.

BlackandGold1, 2023-12-15:
Wbec läuft. Alle Daten sind zu sehen nur Laden geht nicht. Was heist " Verbunden 10 oder chgStat = 10" ? Bei Laden erlaubt ist nen -
Wenn ich die Ladeleistung auf null Stelle bleibt chgStat bei 10.  erst wenn ich das Auto abstecke und wieder anschließe ist chgStat bei 4.

hawa-lc4, 2023-12-15:
Schwer zu sagen was da das Problem ist. Wenn dann bräuchte es da schon die gesamte Ausgabe der /json Informationen.
Am wahrscheinlichsten ist das hier ein remote lock vorliegt; warum der vorliegt oder vorliegen sollte ist so nicht zu erklären.
Schau bitte mal nach was in der json unter "remLock" steht, da sollte eine 1 drin sein. Wenn nicht: Fehler.

Was die Info zu den Charging State angeht so bin ich hier am schnellsten in diesem Forum fündig geworden: [https://forum.iobroker.net/post/752005](url)
Eine Erklärung der ModBus Register der Heidelberg Energy Control habe ich hier: 
[Heidelberger_3_Wallbox_Energy_Control_Externes_Lastmanagement.pdf](https://github.com/steff393/wbec/files/13685859/Heidelberger_3_Wallbox_Energy_Control_Externes_Lastmanagement.pdf)

BlackandGold1, 2023-12-15:
Dein erster Link funktioniert nicht. remlock steht auf 1. Wäre er 0 hätte ich das Blinken.

hawa-lc4, 2023-12-15:
ok, Fehler im ersten Link, jetzt funktioniert's. (Mit copy&paste hätte es vorher auch schon funktioniert ;) )
Die 10 bedeutet einen Code F aber der ist vom Hersteller nicht dokumentiert.
Das heißt das dir vermutlich auf Anhieb keiner sagen kann was die WB hier falsch erkennt, am RS485/ModBus liegt's aber mit großer Sicherheit nicht.
Teil doch mal die komplette json Ausgabe mit uns.

BlackandGold1, 2023-12-15:
{
    "wbec": {
        "version": "v0.4.9",
        "bldDate": "May 28 2023 18:43:20",
        "timeNow": "16:32:27"
    },
    "box": [
        {
            "busId": 1,
            "version": "108",
            "chgStat": 2,
            "currL1": 0,
            "currL2": 0,
            "currL3": 0,
            "pcbTemp": 155,
            "voltL1": 236,
            "voltL2": 237,
            "voltL3": 235,
            "extLock": 1,
            "power": 0,
            "energyP": 0,
            "energyI": 1296.636,
            "energyC": 0,
            "currMax": 16,
            "currMin": 6,
            "logStr": "00.779.2909/01 2115 005142",
            "wdTmOut": 60000,
            "standby": 4,
            "remLock": 1,
            "currLim": 0,
            "currFs": 0,
            "lmReq": 0,
            "lmLim": 255,
            "resCode": "0"
        }
    ],
    "modbus": {
        "state": {
            "lastTm": 356227,
            "millis": 358984
        }
    },
    "rfid": {
        "enabled": false,
        "release": false,
        "lastId": ""
    },
    "pv": {
        "mode": 0,
        "watt": 0
    },
    "wifi": {
        "mac": "E8:DB:84:C5:FF:69",
        "rssi": -78,
        "signal": 44,
        "channel": 13
    }
}

hawa-lc4, 2023-12-16:
Auf den ersten Blick kann ich hier keine Fehlerursache erkennen.
Unterschiede gibt es zu meiner Installation:
- deine WB scheint älter zu sein, der "logStr" ist bei mir anders ("00.779.2909/02 035781 2212 HD_EnergyControl"), was aber nichts zu bedeuten haben sollte.
- den "currFs" habe ich auf 10A eingestellt um bei Ausfall des wbec oder der ModBus Kommunikation weiterhin moderat laden zu können.
- der "chgstat 10" scheint in der Tat eine Fehlermeldung der WB selbst zu sein. Nach meinen Recherchen im web könnte hier ein DC-Fehlerstrom erkannt worden sein. Vergleiche bitte mal das blinken genau mit den Mustern in der Bedienungsanleitung
[Wallbox-Energy-Control-Bedienung-DEU.pdf](https://github.com/steff393/wbec/files/13692327/Wallbox-Energy-Control-Bedienung-DEU.pdf) auf Seite 8, Abbildung 5 oder auch andere Muster.

BlackandGold1, 2023-12-16:
Also die Box verhält sich normal. chgstate ohne Auto = 2, chgStat =4 Auto angeschlossen und currlim =0; Chgstat=10 wenn Currlim=160 ausserdem Blitzt alle 10 Sekunden die weße Leuchte ohne Blau. Dieses  Fehlerbild gibt es so nicht in der Beschreibung.  Dachte an Fehler vom Auto aber woanders lädt es sofort.

closed 2023-12-27 by steff393