steff393/wbec #113: Mit cfgMqttWattJson definiert wird wbec unbenutzbar

TychoGold, 2024-01-07:
Hallo an alle,

ich habe heute Tasmota neu auf meinen Zählerausleser geflasht. Das Topic sieht jetzt folgend aus:
tele/Electric_Energy/SENSOR
{"Time":"2024-01-07T21:46:38","eHZ":{"Power_curr":1759,"Total_in":2614.71,"Total_out":1870.62}}

Wenn ich cfgMqttWattJson auf "Power_curr\":" setze und einen Reset ausführe, wird die GUI nicht mehr angezeigt, mit Glück kann man dann nach gefühlt 20 mal probieren im /edit die Variable wieder auf "" setzen, dann reagiert WBEC wenigstens wieder nach Reset.

Eigentlich lief das schonmal, ich habe nur den Suchtext und den Pfad geändert, jetzt kommt das beschriebene Verhalten.
Ist an dem String irgendwas nicht korrekt?

Viele Grüße und danke für eure Antworten
Tycho

steff393, 2024-01-07:
Ich vermute der Nachrichteninhalt ist zu lang.

TychoGold, 2024-01-07:
Abend Steff,
kann fast sein, da es vorher kürzere Strings waren. Ich probier das aus und gebe Rückmeldung.
Danke.

steff393, 2024-01-07:
Habe sowas ähnliches schon mal beobachtet, aber mir fällt es schwer zu sagen, was die "erlaubte"/mögliche Länge ist.

TychoGold, 2024-01-08:
Guten Morgen,
das scheint leider nicht das Problem zu sein. Mit dem Fantasiewert
{"Time":"1","s":{"Power":100,"In":0.000,"Out":0.000}}
und angepassten "Power\":"
habe ich das gleiche Verhalten. Sogar, wenn ich den Sensor ausschalte und dann den Reset ausführe, kommt die wbec GUI nicht hoch.
Werden irgendwo Logs geschrieben?
Grüße
Tycho

TychoGold, 2024-01-08:
Hallo nochmal,
ich habe das Problem identifiziert: Nicht der eingehende String ist zu lang, sondern der Suchstring macht das Problem. Wenn ich den auf "r\\":" runterbreche, dann funktioniert alles wie es soll.
Der String im Beispiel war eigentlich auch nicht kürzer? Wie dem auch sei - ich schließe den Issue mal. Man sollte aber darüber nachdenken, ob man die Stringsuche anders gestaltet, damit das keine Probleme macht.
Grüße
Tycho

closed 2024-01-08 by TychoGold