steff393/wbec #114: LOG funktioniert nicht

BlackandGold1, 2024-01-08:
Was genau muss ich einstellen damit ich was in der Ladeübersicht sehe?  Wbec 0.5.1

steff393, 2024-01-09:
Das Ladelog funktioniert leider nur mit wbecPro, wbecLan oder wbecPremium (also 1.5.1 oder 2.5.1)

closed 2024-01-09 by steff393

snowy-18, 2024-05-22:
Hallo Stefan, ich hänge mich mal hier dran.
Mein wbecPremium hat ein Jahr lang darauf gewartet angeschlossen zu werden - vor ein paar Wochen bin ich endlich dazu gekommen und alles läuft wunderbar, außer der Ladelog. 
Habe mit 1.5.1 begonnen und bin mittlerweile auf 1.5.2, die Daten des Energiezählers passen, aber der Log bleibt leer.
Hast du eine Idee woran das liegen könnte?

steff393, 2024-05-23:
Hi, hast du 
cfgChargeLog           :"Logbuch der Ladevorgänge: 0:inaktiv, 1:aktiv"
auf 1 gesetzt (und danach einen Reset von wbec durchgeführt)?

snowy-18, 2024-05-23:
Der Wald und die Bäume und so. Danke!