steff393/wbec #115: Wiring Switch on D7

screwb0x, 2024-01-15:
Hi there,
just curious on how to wire the switch to pin D7 as I need a simple and "wife compliant" solution to change PV-charge modes.
Should I just wire the switch in a way that it connects D7 to GND if closed? 
Additional to this a three way switch and another pin of the esp8266 to switch between all of the 3 states of the PV-excess charging would be very nice!
So if this is not that of a big deal correct or add the type of wiring the switch as a note on the wiki page.

ArWi90, 2024-03-22:
Hi @screwb0x,
have you solved this? I found some annotations in the code, in "button.cpp":
- "if RFID is enabled, then it's not possible to use the GPIOs for other purposes"
- "if RFID is not active, then PV_SWITCH can be used to change the PV mode"
- "PIN_PV_SWITCH, INPUT_PULLUP); // USE INTERNAL PULL-UP"

So, internal pull-up is active. The logic indicates, that manual Pull-Down to GND activates the PV mode. Could you give it a try and get back with results?

steff393, 2024-03-23:
Hi, yes, that's right. 
At ESP8266 the Pin D7 needs to be switched to GND. In wbecPro it's G25 to be switched to GND.

Note: In future I want to use one GPIO in wbecPro for a new function: To limit the current to 6A (to cover § 14a EnWG - Netzentgelte für steuerbare Verbrauchseinrichtungen). I still have to see if this will be done with G25 or another GPIO.

closed 2024-03-23 by steff393

screwb0x, 2024-03-29:
Soldered a simple two way switch to D7 which is GPIO13 on NodeMCU and switched it a couple of times but nothing changes on the Webinterface. Then i thought it must have been something with the "if RFID is enabled, then..." comment and searched the settings but found nothing regarding this on the config page. 
So again i'm stuck.

steff393, 2024-03-30:
Have you activated the feature by "cfgBtnDebounce":300 parameter?

screwb0x, 2024-03-30:
No, it is set at standard „0“. Will try it!
But while we’re at it, and maybe it’s obvious, but is this meant for a 2 way switch or a button/trigger like switch? I never really read for what exact switch type it is. And „debouncing“ switches is something i only know from triggers. That’s why i got to this question…

screwb0x, 2024-03-30:
Okay, set debounce to 300, rebooted the ESP and tadaa the switch is working and switches between PV and „aus“.
Thanks for your patience! Issue is definitely closed.
And to be specific: I‘m using a two way switch and no trigger button. Works as intended.