steff393/wbec #116: MQTT password allows only 31 characters

lk3de, 2024-01-21:
Hi,

I am using v0.5.1 and tried to add my MQTT credentials. However, I realized that the parameter `cfgMqttPass` only allows up to 31 characters. My MQTT password has more than that. I assume that only the first 31 characters are stored, which obviously won't work - I'm getting a return code 5 ([MQTT_CONNECT_UNAUTHORIZED](https://github.com/knolleary/pubsubclient/blob/master/src/PubSubClient.h#L54)) :

```
01:00:00: CFG : cfgWbecVersion: v0.5.1
01:00:00: CFG : cfgBuildDate: Jan 20 2024 21:07:32
01:00:00: CFG : cfgCntWb: 1
01:00:03: MB  : HwVersion: 15
01:00:03: RFID: Disabled (rfid.txt not found)
16:48:40: MQTT: Attempting MQTT connection...16:48:40: MQTT: failed, rc=5 try again in 5 seconds
```

After creating a test user on my MQTT broker with a password that has exactly 31 characters, it worked:

```
01:00:00: CFG : cfgWbecVersion: v0.5.1
01:00:00: CFG : cfgBuildDate: Jan 21 2024 17:25:58
01:00:00: CFG : cfgCntWb: 1
01:00:03: MB  : HwVersion: 15
01:00:03: RFID: Disabled (rfid.txt not found)
17:31:25: MQTT: Attempting MQTT connection...connected
17:31:25: MQTT: Publish to openWB/set/lp/1
```

With 32 characters or more, it doesn't work anymore.

I have fixed this for myself by changing the array size of the `cfgMqttPass` parameter to 128. Maybe this can be changed in the official build as well. :)

steff393, 2024-01-24:
Hi, I don't think it's a valid, long-term solution. The maximum password length specified by MQTT is 65535 (!) characters. This is wasted memory. Even if we would increase to 127, then somebody will insist on having a password with 128 characters.

lk3de, 2024-01-25:
You are right, it's more a workaround than a sustainable solution. However, I think that the probability of someone having a password with >127 characters is quite low. Even password generators nowadays create passwords with far less characters (50-64 is what I've seen). Which means that, for the next 5+ years at least, this should be more than enough.

PR #117 (2024-03-24): Extend MQTT password length to 127 chars
Fixes #116
closed 2024-03-24 by steff393