steff393/wbec #118: Einstellungen bei Zeitsteuerung wird nicht gespeichert.

screwb0x, 2024-01-21:
Firmware: wbec_firmware_0_5_1
Hardware:  NodeMCU + Adapterboard von steff393

Alles funktioniert soweit bestens. Aber wenn man über die neuen time.html und time.js eine Ladezeit zum start einer Ladung einstellen möchte kann man auf "Speichern" drücken aber nach einem reload der Seite ist die Einstellung nicht gespeichert. Die Button "Laden" und "Deaktivieren" bewirken ebenfalls nichts.

Schritte zum nachstellen:
Zeitbalken z.B. auf 3:00 bis 5:00Uhr schieben, Maximale Stromstärke auf dem Schieber bis 16A schieben und den Haken für einmalig setzen. "Speichen" klicken, nichts passiert. Nach einem Reload der Seite ist der Ausgangszustand wiederhergestellt und die eingestellten Werte sind verloren.
Firmware mehrfach neu geflashed, die Daten aus dem neuesten data.zip mehrfach erneut hochgeladen. Keine Änderung des Verhaltens.

steff393, 2024-01-24:
Hi, die Zeitsteuerung ist doch in den 0.x.x-Versionen gar nicht enthalten, s. hier: 
https://github.com/steff393/wbec/releases/tag/v0.5.0
Das ist eine Funktion von wbecPro, wbecLan und wbecPremium.

closed 2024-01-24 by steff393

screwb0x, 2024-01-24:
Schade, aber dann seltsam das in der data.zip die time.js etc. enthalten ist und man sie auch hochladen kann…