steff393/wbec #120: Wallbox Status A setzen, wenn nicht erreichbar?

abuesen, 2024-02-03:
Ich habe zwei Heidelberg an evcc per MQTT laut wbec-Anleitung angebunden. 
Wenn eine WB offline ist, wird per MQTT der `Status F` gesetzt, Der ist laut angewendeter Dokumentation nur gültig, wenn ein Fz verbunden ist: https://evsim.gonium.net/#der-control-pilot-cp 

Das evcc meldet dadurch regelmäßig Fehler: `[lp-2  ] ERROR 2024/02/03 18:04:41 charger: invalid status: F`
Leider unterstützt es eine Wallbox Status "offline" nicht. Deshalb wäre `Status A` besser geeignet.

Ich versuche die Fehlermeldung weg zu bekommen, da ich dadurch "echte" Fehlermeldungen übersehe.

Siehe: https://github.com/evcc-io/evcc/discussions/12039#discussion-6174361

steff393, 2024-03-24:
Hi,
der Status F sollte von diesem Code kommen: 
https://github.com/steff393/wbec/blob/f3471ca92bc9c8223c2e9203b153a7e2e34b7aa1/src/mqtt.cpp#L217-L225

`content[i][1]` ist das gleiche wie `"chgStat"` unter http://wbec.local/json

Was ich aber noch nicht nachvollziehen kann ist, warum du einen Ladestatus außerhalb von 2-7 haben kannst, wenn eine WB offline ist. Normalerweise sollte der alte Wert erhalten bleiben.

Oder meinst du den Zustand, dass die Wallbox noch nie geantwortet hat? Dann wäre in der Tat `chgStat=0` und der status=F.
Dann würde es folgender Code lösen, oder?
`case 0:  ps = 0; cs = 0; status = 'A'; break;`

abuesen, 2024-03-24:
Abend!

Es kann sein, dass die WBox im Standby ist nd wbec danach einen Neustart bekommt. Das wäre der Fall, den Du beschreibst?

Das kann ich morgen probieren, bin gerade unterwegs. 
WBox aus-/einschalten schauen, danach ausschalten und schauen. Wenn dann der alte Status bleibt, erster Schritt abgeschlossen.
Dann wbec neustarten und schauen. 

Da würde sicher `case 0:` helfen.

closed 2024-03-30 by steff393

abuesen, 2024-04-10:
Danke für das Update. Der Test war erfolgreich.

Reset wbec nach dem Abschalten der 2. WB  
2.5.1: Status F
2.5.2: Status A