steff393/wbec #122: Docs: Multiple wbec instances?

sja, 2024-02-26:
I'd like to know, how wbec behaves if multiple of them are in the same network.

- What does that mean for DNS `wbec.local`?
- Am I able to change the hostname?
- Is it possible to link between each-others inside the webinterface?
- Where can I find that in the docs/wiki?

abuesen, 2024-02-26:
The `wbec.local` is just a DNS-name corresponding to the devices IPv4 address. The device tries to register that name with the DHCP request. If you connect the wbec to your network it will get an IPv4 via DHCP or you have to configure a static IPv4 at the device. You can use that IP instead of the name. Of course with DHCP it is possible this IP changes. 

If you own a router that manges DHCP and DNS, typicalliy your internet-router, it will manage the name of the device. For example I own a FritzBox (AVM) and configured the name `wbec` for the device. So it is reachable with `wbec`or `wbec.fritz.box` which is the local domain of this router.

So if integrated in a network the name `wbec.local` is not resolved by the device, it is resolved by your DNS.device in your network. There you have to look. 

There is no need to link between multible wbec, yet. It is possible to connect several Heidelberg-Wallboxes to one wbec. You need to wires between the wallboxes as a chain, it is a _modbus_. At the moment it's limited to two boxes and in my opinion ther is n loadmanagement between the boxes. How many do you want to connect?

You'll get a specialised documentation (german language) for wireing the wallboxes if you buy a wbec. Alternative one can google for modbus-wireing.

abuesen, 2024-02-26:
Hmmm, you seem to be german. Why do we talk in english? ;-)

closed 2024-03-23 by steff393