steff393/wbec #123: Minify HTML, JS und CSS

istmichseegal, 2024-02-29:
Ich würde dir vorschlagen deine HTML, JS und CSS Dateien mal durch eine Minifier laufen zu lassen bevor du sie zum Download als Release anbietest. Bei mir läuft dadurch die UI merkbar schneller.

steff393, 2024-03-23:
Gute Idee!
Ich hab das jetzt mal für den aktuellen Stand (x.5.1) machen lassen: [dataMin.zip](https://github.com/steff393/wbec/files/14731675/dataMin.zip)
Achtung: Wer selbst Anpassungen am HTML-Code machen möchte sollte lieber bei den Original-Dateien bleiben.

Genutzte Minifier:
JS: https://www.ultrapixel.de/tools/javascript-minifier
CSS und HTML: https://kangax.github.io/html-minifier/   (mit Default-Einstellungen)
Der komprimierte JS-Code wurde direkt in den HTML-Code (vor dessen Komprimierung) kopiert. Daher sind in o.g. ZIP keine .js-Dateien mehr enthalten.

csh-de, 2024-04-04:
sehr gut, kannst du das auch für die x.4.9 Data hochladen bitte?

steff393, 2024-04-06:
Hier sind die neuesten Daten passend zum Stand x.5.2:
https://github.com/steff393/wbec/releases/download/v0.5.2/data-min.zip

Für ältere Versionen werde ich es nicht mehr nachziehen.

closed 2024-04-06 by steff393

steff393, 2025-03-09:
Für x.5.4 habe ich jetzt nochmal einen neuen Satz gemacht, allerdings dieses Mal mit anderem JS-Minifier:
JS: https://www.toptal.com/developers/javascript-minifier
CSS und HTML: https://kangax.github.io/html-minifier/ (mit Default-Einstellungen)
Der komprimierte JS-Code wurde direkt in den HTML-Code (vor dessen Komprimierung) kopiert. Daher sind in o.g. ZIP keine .js-Dateien mehr enthalten.